realsenseai/realsense-ros #2490: modify the resolution but there is no change in rqt_image_viewer

NKdryer, 2022-09-26:
Hello, I published the topics of realsense by running `roslaunch realsense2_camera rs_camera.launch color_width:=462 color_height:=260`, then I open rqt_image_viewer but the image has no difference between `roslaunch realsense2_camera rs_camera.launch`, Do I miss some operations? Or how can I know whether the resolution has been modified? Thank you in advance.

NKdryer, 2022-09-27:
Thanks for the explanation. So whether 346 * 260 is a resolution supported by RealSense camera? If not, how can I display the images or adjust the collected images to this resolution.

MartyG-RealSense, 2022-09-27:
The data sheet document for the RealSense 400 Series cameras provides the following information.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

An RGB color resolution supported by all 400 Series camera models is 424x240.

The D415, D435, D445i and D435f models support 320x240 and 320x180 RGB color resolutions.

The post-processing **decimation filter** provides the ability to reduce a selected resolution but it is not possible to define a precise custom resolution such as 346x260, unfortunately.

https://dev.intelrealsense.com/docs/post-processing-filters#decimation-filter

In ROS, the function **image_proc/resize** described in Section 4.4 of the link below may help you to adjust the image size.

http://wiki.ros.org/image_proc

MartyG-RealSense, 2022-10-03:
Hi @NKdryer  Do you require further assistance with this case, please?  Thanks!

NKdryer, 2022-10-03:
Thanks for your explanation about this problem! But I still had another problem when using this camera in Linux.

For those **.yaml** files that contain calibration results for the RealSense camera and another camera, is there exists an interface in the **.launch** file that lead the camera to read the **.yaml** file at the specified location in my computer?

MartyG-RealSense, 2022-10-03:
Hi @NKdryer  Thanks very much for the update!

The link below may be a helpful reference for loading a .yaml file from the launch file.

https://roboticsbackend.com/ros-param-yaml-format/#In_a_launch_file

NKdryer, 2022-10-03:
So if I load the **.yaml** file containing calibration results in the **.launch** file correctly, would the **calibrated** images be displayed when selecting RealSense image_raw in the visualization gui?

MartyG-RealSense, 2022-10-03:
My understanding is that loading a .yaml file into the launch file should have the same effect on the image that it would if you had defined the parameters in the launch file directly.

closed 2022-10-17 by MartyG-RealSense