realsenseai/realsense-ros #2491: Can I change the default option of enable auto white balance" to "false" in rs_camera.launch?

RI-SMJ, 2022-09-27:
Dear, Realsense-ROS team.

Hello, Realsense-Ros team.
I use 2 L515 cameras in my ROS system, and now I want to know about some options of rs_camera.launch file.
In my robot environment, there are some problems because of auto white balance.
So, I want to change the value of the auto white balance as false, but there are no arg option in rs_camera.launch.
Can I add this option to your rs_camera.launch file?
And If i can, how should I write about this?
I'd appreciate it if you could answer me.

Thank you

MartyG-RealSense, 2022-09-27:
Hi @RI-SMJ  Yes, you can edit the **rs_camera.launch** launch file or create your own custom launch file.

Adding the code below to the launch file should disable the auto white balance option.

```
<rosparam>
  /camera/rgb_camera/enable_white_balance: false
</rosparam>
```

If you download the source code version of the ROS wrapper then in the source code folder you can find the launch files in the **\realsense2_camera\launch** directory.  you can edit the text of the file with a text editing program.

***********

Alternatively, you can try disabling it from the rs_camera roslaunch instruction so that you do not need to edit the launch file.

**roslaunch realsense2_camera rs_camera.launch enable_white_balance:=false**

RI-SMJ, 2022-09-27:
> Hi @RI-SMJ Yes, you can edit the **rs_camera.launch** launch file or create your own custom launch file.
>
> Adding the code below to the launch file should disable the auto white balance option.
>
> ``` [...]

I'm trying "roslaunch realsense2_camera rs_camera.launch enable_white_balance:=false" but, when I checked in rqt_reconfigure, enable_auto_white_balance is still on.
[image]

Also, when I type Tab during roslaunch, the option related to white_balance does not apper...
[image]

Is it right that the "enable_white_balance" an "enable_auto_white_balance" are the same?

MartyG-RealSense, 2022-09-27:
The developer of the ROS1 wrapper provided the information about enable_white_balance at https://github.com/IntelRealSense/realsense-ros/issues/1354#issuecomment-681126057

However, any parameter that is in rqt_reconfigure should have an equivalent parameter of the same name that can be set with a launch file rosparam.  So please try:

**roslaunch realsense2_camera rs_camera.launch enable_auto_white_balance:=false**

MartyG-RealSense, 2022-10-03:
Hi @RI-SMJ  Do you require further assistance with this case, please?  Thanks!

RI-SMJ, 2022-10-04:
> Hi @RI-SMJ Do you require further assistance with this case, please? Thanks!

No,sir. All issues have been resolved. Thank you very much for your kind reply.

closed 2022-10-04 by RI-SMJ

MartyG-RealSense, 2022-10-04:
You are very welcome.  It's excellent news that you achieved a solution.  Thanks very much for the update!