realsenseai/realsense-ros #2494: Problem recording rosbag when running SLAM with D400 series

W4li8, 2022-10-01:
Hi, we are trying to run the SLAM package from [the wiki](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i) but as soon as we start a `rosbag record` in parallel to the SLAM, we run into the following `[ERROR] Rtabmap.cpp:1277::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 1498 is ignored!` and everything just stops. It works fine without the data recording.

The setup we currently have is Ubuntu 20.04 with ROS Noetic installed on a high quality bootable SD card (100MB R/W) running on an i5 processor with 8GB RAM. Contrary to the wiki, we use the newer D455 camera and it is the only external peripheral.

We plan to export this to a Jetson Nano (SUB) where we experience similar issues except we haven't tested it to the same extent. On this SUB module, we redirect the boot from the 16GB EMMC to an SD card to expand the system's memory.

The gyro/accel of the D455 are explicitly enabled and run at around 200Hz with the camera fps at 30. This eliminates 'no odometry provided part' - it gets lost somewhere, causing the entire thing to crash.

Do you know what could be the cause of the SLAM crashing when a rosbag recording is started? 

As a lead into what might be the issue, sometimes the following errors come out too
`[ERROR] [1664314681.061529311]: Critical Error, NaNs were detected in the output state of the filter. This was likely due to poorly conditioned process, noise, or sensor covariances.`
`ERROR [140165586614016] (ds5-thermal-monitor.cpp:94) Error during thermal compensation handling: query is available during streaming only [camera/realsense2_camera-3] process has finished cleanly` (or not so cleanly)

On a similar note, what are the minimal system requirements recommended to run the proposed SLAM and where can I find more documentation about the package itself?

Best, Filip

MartyG-RealSense, 2022-10-01:
Hi @W4li8  Because of the age of the D435i SLAM guide, its prerequisites use ROS Kinetic packages instead of newer ROS versions such as the Noetic version of ROS that you are using.  I would suggest trying the guide with Noetic-compatible versions of the prerequisites if you have not done so already.  For example:

sudo apt-get install ros-noetic-robot-localization

sudo apt-get install ros-noetic-rtabmap-ros

sudo apt install ros-noetic-imu-filter-madgwick

*******

Before doing that though, please try activating rosbag record whilst the robot is stationary in order to eliminate the possibility that the errors are being caused by the robot moving too quickly during recording and the computing device being unable to keep up with the processing burden of map calculations and data recording simultaneously.

Also, does reducing the FPS from 30 to 15 make a positive difference?

Thermal compensation is a D455-only feature where adjustments are automatically made to the camera's calibration based on changes in temperature.

W4li8, 2022-10-03:
HI @MartyG-RealSense, I adjusted the installation guide for ros-noetic indeed. Regarding the reduction of fps, I modified the launch file with the following, does it seem correct? I ask because `rostopic hz` remains around 30 for color/image_raw and around 200 for the imu.
```
    <include unless="$(arg offline)" 
        file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="align_depth" value="true"/>
        <arg name="linear_accel_cov" value="1.0"/>
        <arg name="depth_fps"    value="15"/>
        <arg name="color_fps"    value="15"/>
        <arg name="gyro_fps"     value="150"/>
        <arg name="accel_fps"    value="150"/>
        <arg name="infra_fps"  value="15"/>
        <arg name="enable_gyro"  value="true"/>
        <arg name="enable_accel" value="true"/>
    
        <arg name="unite_imu_method" value="linear_interpolation"/>
    </include>
```
I will now take a better look at the links you sent.

The issue is likely to be linked with processing burden as you call it but I wonder where it might be coming from and how I could modify the launch file while maintaining reasonable performance.

---

Aside from that, we got lucky with one of our recordings, do you see a reason for these gaps?

[image]

MartyG-RealSense, 2022-10-04:
I suspect that your custom configuration may be being deemed invalid by the launch and the default stream configuration for your particular camera model is being applied instead.  This is because when defining a custom stream configuration, three factors must be provided for each stream type - width, height and FPS.  So if a 15 FPS value is custom-defined but stream width and height are not also defined (they are -1 by default in the rs_camera launch file), the default resolution and FPS speed (usually 30) would be applied automatically.

Please try the settings below:

```
<arg name="depth_width"    value="640"/>
<arg name="depth_height"    value="480"/>
<arg name="depth_fps"    value="15"/>

<arg name="color_width"    value="640"/>
<arg name="color_height"    value="480"/>
<arg name="color_fps"    value="15"/>

<arg name="infra_width"    value="640"/>
<arg name="infra_height"    value="480"/>
<arg name="infra_fps"    value="15"/>

<arg name="enable_gyro"  value="true"/>
<arg name="enable_accel" value="true"/>
<arg name="unite_imu_method" value="linear_interpolation"/>
<arg name="gyro_fps"     value="200"/>
<arg name="accel_fps"    value="200"/>
```

Regarding the gaps, they seem to occur for all stream types simultaneously, which could suggest that there is a communication problem with the camera during those time periods.  Could you check the launch log in the ROS terminal please to see whether **control_transfer returned** warning messages are generated when those gaps occur?

W4li8, 2022-10-04:
The settings you propose work, thanks. Also, I have checked shinkansan's adaptation of the wiki guide. The main difference I found is I record all topics instead of the selection he proposes. Since these other topics should require lighter data transfer I see no problem there.

Regarding the gap, there is no **control_transfer**. Here's what happened during my test with your proposed settings: I ran the SLAM for about 2min and then turned on the rosbag, gave it some time to stabilize the odometry quality and then continued. Aside from the warnings that started to appear (I attach those here for completeness, it is probably not linked to the camera)
```
[ WARN] [1664917606.440181963]: Many occupancy grids should be loaded (~394), this may take a while to update the map(s)...
[ WARN] [1664917606.449819460]: Many clouds should be added to octomap (~394), this may take a while to update the map(s)...

[ WARN] [1664917612.985705168]: Map(s) updated! (6.545540 s)
[ WARN] [1664917612.986120093]: /scan_map topic is deprecated! Subscribe to /cloud_map topic instead with <param name="Grid/Sensor" type="string" value="0"/>. Do "$ rosrun rtabmap_ros rtabmap --params | grep Grid" to see all occupancy grid parameters.
[ WARN] [1664917612.986311383]: Graph has changed! The whole cloud is regenerated.

[ WARN] [1664917632.954854872]: /scan_map topic is deprecated! Subscribe to /cloud_map topic instead with <param name="Grid/Sensor" type="string" value="0"/>. Do "$ rosrun rtabmap_ros rtabmap --params | grep Grid" to see all occupancy grid parameters.

[ WARN] [1664917634.456776111]: /proj_map topic is deprecated! Subscribe to /grid_map topic instead.
```
The gap occurred again and the closest error in the terminal is
`04/10 14:07:16,850 ERROR [139808849553152] (ds5-options.cpp:88) Asic Temperature value is not valid! `
Shortly after it crashes
```
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/ubuntu/.ros/log/b6d8225c-4423-11ed-8e9e-1fbdc4b31f6e/camera-realsense2_camera-2*.log
 04/10 14:07:34,040 ERROR [139809319261952] (ds5-thermal-monitor.cpp:94) Error during thermal compensation handling: query is available during streaming only
```
I attach the logs and bag preview, in case I forgot to mention something relevant.

[image]
[partial_log.txt](https://github.com/IntelRealSense/realsense-ros/files/9711046/partial_log.txt)

MartyG-RealSense, 2022-10-05:
The **Asic temperature is not valid** warning is likely not relevant to the problem.  It just means that the camera was not ready to provide a temperature reading for its internal circuit board at the moment that the SDK requested that information from the camera, but this is a self-recovering issue.

My research into the **Many occupancy grids should be loaded (~394), this may take a while to update the map(s)...** warning indicates that it is an issue related to the 'MapsManager.cpp' file of **rtabmap_ros**: https://github.com/introlab/rtabmap_ros/blob/master/src/MapsManager.cpp#L531

[image]

Your particular type of thermal compensation error - **query is available during streaming only** - makes me think that the camera may be disconnecting, which could account for the gaps in the data.  If a RealSense camera disconnects during streaming then so long as it can reconnect itself within 5 seconds then a timeout error will not occur and it can continue streaming from the point that it left off.

W4li8, 2022-10-05:
Indeed, those warnings aren't linked to the camera directly and I'll look into that separately.

What reasons would you deem likely to cause the camera to disconnect? The USB is well in place.

MartyG-RealSense, 2022-10-06:
If you are using a moving robot then the USB connector might shift slightly in the port without coming out and cause a disconnection and then a reconnection.  If you are not doing so already, using a USB-C cable with screw-locks on its micro-sized connector so that you can fix the connector firmly into the base of the camera via the screw-holes either side of the camera's port may help to reduce motion-related disconnect / reconnect.

[image]

W4li8, 2022-10-11:
I'm closing this issue. Reducing the data transfer as suggested got it working on my test computer. I will reopen if I run into new issues on the Jetson Nano 🙂  Thanks for your help @MartyG-RealSense

closed 2022-10-11 by W4li8

MartyG-RealSense, 2022-10-11:
You are very welcome, @W4li8 - I'm pleased to hear that you achieved a solution.  Thanks very much for the update!

Shrioe, 2024-06-15:
Hello @W4li8 , i faced the smae probelm with you while creating the map too, i tried to edit the rs_camera.launch file as shown below:

<arg name="depth_width"    value="640"/>
<arg name="depth_height"    value="480"/>
<arg name="depth_fps"    value="15"/>

<arg name="color_width"    value="640"/>
<arg name="color_height"    value="480"/>
<arg name="color_fps"    value="15"/>

<arg name="infra_width"    value="640"/>
<arg name="infra_height"    value="480"/>
<arg name="infra_fps"    value="15"/>

<arg name="enable_gyro"  value="true"/>
<arg name="enable_accel" value="true"/>
<arg name="unite_imu_method" value="linear_interpolation"/>
<arg name="gyro_fps"     value="200"/>
<arg name="accel_fps"    value="200"/>

But when I run "roslaunch realsense2_camera opensource_tracking.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation", there is no more NaN values errors but still:
"
15/06 14:26:33,788 WARNING [281472450236784] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ WARN] [1718432786.188037476]: Failed to meet update rate! Took 0.0067153130000000005317
15/06 14:26:33,890 WARNING [281472450236784] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b

In rviz,

There is no map shown.
I cannot tick the "map" or "camera_link" because under "frame", there is only "All enabled".
Then, under "image", there is no image shown even though i changed to "cameracolor/image_raw"
[image]

May I ask can I get some suggestions from you on how did you solve the issue and how can you create the final map?

Thank you!