realsenseai/realsense-ros #2495: process has died ddynamic reconfigure

Akumar201, 2022-10-03:
Greetings, 
I would like to know how can I solve this problem ? While running the command roslaunch realsense2_camera rs_camera.launch
I am getting this error 

/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /home/akumar/catkin_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleE
[camera/realsense2_camera_manager-2] process has died [pid 12234, exit code 127, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/akumar/.ros/log/fd82edfc-434e-11ed-9d35-43b6dbe9278d/camera-realsense2_camera_manager-2.log].
log file: /home/akumar/.ros/log/fd82edfc-434e-11ed-9d35-43b6dbe9278d/camera-realsense2_camera_manager-2*.log

Seems indeed linked to ddynamic reconfigure but ... can't figure out how to deal with that.

MartyG-RealSense, 2022-10-03:
Hi @Akumar201  The error **undefined symbol: _ZN20ddynamic_reconfigure** is typically related to ddynamic_reconfigure being missing.  It can be installed for ROS Noetic using the command below.

**sudo apt-get install ros-noetic-ddynamic-reconfigure**

Akumar201, 2022-10-06:
Thank you for your response.  I resolved it by removing it completely and reinstalling using apt 

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

@MartyG-RealSense Thank you for your response , it was really helpful

closed 2022-10-06 by Akumar201

MartyG-RealSense, 2022-10-06:
You are very welcome, @Akumar201 - I'm pleased to hear that you achieved a solution.  Thanks for the update!