realsenseai/realsense-ros #2496: D435i how to turn auto exposure ON/OFF on launch?

rakshith95, 2022-10-05:
Hello, 
I'm unable to find a parameter to turn the auto exposure ON/OFF for the D435i cameras.   The only way I can do it now is to run `rosrun rqt_reconfigure rqt_reconfigure` and do it through the GUI, but I wish to do this from a script.

Thank you!

MartyG-RealSense, 2022-10-05:
Rosparam code for depth and color auto-exposure enable / disable is below.  

**OFF**

<rosparam>
  /camera/rgb_camera/enable_auto_exposure: false
  /camera/stereo_module/enable_auto_exposure: false
</rosparam>

**ON**

<rosparam>
  /camera/rgb_camera/enable_auto_exposure: true
  /camera/stereo_module/enable_auto_exposure: true
</rosparam>

rakshith95, 2022-10-10:
@MartyG-RealSense setting the parameters doesn't seem to work for me. The

MartyG-RealSense, 2022-10-15:
hi @rakshith95  Do you require further assistance with this case, please?  Thanks!

rakshith95, 2022-10-17:
No, thank you.

closed 2022-10-17 by rakshith95

MartyG-RealSense, 2022-10-17:
Thanks very much @rakshith95 for the update!

chenhanpolyu, 2023-03-08:
@MartyG-RealSense Hi, I am also facing with an auto exposure problem. I found that when I enable the laser emitter, my d435i cannot auto-exposure correctly. I test it with realsense-viewer. When I enable auto-exposure with emitter off, the actual exposure of infrared image can change with the lighting condition. But once I turn on the emitter (when the infrared camera is on), the actual exposure is freezing and will not change with lighting. In addition, when I turn on the camera with auto-exposure on and emitter on, the actual exposure is fixed at the maximal value at current fps (about 31400 us for 30 fps). I hope to confirm is this normal? Thank you!

MartyG-RealSense, 2023-03-08:
Hi @chenhanpolyu  How the depth auto-exposure should operate when correct is that exposure can update but the **exposure value** can no longer update until auto-exposure is disabled.  Instead, depth auto-exposure should be read from the camera **metadata** parameter ACTUAL_EXPOSURE

SamerKhshiboun, 2023-03-22:
Take a look on this PR #2612, it mainly for ROS2, but it should be the same solution for ros1 by editing your launch file.