realsenseai/realsense-ros #2498: D435i Orin dev kit performance?

verderog, 2022-10-07:
I'm running the `ros2-beta` branch and have ROS2 Humble Hawksbill operating successfully with the D435i camera on my NVIDIA Orin dev kit.  My librealsense lib build is at 2.51.1.  

I see the following when running `ros2 launch realsense2_camera rs_intra_process_demo_launch.py intra_process_comms:=true`:

```
[component_container-1] [INFO] [1665157605.224020588] [frame_latency]: Got msg with address 0xffff94bfc5e0 with latency of 0.038842 [sec]
[component_container-1] [INFO] [1665157605.258509746] [frame_latency]: Got msg with address 0xffff94bfc5e0 with latency of 0.0399335 [sec]
[component_container-1] [INFO] [1665157605.290734940] [frame_latency]: Got msg with address 0xffff94bfc5e0 with latency of 0.0388389 [sec]
[component_container-1] [INFO] [1665157605.324291127] [frame_latency]: Got msg with address 0xffff94bfc5e0 with latency of 0.0390485 [sec]
```

These numbers are higher than what I was expecting given that intra-process comms are enabled.  Surprisingly, there's not much of a difference if I enable or disable intra_process_comms -- the latency numbers are about the same.  I'm relatively new to the RealSense and am hoping for a sanity check.  Can anyone else confirm these numbers make sense?

MartyG-RealSense, 2022-10-07:
Hi @verderog  As far as I am aware, you are the first RealSense ROS user to report results from this launch file and so there is not another set of figures from other users to compare to, unfortunately.

Checking the launch file contents, it looks as though this launch file disables depth and only publishes the color stream.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-beta/realsense2_camera/launch/rs_intra_process_demo_launch.py#L33-L45

You may experience better performance on a Jetson board if you have built librealsense with **CUDA** support enabled, as CUDA support accelerates YUY to RGB color conversion operations (and also accelerates pointcloud generation and depth-color alignment).  CUDA will be automatically enabled if you installed librealsense from Jetson packages, or if you built librealsense from source code with CMake and included the term **-DBUILD_WITH_CUDA=true** in the CMake build instruction.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

Another method for benchmarking performance would be to perform an rs_camera py launch:

**ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x30**

and then check the actual published rate of the depth stream with this instruction:

**rostopic hz /camera/depth/image_rect_raw**

verderog, 2022-10-07:
@MartyG-RealSense Thanks for the comments!

I compiled the SDK from source with CUDA so I'm good there.  

When I last ran `ros2 topic hz /camera/color/image_raw`, I was seeing 29fps being reported.  I'm using https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_image_pipeline as a baseline for my performance testing.  When I launch the NVIDIA side of things (`ros2 run isaac_ros_image_proc isaac_ros_image_proc --ros-args -r /image_raw:=/camera/color/image_raw --ros-args -r /camera_info:=/camera/color/camera_info`), the reported framerate dips from 29 to 24.

I'll play around with the option you suggested.  Thanks for the info.

MartyG-RealSense, 2022-10-08:
Thanks very much @verderog for the performance information - good luck with your testing!

MartyG-RealSense, 2022-10-19:
Hi @verderog  Do you have an update about this case that you can provide, please?  Thanks!

verderog, 2022-10-21:
@MartyG-RealSense No new updates.  I'll go ahead and close this issue out.

closed 2022-10-21 by verderog

MartyG-RealSense, 2022-10-22:
Thanks very much @verderog for the update.