realsenseai/realsense-ros #2499: No support for macOS yet? (M1 mac)

JinraeKim, 2022-10-07:
Hi, I'm looking for using this on M1 mac.
Is there any way to use it? I cannot find any installation guide for mac.
It doesn't matter to use docker if needed.

MartyG-RealSense, 2022-10-07:
Hi @JinraeKim  If you are seeking to install the librealsense SDK on Mac M1, there is a detailed step by step guide at the link below.

https://lightbuzz.com/realsense-macos/

Installation support for M1 is also provided by **brew install librealsense**

https://formulae.brew.sh/formula/librealsense

MartyG-RealSense, 2022-10-07:
You are very welcome!

Yes, the librealsense SDK should be installed first and the RealSense ROS wrapper built secondly.

JinraeKim, 2022-10-07:
> You are very welcome!
> 
> Yes, the librealsense SDK should be installed first and the RealSense ROS wrapper built secondly.

Sorry for keeping asking.
As I know, I can only install ROS2 on M1 Mac, not ROS(1).

Then, should I follow the instruction for ROS2 in this repo as well?

MartyG-RealSense, 2022-10-07:
I researched your question but did not find an instance of running the ROS wrapper directly on MacOS.  You may need to use Ubuntu on your Mac in Docker or in a Virtual Machine (VM), as described for M1 in the guide at the link below.

https://yangrobotics.com/ubuntu-and-ros-setup-on-m1-macbook-air

The creator of the guide stated that they were able to use ROS1 Melodic on their M1 with Docker.

JinraeKim, 2022-10-10:
I could also test ROS1 on my M1 mac with docker, but my concern is the use of Realsense tgt cuz it must be related to USB data stream as well.
Anyway, thanks for finding good references!

MartyG-RealSense, 2022-10-20:
Hi @JinraeKim  Do you require further assistance with this case, please?  Thanks!

JinraeKim, 2022-10-20:
> Hi @JinraeKim Do you require further assistance with this case, please? Thanks!

No, thanks! I'll close this issue with this comment.

closed 2022-10-20 by JinraeKim