realsenseai/realsense-ros #2503: ROS2-beta: RGB and depth FPS is lower than configured

guoqingzh, 2022-10-13:
launch ROS2 realsense node and echo rgb/depth topic with ros2 topic hz. The report FPS is much lower than configurated FPS

MartyG-RealSense, 2022-10-13:
Hi @guoqingzh  If you are launching in ros2_beta with **rs_launch.py**, do you get improved FPS if you launch with **ros2 run**

`ros2 run realsense2_camera realsense2_camera_node`

guoqingzh, 2022-10-13:
> Hi @guoqingzh If you are launching in ros2_beta with **rs_launch.py**, do you get improved FPS if you launch with **ros2 run**
> 
> `ros2 run realsense2_camera realsense2_camera_node`

this is the command I used:

ros2 launch realsense2_camera rs_launch.py camera_name:='camera2' serial_no:="'_xxxxxxxxx'" rgb_camera.profile :='1280,720,30'

MartyG-RealSense, 2022-10-13:
What is the actual FPS that is being reported?

Also, can you confirm whether the camera is being detected as being on a USB 3 connection.  If it was being detected as USB 2.1 (either because the connection was mis-detected or the camera is plugged into a USB 2 port) then the only FPS speed supported for 1280x720 resolution would be 6 FPS (5 on the D455 model).

MartyG-RealSense, 2022-10-13:
Thank you.  And what is the FPS for color when launching with **ros2 run realsense2_camera realsense2_camera_node**

guoqingzh, 2022-10-13:
> Thank you. And what is the FPS for color when launching with **ros2 run realsense2_camera realsense2_camera_node**

it is the same

MartyG-RealSense, 2022-10-13:
I note from the title of this case that you are using both depth and RGB.  

As you are using **camera_name:='camera2' ** do you have more than one camera enabled at the same time?

Does the FPS improve if you disable the RGB option **auto_exposure_priority** with the instruction below?

`ros2 param set /camera/camera rgb_camera.auto_exposure_priority false`

guoqingzh, 2022-10-13:
> As you are using **camera_name:='camera2' ** do you have more than one camera enabled at the same time?
> 
> I note from the title of this case that you are using both depth and RGB. Does the FPS improve if you disable the RGB option **auto_exposure_priority** with the instruction below?
> 
> `ros2 param set /camera/camera rgb_camera.auto_exposure_priority false`

Yes, the idea was to run 4 RS camera streaming both RGB and depth with difference resolution and FPS. But the FPS drop is reproducible with only 1 RS.  I will try the auto_exposure_priority params and let you know the result

guoqingzh, 2022-10-14:
@MartyG-RealSense the suggested parameters auto_exposure_priority false doesn't help. For depth, I set it to 30 FPS but only got ~13 FPS. But if  I set depth_module.enable_auto_exposure to true , that seems to help (increase from 13 fps to 27 fps but still not 30 fps )

MartyG-RealSense, 2022-10-14:
For the disabling of auto_exposure_priority to work, **auto_exposure** must also be set to true in order for a constant FPS rate to be enforced.  So if you have auto_exposure_priority = false and auto_exposure = true and FPS increases from 13 to 27, then the FPS rate enforcement mechanism may be working correctly.  There could be another factor pulling the FPS down from 30 to 27 though.

Is 27 fast enough for your particular project if 30 cannot be achieved?

MartyG-RealSense, 2022-10-20:
Hi @guoqingzh  Do you have an update about this case that you can provide, please?  Thanks!

guoqingzh, 2022-10-20:
@MartyG-RealSense  we are doing performance testing. We want the FPS to be exactly the value we set with +/- 1 error maximum .

MartyG-RealSense, 2022-10-20:
Which versions of the librealsense SDK and ros2_beta wrapper are you using, please?  For example, librealsense version **2.51.1** should be matched with ros2_beta wrapper version **4.51.1**

Do you see any improvement in performance if you reset the cameras at launch with **initial_reset**

`ros2 launch realsense2_camera rs_launch.py camera_name:='camera2' serial_no:="'_xxxxxxxxx'" rgb_camera.profile :='1280,720,30' initial_reset:=true`

dheera, 2022-10-20:
I'm also having massive issues with FPS on the ROS2 realsense wrapper.
I have noetic (ROS1) and foxy (ROS2) installed on the same machine. I recorded bags of /camera/color for both ROS1 and ROS2, the ROS1 bags have no issue, framerate is exactly as expected, ROS2 has significant frame drop.

MartyG-RealSense, 2022-10-21:
Hi @dheera  Please create a new case for your problem by visiting the link below and clicking the green **New Issue** button.  Thanks!

https://github.com/IntelRealSense/realsense-ros/issues

MartyG-RealSense, 2022-10-25:
Hi @guoqingzh  Do you require further assistance with this case, please?  Thanks!

Gilaadb, 2022-10-25:
Hi @guoqingzh. I also had issues with FPS using D455 camera and running depth and RGB streams. The issue is resolved if running in auto exposure mode. There is a bug (probably) that blocks me from setting auto exposure to true when starting the camera node, so what I do is add a delayed action in the launch file to set it a bit later:

`
TimerAction(
    period=10.0,
    actions=[
        ExecuteProcess(
            cmd=[[
                'ros2 param set /camera/camera depth_module.enable_auto_exposure true'
            ]],
            shell=True
        )
    ]
),
`

guoqingzh, 2022-10-25:
> Hi @guoqingzh. I also had issues with FPS using D455 camera and running depth and RGB streams. The issue is resolved if running in auto exposure mode. There is a bug (probably) that blocks me from [...]

Yes, Gilad. Alex told me the work around. I am reporting this to librealsense to let them fix the "bug"

closed 2022-10-31 by guoqingzh

SamerKhshiboun, 2023-01-31:
see #2507 for more information about this issue.