realsenseai/realsense-ros #2504: Error with building from source: Could not find a package configuration file provided by "librealsense2"

hassan-tht, 2022-10-14:
Hi

I'm trying to install the ros wrapper by building from source, but I get the following error when I try to colcon build:
`CMake Error at CMakeLists.txt:65 (find_package):`
`  By not providing "Findlibrealsense2.cmake" in CMAKE_MODULE_PATH this`
`  project has asked CMake to find a package configuration file provided by "librealsense2", but CMake did not find one.`
`  Could not find a package configuration file provided by "librealsense2"  with any of the following names:`
`    librealsense2Config.cmake`
`    librealsense2-config.cmake`
`  Add the installation prefix of "librealsense2" to CMAKE_PREFIX_PATH or set  "librealsense2_DIR" to a directory containing one of the above files.  If  "librealsense2" provides a separate development package or SDK, be sure it  has been installed.`

 I followed the instructions to install the dependencies and successfully built the librealsense from source by doing this:
` git clone https://github.com/IntelRealSense/librealsense`
` cd librealsense/`
` mkdir build`
` cd build/`
` cmake ..`
` sudo make` 
` sudo make install`

I verified this by running the realsense-viewer which launched successfully

Next I tried to install realsense-ros by doing this:
` cd src` 
` git clone https://github.com/intel/ros2_intel_realsense.git` 
` cd ..` 
` colcon build --base-paths src/ros2_intel_realsense` 
and I get the error mentioned above

Should I change the CMake file to point to where the librealsense is installed?

MartyG-RealSense, 2022-10-14:
Hi @hassan-tht  The colcon instruction that you are using is from a RealSense ROS2 wrapper that was not developed by the current RealSense ROS team and has been obsolete for 3 years.   The current ROS2 wrapper **ros2_beta** that supports Ubuntu 22.04 and ROS2 Humble can be found at the link below.

https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta

hassan-tht, 2022-10-18:
Yes that did it, thanks!

closed 2022-10-18 by hassan-tht

MartyG-RealSense, 2022-10-18:
You are very welcome, @hassan-tht - it's great to hear that you were successful with the ros2_beta wrapper installation.  Thanks very much for the update!