realsenseai/realsense-ros #2505: How to use opensource_tracking's odom on move_base

riderpunch, 2022-10-18:
Hello, I am a new user of ros. I plan to use d435i for opensource_tracking.launch to execute slam and automatically control the movement of create2 robot through move_base.
Opensource_tracking.launch already has odom data, but it seems that it cannot be directly applied to move_base. I would like to ask how to modify it so that it can be applied to move_base.

MartyG-RealSense, 2022-10-18:
Hi @riderpunch  Data published by the RealSense ROS wrapper is not commonly used for autonomous navigation in SLAM, but instead for generating maps of an area or obstacle detection and avoidance.  

The project that most comes to mind when considering using wrapper data for navigation is the **SAWR** one at the link below, which is a complete hardware and software blueprint for creating a ROS-driven mobile robot.  The guide was written in 2017 shortly before the RealSense 400 Series cameras were launched but the author believed that the system described in the guide could be adapted for the 400 Series.

https://www.intel.com/content/www/us/en/developer/articles/technical/build-an-autonomous-mobile-robot-with-the-intel-realsense-camera-ros-and-sawr.html

In that project, autonomous navigation was enabled via move_base using a **SAWR navigation** package with made use of **gmapping** (though the author stated that other mapping solutions could be used).

https://github.com/intel/sawr/tree/master/sawr_navigation

riderpunch, 2022-10-19:
Thank you for your information.
I will study his approach first.

MartyG-RealSense, 2022-10-25:
Hi @riderpunch  Do you require further assistance with this case, please?  Thanks!

riderpunch, 2022-10-25:
I studied his method
The key point is to use the depth to laserscan ros package to instead of lidar, but because of his package, I compiled a lot of errors, so I only use the two ros packages of gammapping and depth to laserscan.
Then the situation I tested is that the machine can reach the specified point in very few cases, and in most cases it rotates in place.
So still looking for other ways.
I have seen this article so far:
https://roverrobotics.com/blogs/guides/fusing-imu-encoders-with-ros-robot-localization
Not sure if it works or not, will try again in a few days.

MartyG-RealSense, 2022-10-25:
Thanks very much for the update.  I look forward to your next report after your tests.  Good luck!

MartyG-RealSense, 2022-11-04:
Hi @riderpunch  Do you have an update about this case that you can provide, please?  Thanks!

riderpunch, 2022-11-04:
Yes.
I found that opensource_tracking.launch is indeed very similar to the above article in terms of fusion.
All use the robot localizotion package and use the classic Kalman filter for tracking.
But since I am a beginner in this area, it may take a long time to learn the meaning of the parameters of each instruction.
If I open this question will affect your work, I can close it.

MartyG-RealSense, 2022-11-05:
It is fine to close the case now and re-open it at a later date or start a new case.  Thanks!

closed 2022-11-05 by riderpunch

riderpunch, 2022-11-29:
I was thinking too complicated.
There is an odom1 option in ukf_template.yaml, just put the the car odom topic to replace it and change world frame to map.
It can complete fusion.
But because the realsense data are all under camera_link, there will be many problems in TF conversion, so I converted all camera_link related TF trees in the realsesnse ros package to base_link
Finally, add the TF tree converted from map and odom when starting the move_base node, so that move_base can use the fused odom.

MartyG-RealSense, 2022-11-29:
Thanks so much for the update about your success and for sharing your solution with the RealSense ROS community!