realsenseai/realsense-ros #2506: Library / Wrapper version warning (Ubuntu 20.04 Ros2)

geoeo, 2022-10-18:
I just cloned the ros2-beta branch and compiled the ros2 wrapper. However at startup I am getting the a warning: `running with a different librealsense version`

Built is v.2.50.0
Running is v.2.51.1

I checked the installed sdk version to be v.2.51.1 via `apt-cache policy librealsense2-dev`

I am wondering where this 2.50.0 version is comming from?

MartyG-RealSense, 2022-10-18:
Hi @geoeo  This message would typically appear if the wrapper was compiled under one librealsense version and then the librealsense version was changed.  Each time that the librealsense version is changed, the ROS wrapper should be built again in order to avoid this warning.

An example of how the quoted warning could be produced is if 2.50.0 had been installed on an earlier occasion, and then the ROS wrapper was built.  If librealsense 2.51.1 was built after that without subsequently rebuilding the ROS wrapper then this warning could occur.

geoeo, 2022-10-18:
So I know that a previous version of the wrapper has been built on that machine. But if I check out a the git repo again and build that repo for the first time how can that warning occur? 

Also just to be clear by wrapper I mean the realsense-ros repo

geoeo, 2022-10-19:
I didnt install the sdk from packages. I always installed it via apt. This is what I dont understand. My current SDK version is 2.51. So why does the ROS2 wrapper complain that at some point a 2.50 was present, even if it isnt currently.

geoeo, 2022-10-19:
The issue is that `ros-galactic-librealsense` is still at version 2.50.0

closed 2022-10-19 by geoeo