realsenseai/realsense-ros #2507: Ros2 depth_module.auto_exposure_enabled via launch file doesnt do anything

geoeo, 2022-10-19:
If I start the camera node via `ros2 launch realsense2_camera rs_record.py config_file:="'../my_config.yaml'"` with `depth_module.enable_auto_exposure: true`, the auto exposure is not enabled.

I can check the state of the parameter with `ros2 param get /camera/camera depth_module.enable_auto_exposure` and it returns true!

If I set it again via `ros2 param get /camera/camera depth_module.enable_auto_exposure true` then it works and the infra1 images changes exposure

MartyG-RealSense, 2022-10-19:
Hi @geoeo  Other RealSense ROS users have tried setting the true / false status of auto_exposure or auto_exposure_priority from roslaunch instruction, launch file or yaml.  It only works when set during runtime after launch.

**ROS1**
rosrun dynamic_reconfigure dynparam set /camera/stereo_module enable_auto_exposure 1

**ROS2_BETA**
ros2 param set /camera/camera depth_module.enable_auto_exposure true

geoeo, 2022-10-19:
Ok thanks for the confirmation

geoeo, 2022-10-19:
@MartyG-RealSense Btw. This seems to be a regression as it used to work under v. 2.50

MartyG-RealSense, 2022-10-19:
I have referred this issue to the RealSense ROS team.

SamerKhshiboun, 2022-10-19:
Hi @geoeo,

My name is Samer, and I'm from the RealSense ROS Team.
Can you please provide me the full command of the ros2 launch cmd ?
Also, if you can provide the launch file (or paste it here).

Thanks,
Samer

geoeo, 2022-10-19:
`ros2 launch realsense2_camera rs_launch.py config_file:="'path/to/file/launch.yaml'"`

```
camera:
    camera:
      ros__parameters:
        device_type: ''
        serial_no: ''
        usb_port_id: ''

        depth_qos: "SYSTEM_DEFAULT"
        infra_qos: "SYSTEM_DEFAULT"

        gyro_fps: 200
        accel_fps: 100

        enable_accel: true
        enable_gyro: true
        enable_depth: true
        enable_infra1: true
        enable_sync: true
        enable_color: false
        enable_infra2: false
        enable_pointcloud: false

        unite_imu_method: 1
        filters: ''

        depth_module:
            profile: 640x480x5
            emitter_enabled: 1
            align_depth: true
            enable_auto_exposure: true
            auto_exposure_limit_toggle: true
            auto_gain_limit_toggle: false
            

        rosbag_filename: ''
```

SamerKhshiboun, 2022-10-19:
Thanks for the config file.
Can you also provide the launch file "rs_launch.py" ?
Should be under realsense2_camera/launch/

SamerKhshiboun, 2022-10-19:
I tried it on my environment, with your config file, and it works as expected.

Terminal 1:
`ros2 launch realsense2_camera rs_launch.py config_file:="'config.yaml'"`

Terminal 2:
`ros2 param get /camera/camera depth_module.enable_auto_exposure`
`Boolean value is: True`

Are you sure the path to your file is correct ? try to put it in your current working dir where you run the ros2 run command (like mine) and then check again.
If that still happen, I will investigate it more. BTW, which camera model are you using ?

Regards,
Samer

geoeo, 2022-10-19:
Yes the parameter is True, but the autoexposure does not seem to be triggered in e.g. infra1.

geoeo, 2022-10-19:
you can check in rviz2. launch the node with the config and look at the image. Then after some seconds set the parameter manually via `ros2 param set` you should now see the autoexposure triggering

MartyG-RealSense, 2022-10-25:
Hi @geoeo  Do you require further assistance from @SamerKhshiboun with this case, please?  Thanks!

geoeo, 2022-10-25:
@MartyG-RealSense  If this is a confirmed bug on your end feel free the close the issue

Nir-Az, 2022-10-25:
Let's keep this open until we get our investigation results and then we'll update this ticket.

geoeo, 2022-11-14:
Interestingly the config flag works for the rgb module

Adum888, 2023-01-17:
Just wanted to ask whether there is any alternative to --> deactivating --> activating the parameter during runtime,by now?

geoeo, 2023-01-17:
@Adum888 my fix for now is to trigger the parameter via ros param set in the launch.py file with a delay.

SamerKhshiboun, 2023-01-31:
Hi @geoeo ,

I started debugging this interesting issue again, and looks like one of the parameters inside the launch file `depth_module.exposure.1` is auto-disabling the `enable_auto_exposure` feature of the depth module, so the FPS is lower than expected.

These `depth_module.exposure.1` and `depth_module.exposure.2` (also `depth_module.gain.1` and `depth_module.gain.2`) are relevant for HDR feature, so I think you can comment out these params in the rs_launch.py, and try launching your ros2 node again: 
`ros2 launch realsense2_camera rs_launch.py 
` 
you can add **`depth_module.enable_auto_exposure:=true`** to the command, although the auto exposure feature should be enabled by default for depth_module.

Another way to get a solution, is to run your node with **`ros2 run`** command, which will ignore the launch file and start the node with its default values:
`ros2 run realsense2_camera realsense2_camera_node
`
with ros2 run you can also override params, but in a different way, using the **`--ros-args -p <name>:=<value>`**, for example:
`ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_color:=true -p enable_infra1:=false -p enable_infra2:=false`

I will continue investigating why and how these params are being affecting each other, and how we can overcome in an elegant way.

Will update ASAP.

geoeo, 2023-02-12:
@MartyG-RealSense No I dont think so

MartyG-RealSense, 2023-02-12:
Okay, thanks very much @geoeo for the update!

@SamerKhshiboun  Do you wish to keep this case open whilst you investigate the parameters that you mentioned in https://github.com/IntelRealSense/realsense-ros/issues/2507#issuecomment-1411214372

SamerKhshiboun, 2023-03-19:
Hi @MartyG-RealSense,
Yes, keep this issue open please, till we finalize our investigation and propose a proper solution for that.

MartyG-RealSense, 2023-03-19:
Okay, that's no problem at all @SamerKhshiboun   :)

SamerKhshiboun, 2023-03-22:
#2612

MartyG-RealSense, 2023-03-23:
Hi @geoeo and @Adum888  Do you require further assistance after the fix implemented by @SamerKhshiboun at 
https://github.com/IntelRealSense/realsense-ros/pull/2612 and merged into the **ros2_development** ROS2 wrapper?  Thanks!

Adum888, 2023-03-23:
Thanks for the fix.
The ticket can be closed from my side

MartyG-RealSense, 2023-03-23:
Thanks very much @Adum888 - I will wait for confirmation from @geoeo as they created this issue.   Thanks very much for the update from your side.

geoeo, 2023-03-24:
I think its fine. It will be a while before I can get back to this. If the error persists I will repoen @MartyG-RealSense

MartyG-RealSense, 2023-03-24:
Thanks very much @geoeo for the update!  Please do feel free to re-open at a later date if you need to.

closed 2023-03-24 by MartyG-RealSense

drapado, 2023-10-20:
I still find depth auto-exposure disabled as default behaviour: https://github.com/IntelRealSense/realsense-ros/issues/2907