realsenseai/realsense-ros #2508: Unable to connect D435 jetson

erbizard, 2022-10-19:
Hi,

I'm trying to connect the D435 to a jetson nano with the realsense-ros
It work well when i use the rs_camera.launch but I only have one camera binded to topics (the left one).

When I use the rs_multiple_devices.launch I have an error message, all my topics stay empty, and some are not even initilized

Here are the initialisation logs :
` roslaunch realsense2_camera rs_multiple_devices.launch
WARNING: Package name "ByTel-ros-steering" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
... logging to /home/agilex/.ros/log/a62bd3de-4fb5-11ed-bb25-48b02d3cba1c/roslaunch-nano-25991.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "ByTel-ros-steering" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
started roslaunch server http://nano:46005/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 0
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file:
 * /camera1/realsense2_camera/clip_distance: -1.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/confidence_fps: 30
 * /camera1/realsense2_camera/confidence_height: 480
 * /camera1/realsense2_camera/confidence_width: 640
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/device_type:
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_confidence: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_infra: False
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_pose: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters:
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: 30
[... 197 lines cut ...]

And there the error logs : 
` 19/10 23:09:54,671 WARNING [547608326528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 19/10 23:09:54,822 WARNING [547608326528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1666192200.485705680]:
[ WARN] [1666192200.517668771]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1666192200.517772577]: The requested device with  is NOT found. Will Try again.
 19/10 23:10:00,516 ERROR [547046277504] (handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 19/10 23:10:00,516 ERROR [547287654784] (sensor.cpp:572) acquire_power failed: failed to set power state
 19/10 23:10:00,517 WARNING [547287654784] (rs.cpp:310) null pointer passed for argument "device"
[ INFO] [1666192206.559226257]:
[ WARN] [1666192206.589995446]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1666192206.590091595]: The requested device with  is NOT found. Will Try again.
 19/10 23:10:06,588 ERROR [547054670208] (handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 19/10 23:10:06,588 ERROR [547287654784] (sensor.cpp:572) acquire_power failed: failed to set power state
 19/10 23:10:06,589 WARNING [547287654784] (rs.cpp:310) null pointer passed for argument "device"
[ INFO] [1666192212.631176770]:
[ WARN] [1666192212.661115352]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1666192212.661211762]: The requested device with  is NOT found. Will Try again.
 19/10 23:10:12,660 ERROR [547063062912] (handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 19/10 23:10:12,660 ERROR [547287654784] (sensor.cpp:572) acquire_power failed: failed to set power state
 19/10 23:10:12,661 WARNING [547287654784] (rs.cpp:310) null pointer passed for argument "device"
[ INFO] [1666192218.704307590]:
[image]
`

MartyG-RealSense, 2022-10-19:
Hi @erbizard  The **rs_multiple_devices** launch file is for launching more than one RealSense camera at the same time (such as a pair of D435i cameras).  So it is not intended for use with only one camera and **rs_camera.launch** is the most suitable choice of launch file for a single D435i camera.

*****************

If you do have more than one D435i camera attached to the Jetson then the rs_multiple_devices launch instruction should be formatted like this:

`roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=<serial number of the first camera> serial_no_camera2:=<serial number of the second camera>`

So if you had two D435i cameras - for example, with two different serial numbers 12345678 and 12345679 - then the launch instruction for camera 1 (12345678) and camera 2 (12345679) would look like this:

`roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=12345678 serial_no_camera2:=12345679`

****************

If you have just one D435i camera but want to access both the **left and right sensors** then you can enable the left and right **Infrared** streams with the rs_camera launch instruction below:

`roslaunch realsense2_camera rs_camera.launch enable_infra1:=true enable_infra2:=true`

The right-side infrared sensor is only accessible on a USB 3 connection.  The line **Device USB type: 3.2** in your launch log indicates that you are on a USB 3 connection and so should be able to enable the right-side (infra2) infrared stream.

erbizard, 2022-10-20:
Hi Marty,

This is not for a D435i but a D435, in the readme it's specified that `The "/camera" prefix is the default and can be changed. Check the rs_multiple_devices.launch file for an example. If using D435 or D415`
Cause the camera prefix wont allow multiple image_row topic I suppose

MartyG-RealSense, 2022-10-20:
Can you confirm please that you are using a single D435 camera.  If you are, can you provide information please about what you are aiming to achieve.  Thanks!

erbizard, 2022-10-20:
Yeah i use one single D435 camera, i'm trying to make a robot follow a line drawn on the floor for the moment, but the problem is since i only can use one of the 2 camera the image isn't centered and the line is always on the left of the robot (that make the robot walk on the line and isn't optimal) I can make it work by adding a static var to the image reader. But with the 2 camera this wiil be cleanier (Sorry for my bad english)

MartyG-RealSense, 2022-10-20:
No problem, your English is very good  :)

rs_multiple_devices.launch would not be the correct launch file to use for one single camera as it is designed for more than one RealSense camera, like in the image below.  A D435 with left and right **sensors** counts as one camera and so is suitable for use with rs_camera.launch.  

[image]

It sounds as though you are using the left Infrared image (Infra1).  Is that correct, please?  If it is corect, does your project have the option to display the depth image instead, as the depth image is constructed from the left and right infrared frames.

erbizard, 2022-10-20:
I actually use the "/camera/color/image_raw" topic that allow me to see only one of the 2 sensor, But it's not infrared it's a camera. Unfortunatly depth didn't do the job since it's a monochrome image and i can't discriminate a line without colors, the best way will be to have a second image_raw topic who give me access to the second camera.
I don't know if it's possible ?

MartyG-RealSense, 2022-10-20:
How about aligning depth to color with **align_depth** set to true, and using the topic **/camera/aligned_depth_to_color/image_raw**

erbizard, 2022-10-20:
Okay i finally understand my mistake, the only rgb camera on the D435 is the fully left, the other 2 are only IR sensor and the middle one is the ir projector . I'll work around with this, thanks for the help, you can close the issue 👍

MartyG-RealSense, 2022-10-20:
You are very welcome.  :)  As you suggested, I will close the issue.  Thanks again!

closed 2022-10-20 by MartyG-RealSense