realsenseai/realsense-ros #2510: cap = cv2.VideoCapture(index) , how to make index static

Akumar201, 2022-10-20:
I am currently using real-sense d435 for detecting an object, however I noted that many times I need to change the index to find out which is for rgb or b&w , it usually changes 10 in 1 time when I run the program, I was wondering is there a way to make it fixed so I need not to find and change it every time.

MartyG-RealSense, 2022-10-20:
Hi @Akumar201   For a D435 camera the /video indexes should be the following values by default:

Depth: /video0
Infrared: /video1
RGB: /video2

In regard to setting a fixed index: whilst I have not seen this attempted with RealSense cameras, Linux users in general who have tried it with their cameras have suggested creating **symlinks** (symbolic links).  A couple of examples of such discussions are below.

https://askubuntu.com/questions/715333/assign-webcam-to-a-specific-dev-video

https://unix.stackexchange.com/questions/512759/multiple-dev-video-for-one-physical-device/581838#581838

Akumar201, 2022-10-20:
Hi do you have any clue why the index might change every time I run the program ? Initially RGB was at 8 but now at 5. I am using Ubuntu 20.04.

MartyG-RealSense, 2022-10-20:
More information about /video numbers on RealSense cameras is provided by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/3416#issuecomment-471720605

Do you have any other cameras or video devices attached to your computer in addition to your D435, such as ordinary USB webcams?

If you are able to launch the RealSense Viewer tool, does the warning message **Multiple realsense udev-rules were found!** appear in the top corner of the Viewer window when it completes launching?  The udev rules are Ubuntu's device handling rules.  

If this warning appears then it can indicate - as the name suggests - that more than one udev device handling rule is installed on your computer.  This could occur if the librealsense SDK has been installed with both the package installation and source code installation methods on the same computer.

closed 2022-10-25 by Akumar201