realsenseai/realsense-ros #2512: focal InRelease' doesn't support architecture 'armhf'

Akumar201, 2022-10-20:
Hi, I was trying to install [Reaslsense SDK][1] for realsense viewer and it didn't went through, now I want to remove it as because of it I am getting an error while updating 

     Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://librealsense.intel.com/Debian/apt-repo focal InRelease' doesn't support architecture 'armhf'

and upgrading 

    The following packages have unmet dependencies:
     dpkg : Breaks: libapt-pkg5.0 (< 1.7~b) but 1.2.35 is to be installed
    E: Broken packages

I already had downloaded the Realsense ros by using the command sudo apt-get install ros-$ROS_DISTRO-realsense2-camera
I am using Ubuntu 20.04

  [1]: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

MartyG-RealSense, 2022-10-21:
Hi @Akumar201  If you installed from packages then you can uninstall all RealSense related packages from your computer using the command below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

Akumar201, 2022-10-21:
Hi @MartyG-RealSense  I used the code , however while doing `sudo apt update` I am still getting the following error

```

Hit:1 http://ports.ubuntu.com/ubuntu-ports xenial InRelease
Hit:2 http://security.ubuntu.com/ubuntu xenial-security InRelease              
Hit:3 http://archive.ubuntu.com/ubuntu xenial InRelease                        
Hit:4 http://lenovo.archive.canonical.com focal InRelease                      
Hit:5 http://ports.ubuntu.com/ubuntu-ports xenial-updates InRelease            
Hit:6 http://archive.ubuntu.com/ubuntu xenial-updates InRelease                
Hit:7 http://packages.ros.org/ros/ubuntu focal InRelease                       
Hit:8 https://packages.microsoft.com/repos/ms-teams stable InRelease           
Hit:9 https://librealsense.intel.com/Debian/apt-repo focal InRelease           
Reading package lists... Done
Building dependency tree       
Reading state information... Done
35 packages can be upgraded. Run 'apt list --upgradable' to see them.
N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://librealsense.intel.com/Debian/apt-repo focal InRelease' doesn't support architecture 'armhf'

```

`sudo apt upgrade`

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 dpkg : Breaks: libapt-pkg5.0 (< 1.7~b) but 1.2.35 is to be installed
E: Broken packages

```

MartyG-RealSense, 2022-10-21:
Are you installing on a computer with an Arm processor, such as Nvidia Jetson?  The librealsense SDK does not support installation on Arm from packages on Ubuntu 20.04 Focal and there are no plans to provide this feature, unfortunately.  The SDK should be built for Focal from source code instead of packages on an Arm computer.

Akumar201, 2022-10-21:
Thank you for your response, can you please tell me how can I remove it such that I don't get any errors?

MartyG-RealSense, 2022-10-21:
You should be able to remove it by repeating the 'purge' instruction above and then not performing the package installation again afterwards.

Akumar201, 2022-10-21:
Hi! I am getting this error after using the above suggested command 

dpkg: error: --purge needs at least one package name argument

```
Type dpkg --help for help about installing and deinstalling packages [*];
Use 'apt' or 'aptitude' for user-friendly package management;
Type dpkg -Dhelp for a list of dpkg debug flag values;
Type dpkg --force-help for a list of forcing options;
Type dpkg-deb --help for help about manipulating *.deb files;

Options marked [*] produce a lot of output - pipe it through 'less' or 'more' !

```

MartyG-RealSense, 2022-10-21:
Possibly there are no packages with the word "realsense" in them to uninstall because your previous installation attempt had errors.  So I would recommend assuming that the packages are no longer on the computer and proceeding with a source code install of librealsense.

Akumar201, 2022-10-21:
I understand, however can you please tell me why I am facing this error ? 

```
ThinkPad-P17-Gen-2i:~$ sudo apt update
 
Hit:1 http://archive.ubuntu.com/ubuntu xenial InRelease
Get:2 http://archive.ubuntu.com/ubuntu xenial-updates InRelease [99.8 kB]                                                                                     
Hit:3 http://ports.ubuntu.com/ubuntu-ports xenial InRelease                                                                                                   
Hit:4 http://ports.ubuntu.com/ubuntu-ports xenial-updates InRelease                                                                                           
Hit:5 http://lenovo.archive.canonical.com focal InRelease                                                                                                     
Hit:6 http://packages.ros.org/ros/ubuntu focal InRelease                                                                                                      
Hit:7 https://packages.microsoft.com/repos/ms-teams stable InRelease                                                                                    
Get:8 http://security.ubuntu.com/ubuntu xenial-security InRelease [99.8 kB]                                                       
Hit:9 https://librealsense.intel.com/Debian/apt-repo focal InRelease                    
Fetched 200 kB in 1s (149 kB/s)                          
Reading package lists... Done
Building dependency tree       
Reading state information... Done
35 packages can be upgraded. Run 'apt list --upgradable' to see them.
N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://librealsense.intel.com/Debian/apt-repo focal InRelease' doesn't support architecture 'armhf'

ThinkPad-P17-Gen-2i:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 dpkg : Breaks: libapt-pkg5.0 (< 1.7~b) but 1.2.35 is to be installed
E: Broken packages

```

MartyG-RealSense, 2022-10-21:
If your computer is a **ThinkPad-P17-Gen-2i** then it is unusual that you would be getting the message **focal InRelease' doesn't support architecture 'armhf'** as this computer apparently has an Intel processor, not an Arm one.  A package installation on Focal should work on a computer with an Intel processor.

If you have used the instruction **sudo apt-get install ros-$ROS_DISTRO-realsense2-camera** then you do not need to install librealsense again as it has already been installed.  This instruction installs both librealsense and the ROS wrapper together at te same time from packages.   So if this install method is used and then an attempt to install librealsense is made afterwards, this could cause a conflict as you would be installing librealsense a second time.

Can you confirm please whether or not the RealSense Viewer can be launched if you input **realsense-viewer** into the Ubuntu terminal (not the ROS terminal).

Akumar201, 2022-10-21:
Hi, thank you so much or your prompt replies. I have used **sudo apt-get install ros-$ROS_DISTRO-realsense2-camera** and rs_camera.launch file is working perfectly, however still getting the same error while updating and upgrading, plus I  realsense-viewer command is not working stating command not found.

Akumar201, 2022-10-21:
I am getting this error now 

```
sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u

Get:1 http://security.ubuntu.com/ubuntu xenial-security InRelease [99.8 kB]
Hit:2 http://archive.ubuntu.com/ubuntu xenial InRelease                                                                                                       
Hit:3 http://ports.ubuntu.com/ubuntu-ports xenial InRelease                                                                                                   
Hit:4 http://lenovo.archive.canonical.com focal InRelease                                                                                                     
Hit:5 http://packages.ros.org/ros/ubuntu focal InRelease                                                                                                      
Get:6 http://archive.ubuntu.com/ubuntu xenial-updates InRelease [99.8 kB]                                                                                     
Hit:7 http://ports.ubuntu.com/ubuntu-ports xenial-updates InRelease                                                                                           
Hit:8 https://packages.microsoft.com/repos/ms-teams stable InRelease                                                                                          
Hit:9 https://librealsense.intel.com/Debian/apt-repo focal InRelease                                                                         
Fetched 200 kB in 1s (194 kB/s)
Reading package lists... Done
N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://librealsense.intel.com/Debian/apt-repo focal InRelease' doesn't support architecture 'armhf'

ThinkPad-P17-Gen-2i:~/catkin_ws/src/realsense-ros/realsense2_camera/launch$ sudo apt-get install librealsense2-utils

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 librealsense2-utils : Depends: libgtk-3-dev but it is not going to be installed
                       Depends: libglfw3 (>= 3.2) but 3.1.2-3 is to be installed
E: Unable to correct problems, you have held broken packages.

```

MartyG-RealSense, 2022-10-21:
Did you register the server's public key and add the server to th elist of repositories before inputting the sudo apt-get instruction, please?

Akumar201, 2022-10-21:
Yes, I did that 

```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
Executing: /tmp/apt-key-gpghome.Ff4s1whCfZ/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1

```

MartyG-RealSense, 2022-10-21:
It looks as though the 'utils' package is dependent on the 'libgtk-3-dev' and 'libglfw3' packages but they are not currently installed on your computer.

What happens if you use the command below and then repeat the sudo apt-get for the 'utils' package afterwards.

`sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev`

Akumar201, 2022-10-21:
Hi, these are the outputs 

```
sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev

Reading package lists... Done
Building dependency tree       
Reading state information... Done
libgl1-mesa-dev is already the newest version (21.2.6-0ubuntu0.1~20.04.2).
libgl1-mesa-dev set to manually installed.
libglu1-mesa-dev is already the newest version (9.0.1-1build1).
libglu1-mesa-dev set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libglfw3-dev : Depends: libxxf86vm-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

```

```
sudo apt-get install librealsense2-utils

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 librealsense2-utils : Depends: libgtk-3-dev but it is not going to be installed
                       Depends: libglfw3 (>= 3.2) but 3.1.2-3 is to be installed
E: Unable to correct problems, you have held broken packages.
akumar@akumar-ThinkPad-P17-Gen-2i:~/catkin_ws/src/realsense-ros
```

MartyG-RealSense, 2022-10-21:
Okay, now try `sudo apt install libgtk-3-dev`

Akumar201, 2022-10-21:
These are the output 
```

sudo apt install libgtk-3-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgtk-3-dev : Depends: libgtk-3-0 (= 3.18.9-1ubuntu3.3) but 3.24.20-0ubuntu1.1 is to be installed
                Depends: gir1.2-gtk-3.0 (= 3.18.9-1ubuntu3.3) but 3.24.20-0ubuntu1.1 is to be installed
                Depends: libatk-bridge2.0-dev but it is not going to be installed
                Depends: libepoxy-dev (>= 1.0) but it is not going to be installed
                Depends: libegl1-mesa-dev but it is not going to be installed
                Depends: libwayland-dev (>= 1.5.91) but it is not going to be installed
                Depends: libxkbcommon-dev but it is not going to be installed
                Depends: libmirclient-dev (>= 0.13.3) but it is not going to be installed
                Depends: nettle-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

```

MartyG-RealSense, 2022-10-21:
It looks as though a workaround to install realsense-viewer on its own is not going to work out, unfortunately.  I do apologize.  The only option if you really need the Viewer may be to remove the librealsense / wrapper installation that you already have and install librealsense and the wrapper separately (librealsense from packages, the wrapper from source code).

Akumar201, 2022-10-21:
I am fine , not using the realsense-viewer, however can you please help to just remove the errors that I am getting whendoing upgrade and update.

MartyG-RealSense, 2022-10-21:
This libapt error does not seem to be specifically related to librealsense as an internet search shows that numerous Ubuntu users in general have experienced it.  In various cases they give the same advice of using the -f flag

**sudo apt-get -f install 
sudo apt-get update 
sudo apt-get upgrade**

Akumar201, 2022-10-21:
I think I found out the solution 
1. I opened my sources.list using the command using `sudo gedit /etc/apt/sources.list`
```

deb [arch=amd64] http://archive.ubuntu.com/ubuntu/ xenial main multiverse restricted universe
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports/ xenial main multiverse restricted universe
deb [arch=armhf] http://ports.ubuntu.com/ubuntu-ports/ xenial-updates main multiverse restricted universe
deb [arch=amd64] http://archive.ubuntu.com/ubuntu/ xenial-updates main multiverse restricted universe
deb [arch=amd64] http://security.ubuntu.com/ubuntu/ xenial-security main multiverse restricted universe
deb https://librealsense.intel.com/Debian/apt-repo focal main
# deb-src https://librealsense.intel.com/Debian/apt-repo focal main

then removed the last two lines from my sources file 
sudo apt update 
sudo apt upgrade
```

MartyG-RealSense, 2022-10-21:
Great, thanks very much for sharing your solution!

Akumar201, 2022-10-21:
However I am still getting error while using sudo apt upgrade 

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 dpkg : Breaks: libapt-pkg5.0 (< 1.7~b) but 1.2.35 is to be installed
E: Broken packages
```
Any suggestion ?

MartyG-RealSense, 2022-10-21:
Another suggestion people have made in relation to this error is to uninstall Ubuntu Advantage.

`sudo apt remove ubuntu-advantage-tools`

Akumar201, 2022-10-21:
Thank you @MartyG-RealSense for your help. I would like to make a comment to help others in future. 
I resolved this issue by using following steps 
Step 1. Go to sources.list using `sudo gedit /etc/apt/sources.list` , remove the listed repository related to realsense librealsense.
Step 2. sudo apt remove ubuntu-advantage-tools
Step 3. sudo apt update
Step 4. sudo apt upgrade

closed 2022-10-21 by Akumar201