realsenseai/realsense-ros #2518: Delayed Timestamps D435

Kubasinska, 2022-10-25:
I have a configuration with two intel d435 , which I launch with the following command: 
```
roslaunch realsense2_camera rs_camera.launch serial_no:=146322076069 initial_reset:=true filters:=pointcloud camera:=camera1 align_depth:=true device_type:=d435 ordered_pc:=true color_width:=848 color_height:=480 color_fps:=30 depth_width:=848 depth_height:=480 depth_fps:=30 infra_width:=848 infra_height:=480 infra_fps:=30 &>/logs/camera1.log &
roslaunch realsense2_camera rs_camera.launch serial_no:=146322075900 initial_reset:=true filters:=pointcloud camera:=camera2 align_depth:=true device_type:=d435 ordered_pc:=true color_width:=848 color_height:=480 color_fps:=30 depth_width:=848 depth_height:=480 depth_fps:=30 infra_width:=848 infra_height:=480 infra_fps:=30 &>/logs/camera2.log &
```

This is the summary of a camera:

```
SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: -1
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: True
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -2.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 848
 * /camera1/realsense2_camera/confidence_fps: -1
 * /camera1/realsense2_camera/confidence_height: -1
 * /camera1/realsense2_camera/confidence_width: -1
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 848
 * /camera1/realsense2_camera/device_type: d435
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_confidence: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_infra: False
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_pose: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters: pointcloud
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: -1
 * /camera1/realsense2_camera/fisheye_frame_id: camera1_fisheye_f...
 * /camera1/realsense2_camera/fisheye_height: -1
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera1_fisheye_o...
 * /camera1/realsense2_camera/fisheye_width: -1
 * /camera1/realsense2_camera/gyro_fps: -1
 * /camera1/realsense2_camera/gyro_frame_id: camera1_gyro_frame
 * /camera1/realsense2_camera/gyro_optical_frame_id: camera1_gyro_opti...
 * /camera1/realsense2_camera/imu_optical_frame_id: camera1_imu_optic...
 * /camera1/realsense2_camera/infra1_frame_id: camera1_infra1_frame
 * /camera1/realsense2_camera/infra1_optical_frame_id: camera1_infra1_op...
 * /camera1/realsense2_camera/infra2_frame_id: camera1_infra2_frame
 * /camera1/realsense2_camera/infra2_optical_frame_id: camera1_infra2_op...
 * /camera1/realsense2_camera/infra_fps: 30
 * /camera1/realsense2_camera/infra_height: 480
 * /camera1/realsense2_camera/infra_rgb: False
 * /camera1/realsense2_camera/infra_width: 848
 * /camera1/realsense2_camera/initial_reset: True
 * /camera1/realsense2_camera/json_file_path: 
 * /camera1/realsense2_camera/linear_accel_cov: 0.01
 * /camera1/realsense2_camera/odom_frame_id: camera1_odom_frame
 * /camera1/realsense2_camera/ordered_pc: True
 * /camera1/realsense2_camera/pointcloud_texture_index: 0
 * /camera1/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera1/realsense2_camera/pose_frame_id: camera1_pose_frame
 * /camera1/realsense2_camera/pose_optical_frame_id: camera1_pose_opti...
 * /camera1/realsense2_camera/publish_odom_tf: True
 * /camera1/realsense2_camera/publish_tf: True
 * /camera1/realsense2_camera/reconnect_timeout: 6.0
 * /camera1/realsense2_camera/rosbag_filename: 
 * /camera1/realsense2_camera/serial_no: 146322076069
 * /camera1/realsense2_camera/stereo_module/exposure/1: 7500
 * /camera1/realsense2_camera/stereo_module/exposure/2: 1
 * /camera1/realsense2_camera/stereo_module/gain/1: 16
 * /camera1/realsense2_camera/stereo_module/gain/2: 16
 * /camera1/realsense2_camera/tf_publish_rate: 0.0
 * /camera1/realsense2_camera/topic_odom_in: odom_in
 * /camera1/realsense2_camera/unite_imu_method: 
 * /camera1/realsense2_camera/usb_port_id: 
 * /camera1/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /camera1/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)
```

I am experiencing a problem with moveit (described here: https://github.com/ros-planning/moveit/issues/3242). I suspect that the problem is caused by delayed camera timestamps. 
By running `rostopic echo -n 1 --offset /camera2/depth/color/points/header `
I have:
```
seq: 119072
stamp: 
  secs: -1
  nsecs: 943370104
frame_id: "camera2_color_optical_frame"
```

This is what I have running `rostopic delay /camera2/depth/color/points/header `
```
subscribed to [/camera2/depth/color/points]
average delay: 0.052
	min: 0.050s max: 0.057s std dev: 0.00162s window: 30
average delay: 0.052
	min: 0.050s max: 0.057s std dev: 0.00126s window: 60
average delay: 0.052
	min: 0.050s max: 0.057s std dev: 0.00132s window: 90
average delay: 0.052
	min: 0.050s max: 0.057s std dev: 0.00126s window: 108
```
How can I align the camera time stamps? Is this a known problem? Or is it completely normal?

I have already tried launching the camera with this launch file where I configure global_time_enabled both false and true do not solve the problem.
```
<launch>
    <include file="/opt/ros/noetic/share/realsense2_camera/launch/rs_camera.launch">
        <arg name="camera" value="camera1"/>
        <arg name="serial_no" value="146322076069"/>
        <arg name="filters" value="spatial,pointcloud"/>
        <arg name="initial_reset" value="true"/>
        <arg name="align_depth" value="true"/>
    </include>
    <rosparam>
        /camera1/realsense2_camera/global_time_enabled: false
        /camera1/realsense2_camera/l500_depth_sensor/global_time_enabled: false
        /camera1/realsense2_camera/rgb_camera/global_time_enabled: false
    </rosparam>
</launch>
```
I also updated the firmware of the cameras but did not solve the problem.

MartyG-RealSense, 2022-10-25:
Hi @Kubasinska  The multiple camera section of the documentation for the RealSense ROS wrapper recommends performing rs_camera.launch in two separate ROS terminals if you are not doing so already.

https://github.com/IntelRealSense/realsense-ros#work-with-multiple-cameras

Do you experience improved results if launching in two separate ROS terminals, please?

*************

The infrared topics are disabled by default in rs_camera.launch, so they should be enabled in the launch instruction by adding the commands below to your launch instruction:

**enable_infra1:=true enable_infra2:=true**

As you are enabling the pointcloud filter, **enable_sync** will automatically be enabled.  This gathers the closest frames of different sensors (infrared, color and depth) to be sent with the same timetag.

Kubasinska, 2022-10-26:
Thanks for the support @MartyG-RealSense !
I tried starting the cameras in two separate terminals; it did not affect the timestamp delay.
As you suggested, I also launched the camera with the parameters `enable_infra1:=true enable_infra2:=true`. No effect on timestamps delay. This is the command I used:

```
roslaunch realsense2_camera rs_camera.launch serial_no:=146322076069 enable_infra1:=true enable_infra2:=true initial_reset:=true filters:=pointcloud camera:=camera1 align_depth:=true device_type:=d435 ordered_pc:=true color_width:=848 color_height:=480 color_fps:=30 depth_width:=848 depth_height:=480 depth_fps:=30 infra_width:=848 infra_height:=480 infra_fps:=30 enable_sync:=true
```
Also, The problem occurs even when I launch a single camera.
Furthermore, when I launched the camera, I noticed one thing: the timestamp delay in the very first messages was minor.  After about 100 messages, it stabilizes (a couple of wall time seconds).

Command: `rostopic echo --offset /camera1/depth/color/points/header/stamp`
Output:
```
---

secs: 0
nsecs:   8475781
---

secs: 0
nsecs:   9465456
---

secs: 0
nsecs:   9071827
---

secs: 0
nsecs:   8201600
---

secs: 0
nsecs:   3754377
---

secs: 0
nsecs:   5653620
---

secs: -1
nsecs: 992389203
---

secs: -1
nsecs: 999967099
---

secs: 0
nsecs:    919104
---

secs: -1
nsecs: 988869190
---

secs: -1
nsecs: 996113062
---

secs: -1
nsecs: 988427401
```
I have 3 RealSense D435, they all exhibit the same problem, so it is not a hardware issue. What could this problem be attributed to?

MartyG-RealSense, 2022-10-26:
I wonder whether the process of logging with **&>/logs/camera1.log &** and **&>/logs/camera2.log &** could be adding some processing overhead to the launch.

align_depth will add further processing demand as depth-color alignment is a processing-intensive operation.

Does the delay still occur if you use a more basic launch instruction without the pointcloud?

```
roslaunch realsense2_camera rs_camera.launch serial_no:=146322076069 camera:=camera1 color_width:=848 color_height:=480 color_fps:=30 depth_width:=848 depth_height:=480 depth_fps:=30 infra_width:=848 infra_height:=480 infra_fps:=30

roslaunch realsense2_camera rs_camera.launch serial_no:=146322075900 camera:=camera2 color_width:=848 color_height:=480 color_fps:=30 depth_width:=848 depth_height:=480 depth_fps:=30 infra_width:=848 infra_height:=480 infra_fps:=30
```

Kubasinska, 2022-10-26:
The delay persist by removing the logging (`&>/logs/camera1.log &`) and also by running `roslaunch realsense2_camera rs_camera.launch serial_no:=146322076069 camera:=camera1 color_width:=848 color_height:=480 color_fps:=30 depth_width:=848 depth_height:=480 depth_fps:=30 infra_width:=848 infra_height:=480 infra_fps:=30`

MartyG-RealSense, 2022-10-26:
There are only a few previous reported uses in this forum of **rostopic delay** to compare your results to, unfortunately.

In https://github.com/IntelRealSense/realsense-ros/issues/1665 the timestamp delay of that case's RealSense ROS user was solved by using a combination of **global_time_enabled:=false** and **initial_reset:=true**

Users in https://github.com/IntelRealSense/realsense-ros/issues/431 thought that it was related to the on-device clock.

A user in https://github.com/IntelRealSense/realsense-ros/issues/1678 who tested with rostopic delay found that their particular timestamp solution was to manually set the message header to current system time.

closed 2022-11-09 by MartyG-RealSense