realsenseai/realsense-ros #2520: Unable to change resolution in ros2-beta

ripdk12, 2022-10-27:
Running:
```
ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_infra1:=true enable_infra2:=true
```
works as intended.
However i want to use a lower resolution like 640x360 @ 90fps, which i have done successfully on the old ros1 wrapper (2.3.2). But when i run:
```
ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_infra1:=true -p enable_infra2:=true -p infra1_height:=360 -p infra1_width:=640 -p infra2_height:=360 -p infra2_width:=640 -p infra1_fps:=90.0 -p infra2_fps:=90.0
```
The resolution is still the standard 480x848 @ 30 fps.
I also tried through the provided launch file, but the parameters don't seem to be configurable? Was this changed? or am i missing something?

MartyG-RealSense, 2022-10-27:
Hi @ripdk12  The **ros2_beta** wrapper uses a 'profile' system with **ros2 launch** for defining custom resolutions instead of defining width and height individually.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x90`

This publishes a stream configuration that is equal to the one below under the old system.

`ros2 launch realsense2_camera rs_launch.py depth_width:=640 depth_height:=480 depth_fps:=30.0 infra1_width:=640 infra1_height:=480 infra1_fps:=30.0`

ripdk12, 2022-10-27:
@MartyG-RealSense You're the man! Just the information i needed!

closed 2022-10-27 by ripdk12

MartyG-RealSense, 2022-10-27:
You are very welcome!  I'm pleased that i could help  :)