realsenseai/realsense-ros #2521: Is there any way to get the gravity direction using the realsense-ros wrapper?

martinakos, 2022-10-27:
Ideally the gravity direction wrt. camera frame, published in some topic.

If this is not possible with realsense-ros, is it possible to get this direction vector using librealsense sdk?

martinakos, 2022-10-27:
I'm using unite_imu_method at the moment as I need the /imu topic for a VIO.  Is there any way to get the /camera/accel/sample and /imu topics simultaneously?

MartyG-RealSense, 2022-10-27:
My understanding is that if you are publishing /imu then then the accel values can be obtained from that topic as **linear_acceleration**, as shown at https://github.com/IntelRealSense/realsense-ros/issues/1286#issue-656550753

MartyG-RealSense, 2022-11-01:
Hi @martinakos  Do you require further assistance with this case, please?  Thanks!

martinakos, 2022-11-01:
Sorry, I was away these last days. I'll check what you suggest and come back with the results.

MartyG-RealSense, 2022-11-01:
In Intel's SLAM navgation guide for ROS and D435i at the link below, imu_filter_madgwick is used in combination with the RealSense ROS wrapper's ROS1 launch file **opensource_tracking.launch**

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

closed 2022-11-07 by martinakos

MartyG-RealSense, 2022-11-07:
It's excellent news that you achieved a solution.  Thanks very much for the update!