realsenseai/realsense-ros #2522: [ros2-beta] Missing TFs in "/tf" topic when tf_publish_rate > 0

Tristis116, 2022-10-28:
When parameter `tf_publish_rate`  > 0, the node [publishes](https://github.com/IntelRealSense/realsense-ros/blob/ros2-beta/realsense2_camera/src/base_realsense_node.cpp#L979) all TFs from`_static_tf_msgs`.
This [line](https://github.com/IntelRealSense/realsense-ros/blob/ros2-beta/realsense2_camera/src/rs_node_setup.cpp#L322) clears static transforms saved in the RS node before adding transforms for each sensor. After all sensors initialization, `_static_tf_msgs` has only TFs for the last sensor. 
So, dynamic transform publisher doesn't publish all TFs like static transform publisher does, but only TFs of the last initialized sensor.

It looks like a mistake, because `_static_tf_msgs`, as its name suggests, should have all static TFs inside it for the dynamic transform publisher.

Tristis116, 2022-10-28:
Hi @MartyG-RealSense
I read about limitations. My point is: `/tf` doesn't have all transformations - only for a sensor initialized the last. And the documentation doesn't say anything about which transformations should be published and which shouldn't.
According to the code, it looks like it should duplicate `/tf_static`. But sensor initialization clears the list of transformations when it initializes every sensor.

MartyG-RealSense, 2022-10-28:
I will highlight your query to the RealSense ROS team.  Thanks very much for your patience!

MartyG-RealSense, 2022-11-10:
Hi @Tristis116  Just an update to let you know that I have been discussing your case with the RealSense ROS team.  At this time it is not known why you are experiencing the reported issue with /tf, unfortunately.

Tristis116, 2022-11-10:
Hi @MartyG-RealSense 
When I start d455 camera with `ros2 launch realsense2_camera rs_launch.py -s`, then /tf_static has transforms for color and depth:
```
transforms:
- header:
    stamp:
      sec: 1668070582
      nanosec: 112172855
    frame_id: camera_link
  child_frame_id: camera_depth_frame
  transform:
    translation:
      x: 0.0
      y: -0.0
      z: -0.0
    rotation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
- header:
    stamp:
      sec: 1668070582
      nanosec: 112172855
    frame_id: camera_depth_frame
  child_frame_id: camera_depth_optical_frame
  transform:
    translation:
      x: 0.0
      y: -0.0
      z: -0.0
    rotation:
      x: -0.5
      y: 0.4999999999999999
      z: -0.5
      w: 0.5000000000000001
- header:
    stamp:
      sec: 1668070582
      nanosec: 314827794
    frame_id: camera_link
  child_frame_id: camera_color_frame
  transform:
    translation:
      x: 0.00043846675544045866
      y: 0.059190280735492706
      z: 0.0001858784380601719
    rotation:
      x: 0.00023960287217050793
      y: 0.00014341560017783195
      z: -0.0007031101267784834
      w: 0.9999997019767761
- header:
    stamp:
      sec: 1668070582
      nanosec: 314827794
    frame_id: camera_color_frame
  child_frame_id: camera_color_optical_frame
  transform:
    translation:
      x: 0.0
      y: -0.0
      z: -0.0
    rotation:
      x: -0.5
      y: 0.4999999999999999
      z: -0.5
      w: 0.5000000000000001
---
```
/tf has only transformation for color:
```
transforms:
- header:
    stamp:
      sec: 1668070614
      nanosec: 989107270
    frame_id: camera_link
  child_frame_id: camera_color_frame
  transform:
    translation:
      x: 0.00043846675544045866
      y: 0.059190280735492706
      z: 0.0001858784380601719
    rotation:
      x: 0.00023960287217050793
      y: 0.00014341560017783195
      z: -0.0007031101267784834
      w: 0.9999997019767761
- header:
    stamp:
      sec: 1668070614
      nanosec: 989107270
    frame_id: camera_color_frame
  child_frame_id: camera_color_optical_frame
  transform:
    translation:
      x: 0.0
      y: -0.0
      z: -0.0
    rotation:
      x: -0.5
      y: 0.4999999999999999
      z: -0.5
      w: 0.5000000000000001
---
```
This behavior doesn't correspond to the `tf_publish_rate` description:
```
    'tf_publish_rate':
        Rate of publishing static_tf
        (default: '0.0') 
```
If I disable color stream, then /tf has only transforms for depth. /tf has only the last transform of /tf_static. And the last transform of /tf_static depends on enabled streams and output of [rs2::device::query_sensors](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1device.html#ac8cab78a80e1ff20a3886fe97c8a5330).
So there is an actual problem for users: ros tf functionality and rviz visualization may not work because of missing transforms. And a potential problem after disabling/enabling specific streams, switching to another camera model, or even firmware update, because published transform may change and break code that relied on previously published transform.

MartyG-RealSense, 2022-11-10:
I have passed your latest information above to the RealSense ROS team.  Thanks very much for the update.

MartyG-RealSense, 2022-11-21:
Hi @Tristis116  The issue is still under discussion with the RealSense ROS team, so I have added an **Enhancement** label to signify that this case should be kept open.  Thanks very much for your patience!

SamerKhshiboun, 2023-06-01:
@MartyG-RealSense, PR was merged.
I think we can close this ticket if @Tristis116 confirms that.

MartyG-RealSense, 2023-06-01:
Thanks so much, @SamerKhshiboun  :)