realsenseai/realsense-ros #2523: Release for Rolling?

ijnek, 2022-10-28:
Thanks for this library! I see the package is available for ROS 2 Humble, may I request for a release to be made in ROS 2 Rolling too? Are there some roadblocks?

MartyG-RealSense, 2022-10-28:
Hi @ijnek  The **ros2_beta** wrapper has support for Rolling.  It should be possible to build the ros2_beta wrapper for Rolling from source code.

[image]

Are you installing from source using the ros2_beta instructions at the link below, please?

https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta#step-3-install-intel-realsense-ros2-wrapper-from-sources

ijnek, 2022-10-28:
Great to hear that ROS 2 Rolling is supported by source installation. I am working on trying to release a package that has an <exec_depend> on realsense-ros for ROS 2 rolling. We'll need realsense-ros released and available as a binary in Rolling to proceed.

Considering developing on ROS 2 Rolling is now common practice, it would be great if you could consider it!

MartyG-RealSense, 2022-10-28:
I will query to the RealSense ROS team about the absence of a distribution for Rolling.

https://index.ros.org/r/realsense2_camera/#rolling

MartyG-RealSense, 2022-10-31:
Hi @ijnek  The RealSense ROS team has responded about your question regarding the absent Rolling distribution.  Rolling is not supported with a RealSense ROS distribution because it is a **rolling development** release of ROS2, meaning that it is continually being updated and is not stable.

[image]

For this reason, if binaries for Rolling were released by the RealSense ROS team then they could become broken the next day by changes made to Rolling.  The stable versions of ROS2 are therefore prioritized.

ijnek, 2022-11-01:
Thanks for taking the time to consider this in your team, I really appreciate it.

The screenshot of the warning you posted talks about two categories of people:

1. Users - Don't plan on releasing packages, and can use a stable distribution of realsense2_camera
1. Maintainers - Plan on releasing packages, and should use Rolling to catch issues early and ensure stability for next stable release.

Anyone that is trying to release a package that depends on realsense2_camera would fall into the "maintainer" category and not the "user" category.

To release a package into any distribution (including Rolling), all dependencies must already be released and available, and the inavailability of realsense2_camera inhibits the release of any packages that depend on it into Rolling.

> The stable versions of ROS2 are therefore prioritized.

Rolling releases are intended for use by other package maintainers, and stable releases are intended for use by end users, so I don't think this should be about prioritizing a distro over another.

I would appreciate if you could reconsider releasing into Rolling, as without the release, we cannot release our example robots into ROS 2 Rolling!

MartyG-RealSense, 2022-11-01:
The RealSense ROS developer team have responded that binaries for ROS2 Rolling will not be supported, unfortunately.  This is to focus resources on stable releases instead of having to track Rolling's stability on a daily basis.  I sincerely apologize for the inconvenience to your project.

Nir-Az, 2022-11-02:
Hi @ijnek ,
I'm Nir from the development team of the ROS wrapper for librealsense2.
We appreciate your approach and request.
Currently the team focus is on stabilizing the Humble release with the new refactored code of `ros2-development`  (previously `ros2-beta`) branch.
We try to support `Rolling` distro as we can from building from source.
With an eye on the next ROS2 distro release.

The major issue with releasing to ROS2 `Rolling` is that if it breaks, we cannot handle `Rolling` releases outside of our full product releases which is `librealsense2` SDK.
We will consider adding it in the future when we have more bandwidth.

ijnek, 2022-11-03:
Thanks for the detailed explanation @MartyG-RealSense and @Nir-Az. In regards to Intel's release cycles, that is something I may not be too familiar with, but in regards to bandwidth, I understand the extra commitment required to maintain a package in Rolling.

Some time in the future (or once things are more stable with Humble), I wish your team would revisit Rolling releases. But for now, I am thankful and appreciate what your team has provided so far for the community :+1:

MartyG-RealSense, 2022-11-03:
Thanks so much @ijnek for the update and for your patient understanding!