realsenseai/realsense-ros #2528: Error starting device: No default profile found

husseinalijaafar, 2022-10-31:
Hello, 
I am trying to use an d435i with a raspberry pi and ros2. 

I have done the following
Add the 99... rules file
```
cd ~/
git clone https://github.com/IntelRealSense/librealsense.git 
cd librealsense/config
sudo cp ./99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger
reboot
```

Install realsense-ros and realsense sdk using the prebuilt binaries
```
sudo apt update
sudo apt install ros-foxy-realsense2-camera ros-foxy-realsense2-description
```

But when I try to run the default bringup script
```
source /opt/ros/foxy/setup.bash
ros2 launch realsense2_camera rs_launch.py 
```

I get the following error 
```
ubuntu@ubuntu:~$ ros2 launch realsense2_camera rs_launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2022-10-31-21-26-19-823102-ubuntu-7243
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [7245]
[realsense2_camera_node-1] [INFO] [1667251581.110485640] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1667251581.110958448] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1667251581.111092613] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1]  31/10 21:26:21,175 WARNING [281473332521376] (ds5-device.cpp:687) The IMU sensor is undefined for PID ad6 and imu_chip_id: �
[realsense2_camera_node-1] [INFO] [1667251581.178200258] [camera.camera]: Device with serial number 036222070794 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1667251581.178465776] [camera.camera]: Device with physical ID /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1667251581.178582650] [camera.camera]: Device with name Intel RealSense USB2 was found.
[realsense2_camera_node-1] [INFO] [1667251581.180806744] [camera.camera]: Device with port number 1-1.3 was found.
[realsense2_camera_node-1] [INFO] [1667251581.181047002] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1667251581.181211427] [camera.camera]: Device 036222070794 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1]  31/10 21:26:21,178 WARNING [281473332521376] (ds5-active.cpp:78) Projector capacity is overrided and disabled by FW
[realsense2_camera_node-1] Device PID = 0xad6, Capabilities Vector = [Active Projector/RGB Sensor/IMU Sensor/Global Shutter/]
[realsense2_camera_node-1] [INFO] [1667251581.191211489] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1667251581.196243993] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1667251581.196496335] [camera.camera]: Device Name: Intel RealSense USB2
[realsense2_camera_node-1] [INFO] [1667251581.196674041] [camera.camera]: Device Serial No: 036222070794
[realsense2_camera_node-1] [INFO] [1667251581.196836696] [camera.camera]: Device physical port: /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1667251581.197090444] [camera.camera]: Device FW version: 05.11.01.100
[realsense2_camera_node-1] [INFO] [1667251581.197198099] [camera.camera]: Device Product ID: 0x0AD6
[realsense2_camera_node-1] [INFO] [1667251581.197269088] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [ERROR] [1667251581.223361496] [camera.camera]: Error starting device: No default profile found
```

I have also tried
```
ros2 launch realsense2_camera rs_launch.py device_type:=d435
```
```
ubuntu@ubuntu:~$ ros2 launch realsense2_camera rs_launch.py device_type:=d435
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2022-10-31-21-27-12-376800-ubuntu-7278
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [7280]
[realsense2_camera_node-1] [INFO] [1667251633.658867322] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1667251633.659331693] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1667251633.659506014] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1667251633.725793354] [camera.camera]: Device with serial number 036222070794 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1667251633.726021894] [camera.camera]: Device with physical ID /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1667251633.726121112] [camera.camera]: Device with name Intel RealSense USB2 was found.
[realsense2_camera_node-1]  31/10 21:27:13,723 WARNING [281473393613216] (ds5-device.cpp:687) The IMU sensor is undefined for PID ad6 and imu_chip_id: �
[realsense2_camera_node-1] [INFO] [1667251633.728730722] [camera.camera]: Device with port number 1-1.3 was found.
[realsense2_camera_node-1] [ERROR] [1667251633.729037959] [camera.camera]: The requested device with device name containing d435 is NOT found. Will Try again.
[realsense2_camera_node-1]  31/10 21:27:13,725 WARNING [281473393613216] (ds5-active.cpp:78) Projector capacity is overrided and disabled by FW
[realsense2_camera_node-1] Device PID = 0xad6, Capabilities Vector = [Active Projector/RGB Sensor/IMU Sensor/Global Shutter/]
...
```

However, when I run `dmesg | less` I can see the device registered
```
[    4.393645] usb 1-1.3: New USB device found, idVendor=8086, idProduct=0ad6, bcdDevice=50.b1
[    4.393657] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    4.393664] usb 1-1.3: Product: Intel(R) RealSense(TM) Depth Camera 435i
[    4.393671] usb 1-1.3: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
....
[ 2760.008442] usb 1-1.3: USB disconnect, device number 10
[ 2760.743063] usb 1-1.3: new high-speed USB device number 11 using dwc_otg
[ 2760.844188] usb 1-1.3: New USB device found, idVendor=8086, idProduct=0ad6, bcdDevice=50.b1
[ 2760.844208] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2760.844222] usb 1-1.3: Product: Intel(R) RealSense(TM) Depth Camera 435i
[ 2760.844234] usb 1-1.3: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[ 2760.845677] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 2760.846073] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0ad6)
[ 2760.849413] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/input/input6
[ 2788.307350] usb 1-1.1.2: new full-speed USB device number 12 using dwc_otg
[ 2788.415161] usb 1-1.1.2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 2788.415178] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2788.415191] usb 1-1.1.2: Product: CP2102 USB to UART Bridge Controller
[ 2788.415204] usb 1-1.1.2: Manufacturer: Silicon Labs
[ 2788.415217] usb 1-1.1.2: SerialNumber: 0001
[ 2788.418501] cp210x 1-1.1.2:1.0: cp210x converter detected
[ 2788.423167] usb 1-1.1.2: cp210x converter now attached to ttyUSB0
```

I am not sure what the problem is, any help is appreciated.
Thanks

MartyG-RealSense, 2022-10-31:
Hi @husseinalijaafar  The log lists your D435i camera's firmware driver version as **05.11.01.100**.  This firmware is very old, dating back to February 2019 and librealsense version 2.17.0.  

I strongly recommend performing an update of your firmware driver, which is stored inside the camera hardware and not on the computer.  The firmware recommended for use with librealsense 2.51.1 is **5.13.0.50** which is the latest firmware at the time of writing this.

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

You can update the firmware in the **RealSense Viewer** tool with the following procedure.

1.  Download to your computer the firmware .bin file  for firmware 5.13.0.50 from the firmware releases page at the link above.

2.  Launch the RealSense Viewer tool - also known as realsense-viewer - and left-click on the **More** option near the top of the Viewer's options side-panel to open its drop-down menu.

3.  Select **Update Firmware** from the menu and navigate to the location on your computer that the .bin file was downloaded to, then select the file.

The camera's firmware driver should then automatically begin updating.

husseinalijaafar, 2022-11-01:
Hi Marty,

Thank you, I did not realize the firmware was that old.
After a firmware update it works, however, I am getting very few frames passed through the `/camera/color/image_raw` topic.

I notice that the following is filling the terminal
`[4767,698374] uvcvideo:  Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp.1024)`

Not sure what this means, any insight is appreciated.

husseinalijaafar, 2022-11-01:
Similarly, 
```
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStdout'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0xffff8905ebe0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0xffff8905ebe0>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0xffff87a1fc70>'
[realsense2_camera_node-1]  01/11 16:26:55,181 WARNING [281472619501984] (ds5-timestamp.cpp:69) UVC metadata payloads not available. Please refer to the installation chapter for details.
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStdout'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0xffff8905eca0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0xffff8905eca0>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0xffff87a1fc70>'
[realsense2_camera_node-1]  01/11 16:26:55,493 ERROR [281472619501984] (backend-v4l2.cpp:417) Non-sequential Video and Metadata v4l buffers
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStdout'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0xffff87a1f640>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0xffff87a1f640>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0xffff87a1fc70>'
[realsense2_camera_node-1]  01/11 16:26:55,493 WARNING [281472619501984] (backend-v4l2.cpp:1165) Video frame dropped, video and metadata buffers inconsistency
```

MartyG-RealSense, 2022-11-01:
Regarding the message **uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32**, there was a past case where I consulted with my Intel RealSense colleagues about it.  They provided the response below.

***********

The message “uvcvideo: Failed to query (GET_CUR) UVC control” indicates a failure on (kernel) driver level to read the requested parameter, for instance if the device was busy with internal operation or that the incoming requests queue is full.  In most cases this message can be disregarded as the SDK has built-in auto-retry mechanism to compensate for the temporalities.

**********

I note that you installed the ROS2 wrapper using the apt-get install instruction so that librealsense and the ROS2 wrapper are installed from packages at the same time. 

The debug log message **UVC metadata payloads not available. Please refer to the installation chapter for details** suggests that there may be a problem with kernel patching.  The ROS wrapper's packages are based on the **RSUSB backend** installation method though, so it should not have such kernel errors as an RSUSB build of librealsense bypasses the kernel and so is not dependent on Linux versions or kernel versions.

Does performance improve if you reset the camera at launch by adding **initial_reset:=true** to the launch instruction?

**ros2 launch realsense2_camera rs_launch.py initial_reset:=true**

husseinalijaafar, 2022-11-01:
Hi Marty,
Thank you for the information.

I tried the same setup on a different raspberry pi, with a different d435i, and I am still getting the same result. Regardless of whether I include the reset option, or with ros2 run. 

Here is the full output.
```
ubuntu@ubuntu:~$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1667327088.703027393] [camera]: RealSense ROS v4.51.1
[INFO] [1667327088.703603284] [camera]: Built with LibRealSense v2.51.1
[INFO] [1667327088.703737087] [camera]: Running with LibRealSense v2.51.1
[INFO] [1667327088.800175926] [camera]: Device with serial number 036222070973 was found.

[INFO] [1667327088.800406293] [camera]: Device with physical ID /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/video4linux/video0 was found.
[INFO] [1667327088.800488272] [camera]: Device with name Intel RealSense D435I was found.
[INFO] [1667327088.802451518] [camera]: Device with port number 1-1.2 was found.
[INFO] [1667327088.802577301] [camera]: Device USB type: 2.1
[WARN] [1667327088.802654124] [camera]: Device 036222070973 is connected using a 2.1 port. Reduced performance is expected.
[INFO] [1667327088.810444711] [camera]: getParameters...
[INFO] [1667327088.811699878] [camera]: JSON file is not provided
[INFO] [1667327088.811882484] [camera]: Device Name: Intel RealSense D435I
[INFO] [1667327088.812006756] [camera]: Device Serial No: 036222070973
[INFO] [1667327088.812080715] [camera]: Device physical port: /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/video4linux/video0
[INFO] [1667327088.812151028] [camera]: Device FW version: 05.13.00.50
[INFO] [1667327088.812215143] [camera]: Device Product ID: 0x0B3A
[INFO] [1667327088.812287643] [camera]: Sync Mode: Off
[WARN] [1667327088.861217225] [camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[INFO] [1667327088.909778679] [camera]: Stopping Sensor: Stereo Module
[INFO] [1667327088.939820705] [camera]: Starting Sensor: Stereo Module
[INFO] [1667327089.020468316] [camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 640, Height: 480, FPS: 15
[INFO] [1667327089.020849934] [camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 640, Height: 480, FPS: 15
[INFO] [1667327089.021045353] [camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[INFO] [1667327089.022496927] [camera]: Stopping Sensor: RGB Camera
[INFO] [1667327089.031814767] [camera]: Starting Sensor: RGB Camera
[INFO] [1667327089.043589971] [camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 15
[INFO] [1667327089.043927110] [camera]: Stopping Sensor: Motion Module
 01/11 18:24:49,048 WARNING [281472426568096] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[INFO] [1667327089.058513900] [camera]: Starting Sensor: Motion Module
[INFO] [1667327089.075622067] [camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 63
[INFO] [1667327089.075848475] [camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[INFO] [1667327089.080886955] [camera]: RealSense Node Is Up!
[WARN] [1667327089.171804187] [camera]: 
 01/11 18:24:49,244 WARNING [281472065845664] (sensor.cpp:1058) HID timestamp not found, switching to Host timestamps.
^C[INFO] [1667327151.817572210] [rclcpp]: signal_handler(signal_value=2)
[INFO] [1667327151.872239137] [camera]: Stop Sensor: Stereo Module
[INFO] [1667327151.872387211] [camera]: Close Sensor. 
[INFO] [1667327151.886491825] [camera]: Close Sensor - Done. 
[INFO] [1667327151.886670473] [camera]: Stop Sensor: RGB Camera
[INFO] [1667327151.886742661] [camera]: Close Sensor. 
[INFO] [1667327151.891974846] [camera]: Close Sensor - Done. 
[INFO] [1667327151.892108441] [camera]: Stop Sensor: Motion Module
[INFO] [1667327151.893830176] [camera]: Close Sensor. 
[INFO] [1667327152.577846030] [camera]: Close Sensor - Done. 
```

Using Foxglove visualiser, I can see that the frame rate of the camera is only 2-3 Hz. 
Any ide awhy this is?

MartyG-RealSense, 2022-11-01:
Which Raspberry Pi board model are you using, please?  Your USB connection is being detected as USB 2.1, which has slower data transfer  speed than USB 3.  

A Pi may have a USB 2.1 connection if the board model is not a Pi 4, as models released before Pi 4 (such as Pi 3) all had a USB 2 port.  Pi 4 introduced the faster USB 3 port to the Raspberry Pi range.

It looks as though multiple stream types (depth, color, RGB, motion module) are being published by the launch, which may be over-burdening the Pi.

How does it perform if you purposely disable some stream types.

**ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_color:=false -p enable_gyro:=false -p enable_accel:=false -p enable_infra1:=false**

husseinalijaafar, 2022-11-01:
Hi Marty,

That is definitely a possibility; we are using Pi 3 (no usb 3).

You're recommendation of disabling certain streams was helpful; Is it possible to disable all streams except rgb and depth? I am not too familiar with the different streams. 

When I try this, the depth is publishing messages but not the raw image
```
ros2 run realsense2_camera realsense2_camera_node rgb_camera.profile:=640x480x15 --ros-args -p enable_color:=true -p enable_gyro:=false -p enable_accel:=false -p enable_infra1:=false -p enable_infra2:=false 
```

MartyG-RealSense, 2022-11-01:
Let's try adding the depth profile so that depth is published, and disabling infra1 and infra2 (the depth profile publishes depth, infra1 and infra2).

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x15 rgb_camera.profile:=640x480x15 enable_depth:=true enable_color:=true enable_gyro:=false enable_accel:=false enable_infra1:=false enable_infra2:=false`

husseinalijaafar, 2022-11-01:
That seems to help. However, I am still getting about 3Hz. (assuming I am doing the calculation correclty)
Is there a better way to monitor this?

The performance requirement for us is 15 Hz. I have disconnected all USB devices atm. 

```
ubuntu@ubuntu:~$ ros2 bag info sensors_v11/

Files:             sensors_v11_0.db3
Bag size:          44.0 MiB
Storage id:        sqlite3
Duration:          15.965s
Start:             Nov  1 2022 22:28:01.292 (1667341681.292)
End:               Nov  1 2022 22:28:17.258 (1667341697.258)
Messages:          50
Topic information: Topic: /camera/color/image_raw | Type: sensor_msgs/msg/Image | Count: 50 | Serialization Format: cdr
```

MartyG-RealSense, 2022-11-01:
Does disabling an RGB setting called **auto_exposure_priority** make a difference?

`ros2 param set /camera/camera rgb_camera.auto_exposure_priority false`

If auto_exposure is true and auto_exposure_priority is false then it should force the FPS to be maintained at a constant rate instead of being allowed to vary.

husseinalijaafar, 2022-11-02:
Hi Marty,

This seems to help, but we still are only able to achieve ~5Hz.
```
Files:             no_depth_no_exposure_2_0.db3
Bag size:          237.6 MiB
Storage id:        sqlite3
Duration:          48.891s
Start:             Nov  2 2022 21:04:22.622 (1667423062.622)
End:               Nov  2 2022 21:05:11.514 (1667423111.514)
Messages:          270
Topic information: Topic: /camera/color/image_raw | Type: sensor_msgs/msg/Image | Count: 270 | Serialization Format: cdr
```

I also disabled depth, as from my understanding that is also resource intensive. Is there any reason why we are only getting such frame rate?

MartyG-RealSense, 2022-11-03:
Depth on its own should not be resource intensive, though a low-end computing device could struggle more than a higher-specification device such as a Pi 4.  

Operations such as pointclouds, depth-color alignment and post-processing filters can be processing-intensive, though you do not seem to have any of these enabled.

The low amount of real memory (1 GB) on a Pi 3 could be a contributory factor, as a memory capacity that small could be used up quickly.  Once a Pi uses up its real memory capacity then it will start using the **swapfile** (which is really drive storage space used as slower 'virtual' memory).  Increasing the size of the swapfile will increase the amount of virtual memory but not increase the speed at which it operates.

When using Ubuntu as the operating system on the Pi 3, better performance may be achieved by using **Ubuntu MATE** instead, as described in the Intel RealSense installation guide for Pi 3 at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

MartyG-RealSense, 2022-11-09:
Hi @husseinalijaafar  Do you require further assistance with this case, please?  Thanks!

husseinalijaafar, 2022-11-09:
Not at the moment. Thank you Marty

closed 2022-11-09 by husseinalijaafar