realsenseai/realsense-ros #2531: [ros2] colcon failed to build ros2 wrapper

Martzi, 2022-11-03:
System: ubuntu focal, x64
ROS: ros2 galaxy
SDK: librealsense2=2.51.1-0~realsense0.7526

My problem is that I tried to compile ros2 wrapper with colcon, but I get the following error. I tried with all the different branches for ros2 (ros2-development, ros2-legacy, old/ros2-511) both fail at the same place.

```
CMake Error at /home/ubuntu/docker-home/robot_ws/install/realsense2_camera_msgs/share/realsense2_camera_msgs/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/home/ubuntu/docker-home/robot_ws/install/realsense2_camera_msgs/share/realsense2_camera_msgs/cmake/realsense2_camera_msgs__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  /home/ubuntu/docker-home/robot_ws/install/realsense2_camera_msgs/share/realsense2_camera_msgs/cmake/realsense2_camera_msgsConfig.cmake:41 (include)
  CMakeLists.txt:93 (find_package)
```

MartyG-RealSense, 2022-11-03:
Hi @Martzi  The wrapper names were changed 3 days ago at the time of writing this, as described at https://github.com/IntelRealSense/realsense-ros/issues/2527

Have you previously been able to install the ROS2 wrapper on Galactic before this change, please?

Martzi, 2022-11-03:
Hi!
I just tried this yesterday and today and the error is the same.

MartyG-RealSense, 2022-11-03:
I will highlight your case to the RealSense ROS team.  Thanks very much for your patience.

Martzi, 2022-11-03:
Yes, I forgot to mention that. It is inside a docker image. Should I give additional details about that?

MartyG-RealSense, 2022-11-03:
Yes please, additional details about your docker image are welcome in case it is a factor in your problem.

Martzi, 2022-11-03:
i use dorowu/ubuntu-desktop-lxde-vnc:focal as a base image [https://hub.docker.com/r/dorowu/ubuntu-desktop-lxde-vnc](https://hub.docker.com/r/dorowu/ubuntu-desktop-lxde-vnc)

I cloned the UR ros2 repo locally, copy to the docker environment and build there with colcon.
```
RUN mkdir -p /home/ubuntu/docker-home/robot_ws/src

ENV COLCON_WS=/home/ubuntu/docker-home/robot_ws

WORKDIR /home/ubuntu/docker-home/robot_ws

COPY /docker-home/robot_ws/src /src

RUN colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
```

MartyG-RealSense, 2022-11-03:
Does it make any difference if you simply use **colcon build** instead of adding the Release arg, since the official instructions only specify `colcon build`

https://github.com/IntelRealSense/realsense-ros#step-5-build

Martzi, 2022-11-03:
> Also, have the rosdep dependencies in Step 4 been installed?
Yes, Step 4 also done

>Can you please tell me what is the cmake version ? (cmake --version)
Thanks for the question, I am using Cmake 3.8 so definitely above 3.24. I will check it on later if it is the issue and reply!

MartyG-RealSense, 2022-11-09:
Hi @Martzi  Do you have an update about this case that you can provide, please?  Thanks!

Martzi, 2022-11-10:
Hi, sorry for the late reply.

Finally I have time to test it out and yes, it was really the cmake version mismatch, it really does not prefer above 3.24.0.
Thans @SamerKhshiboun and @MartyG-RealSense !

closed 2022-11-10 by Martzi

MartyG-RealSense, 2022-11-10:
You are very welcome, @Martzi - it's excellent news that the problem is resolved.  Thanks very much for the update!