realsenseai/realsense-ros #2532: conflicting implementation for realsense2_camera::DeviceInfo

Akumar201, 2022-11-03:
Hi, I am trying to write a publisher to publish data to another computer where ROS will be running. However, I am getting the following error. I am sharing the code with you. I am getting the following error

```
Compiling ros_rust_bridge v0.1.0 (/home/akumar/rosrust/my_won/ros_rust_bridge)
error[E0428]: the name `DeviceInfo` is defined multiple times
 --> src/main.rs:5:1
  |
5 | roslibrust_codegen_macro::find_and_generate_ros_messages!();
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  | |
  | `DeviceInfo` redefined here
  | previous definition of the type `DeviceInfo` here
  |
  = note: `DeviceInfo` must be defined only once in the type namespace of this module
  = note: this error originates in the macro `roslibrust_codegen_macro::find_and_generate_ros_messages` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: use of undeclared crate or module `std_msg`
  --> src/main.rs:20:40
   |
20 |     let publisher = client.advertise::<std_msg::String>("talker").await?;
   |                                        ^^^^^^^ use of undeclared crate or module `std_msg`
   |
help: there is a crate or module with a similar name
   |
20 |     let publisher = client.advertise::<std_msgs::String>("talker").await?;
   |                                        ~~~~~~~~

warning: unused import: `roslibrust_codegen_macro::find_and_generate_ros_messages`
 --> src/main.rs:3:5
  |
3 | use roslibrust_codegen_macro::find_and_generate_ros_messages;
  |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  |
  = note: `#[warn(unused_imports)]` on by default

error[E0119]: conflicting implementations of trait `roslibrust::RosServiceType` for type `realsense2_camera::DeviceInfo`
 --> src/main.rs:5:1
  |
5 | roslibrust_codegen_macro::find_and_generate_ros_messages!();
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  | |
  | first implementation here
  | conflicting implementation for `realsense2_camera::DeviceInfo`
  |
  = note: this error originates in the macro `roslibrust_codegen_macro::find_and_generate_ros_messages` (in Nightly builds, run with -Z macro-backtrace for more info)

Some errors have detailed explanations: E0119, E0428, E0433.
For more information about an error, try `rustc --explain E0119`.
warning: `ros_rust_bridge` (bin "ros_rust_bridge") generated 1 warning
error: could not compile `ros_rust_bridge` due to 3 previous errors; 1 warning emitted

```

**CODE**
```
use log::*;
use roslibrust::ClientHandle;
use roslibrust_codegen_macro::find_and_generate_ros_messages;

roslibrust_codegen_macro::find_and_generate_ros_messages!();

/// This example creates a client, and publishes a message to the topic "talker"
/// Running this example at the same time as subscribe_and_log will have the two examples
/// pass messages between each other.
/// To run this example a rosbridge websocket server should be running at the deafult port (9090).
#[tokio::main(flavor = "multi_thread")]
async fn main() -> Result<(), Box<dyn std::error::Error + Send + Sync>> {
    simple_logger::SimpleLogger::new()
        .with_level(log::LevelFilter::Debug)
        .without_timestamps() // required for running wsl2
        .init()
        .unwrap();

    let client = ClientHandle::new("ws://localhost:9090").await?;
    let publisher = client.advertise::<std_msgs::Header>("talker").await?;

    loop {
        let msg = std_msgs::Header::default();
        info!("About to publish");
        publisher.publish(msg).await.unwrap();
        info!("Published msg...");
        tokio::time::sleep(tokio::time::Duration::from_secs(1)).await;
    }
}

```

MartyG-RealSense, 2022-11-04:
Hi @Akumar201  There are not any previous reported cases of RealSense being used with the ROS bridge **rosrust**.  

https://github.com/adnanademovic/rosrust

Looking through the logs, there does not seem to be anything that the RealSense ROS team could do to adjust rosrust to resolve this error, unfortunately.

There is a RealSense binding for rust called **realsense-rust** available, but it is for the librealsense SDK rather than the RealSense ROS wrapper.

https://github.com/Tangram-Vision/realsense-rust

***********

Another approach to transferring RealSense ROS data between two computers would be to use **ROS_DOMAIN_ID** for ROS2 - as described at https://github.com/IntelRealSense/realsense-ros/issues/2502#issuecomment-1277420736 - or **ROS_MASTER_URI** in ROS1, as described in the Intel guide at the link below.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

Akumar201, 2022-11-10:
Hi @MartyG-RealSense , I resolved this issue by finding out a alternative method using ROS_MASTER_URI, however this script is running successfully in the system where RealSense software is not installed. For your reference I have tried from scratch, where I install RealSense using the follwoing command  `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` as suggested by you. Luckily I don't need ros wrapper on the system from where I have to send the data. 

I am closing this issue as it has been solved. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2022-11-10 by Akumar201

MartyG-RealSense, 2022-11-10:
Thanks very much @Akumar201 for the update - I'm pleased to hear that you achieved a solution!