realsenseai/realsense-ros #2533: Check for multiple Version isnatlled

Akumar201, 2022-11-03:
Hi, I would like to know how can I check if I have multiple version of realsense install ?

MartyG-RealSense, 2022-11-04:
Hi @Akumar201  A RealSense team member provides advice at https://github.com/IntelRealSense/librealsense/issues/1484#issuecomment-378854482 for checking for multiple versions of the RealSense SDK.

If the SDK has been installed using both packages and source code on the same computer then a clue to this will be if the RealSense Viewer tool (realsense-viewer) is launched and the red warning message **Multiple realsense udev-rules were found!** appears in the top corner of the Viewer window.

If you install librealsense from packages then you can remove all RealSense related packages from your computer with the command below so that you can be confident that the next package installation of the RealSense SDK that you perform will be the only one on the computer.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

If you built librealsense from a source code folder with CMake then I would recommend finding each 'librealsense' source code folder if you have multiple versions of them, going to the 'build' sub-directory of the folder and inputting the CMake build instruction below.

**sudo make uninstall && make clean**

Akumar201, 2022-11-04:
Hi, thank you for your response, I guess while installing I did some mistake, now when I am doing `sudo apt update`, I am getting the following hints, 

```
Hit:1 http://security.ubuntu.com/ubuntu xenial-security InRelease
Hit:2 http://archive.ubuntu.com/ubuntu xenial InRelease                                                                         
Hit:3 http://packages.ros.org/ros/ubuntu focal InRelease                                                                        
Hit:4 http://lenovo.archive.canonical.com focal InRelease                                                                       
Hit:5 http://archive.ubuntu.com/ubuntu xenial-updates InRelease                                                          
Hit:6 https://packages.microsoft.com/repos/ms-teams stable InRelease                               
Hit:7 http://ports.ubuntu.com/ubuntu-ports xenial InRelease                  
Hit:8 http://ports.ubuntu.com/ubuntu-ports xenial-updates InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.

```
However, I am using Ubuntu 20.04, focal. I guess my mistake is I somehow downloaded the xenial version also. I tried using the suggested command but the problem persists `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`, please guide me through this.

MartyG-RealSense, 2022-11-04:
Can you link me to the RealSense SDK installation instruction page that you are using, please?

Akumar201, 2022-11-04:
I used this one https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy

MartyG-RealSense, 2022-11-04:
Thanks very much.  So you installed the librealsense SDK and the RealSense ROS wrapper from packages at the same time using the 'Method 1' apt-get install instructions?

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#method-1-the-ros-distribution

Because Method 1 installs both librealsense and the wrapper at the same time, the librealsense SDK should not be installed separately before or after using **sudo apt-get install ros-$ROS_DISTRO-realsense2-camera** otherwise you could have two potentially conflicting versions of librealsense on the same computer.

If you can launch your RealSense camera in ROS successfully with `roslaunch realsense2_camera rs_camera.launch` and it performs well then it is likely that you do not need to be concerned about the possibility of multiple versions.

MartyG-RealSense, 2022-11-14:
Hi @Akumar201  Do you require further assistance with this case, please?  Thanks!

Akumar201, 2022-11-14:
This issue has been resolved.

closed 2022-11-14 by Akumar201

MartyG-RealSense, 2022-11-14:
That's great to hear, @Akumar201 - thanks very much for the update  :)