realsenseai/realsense-ros #2538: How to get Color images and depth images which are taken at the same angle

Xue21, 2022-11-09:
How cam I get  the color and  the depth image at the same time? I want the Color images and depth images which are taken at the same angle.

MartyG-RealSense, 2022-11-09:
Hi @Xue21  Which programming language are you using please (C++, Python, C#, etc).  Thanks!

Xue21, 2022-11-09:
Hi@MartyG-RealSense I am using the librealsense SDK, and I do not want to turn on the laser.

MartyG-RealSense, 2022-11-09:
Thanks very much for the confirmation.  Are you using C++ or Python language?

Xue21, 2022-11-09:
Hi@MartyG-RealSense I am using Python

MartyG-RealSense, 2022-11-09:
The pre-made librealsense Python example program **opencv_viewer_example.py** may provide what you are aiming to achieve.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

[image]

If you wish to align depth and color together into a single combined image then the example program **align_depth2color.py** could be used.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

[image]

Xue21, 2022-11-09:
Hi@MartyG-RealSense, Thanks very much.How can I save it as images or mp4?

Xue21, 2022-11-09:
Hi@MartyG-RealSense,Thanks very much!

closed 2022-11-15 by Xue21