realsenseai/realsense-ros #2539: From single-threaded to multi-threaded

wash0126, 2022-11-11:
I'm trying to use the D435i with VIM4, but it's too loud to work.
When I check the CPU usage rate with Htop, I feel that the cores being used are limited.
I think ROS2 can use more cores for MultiThreadedExecutor.
Where can I make this change?

MartyG-RealSense, 2022-11-11:
Hi @wash0126  If the librealsense SDK is built from source code with CMake with the build flag **-DBUILD_WITH_OPENMP** included in the CMake build instruction then the SDK can take advantage of multiple cores when performing depth-color alignment or YUY to RGB color conversion.

In regard to MultiThreadedExecutor, ROS' official documentation page for it at the link below indicates that it can be enabled by calling `add_node(..) `for each node.

https://docs.ros.org/en/foxy/Concepts/About-Executors.html

wash0126, 2022-11-17:
Hi

I tried building the librealsense SDK.
However, while it did improve some CPU utilization, it wasn't as much as we needed.
I still feel it's a ROS2 issue.
MultiThread when launching Node with LaunchFile should ROSAnswes listen? Or do you know a solution?

MartyG-RealSense, 2022-11-17:
I do not have further ROS2 suggestions to offer, unfortunately.  

If loud fan noise is a problem on your VIM4 then - assuming that you are using Ubuntu - you could experiment with **throttling** the fan speed by installing fan control software, starting with a small amount of throttling during testing to ensure that the board does not overheat.  Googling for the term **fan control software ubuntu** provides some examples of such software.

wash0126, 2022-11-20:
Hi.
Thank you for your help with my problem.

MartyG-RealSense, 2022-11-20:
You are very welcome, @wash0126  - I'm pleased that I could help!

closed 2022-11-21 by wash0126