realsenseai/realsense-ros #2540: Change name topics

alvaropascuaI, 2022-11-11:
I will launch 2 cameras at the same time, i need one camera publish messages for differents topics than the other camera.
how i can change the name of topics for camera ¿?

MartyG-RealSense, 2022-11-11:
Hi @alvaropascuaI  If you are using the ROS1 wrapper then you could launch each camera with **rs_camera.launch** in a separate ROS terminal and define an individual **camera** name in the launch instruction, such as 'cam1' and 'cam2'.  For example:

**Terminal 1**
roslaunch realsense2_camera rs_camera.launch camera:=cam_1 serial_no:=<serial number of the first camera>

**Terminal 2**
roslaunch realsense2_camera rs_camera.launch camera:=cam_2 serial_no:=<serial number of the second camera>

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#work-with-multiple-cameras

************

If you wanted to publish different stream types for each camera (such as depth only from one camera and color only from another) then you could also use the roslaunch instruction in each individual ROS terminal to define this.  For example:

**Camera 1 - depth enabled, color disabled**

enable_depth:=true enable_color:=false

**Camera 2 - depth disabled, color enabled**

enable_depth:=false enable_color:=true

*********

If you are using the ROS2 wrapper instead of ROS1 then please let me know.  Thanks!

alvaropascuaI, 2022-11-11:
Yes, i am using ROS2 wrapper

MartyG-RealSense, 2022-11-11:
The ROS2 wrapper **ros2-development** (formerly called ros2_beta) has a pre-made launch file called **rs_multi_camera_launch.py** that defines separate names for two cameras ('camera1' and 'camera2').

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_multi_camera_launch.py

alvaropascuaI, 2022-11-11:
Okay, thank you very much, I'll close the issue, thanks for everything

MartyG-RealSense, 2022-11-11:
You are very welcome, @alvaropascuaI   :)

alvaropascuaI, 2022-11-14:
i have the following question, if i would change name of topics, for example camera1 for camera_left ¿?
[image]

alvaropascuaI, 2022-11-22:
Hi, i need launch file rs_multi_camera, but i need publish imu topics, but how can i pass parameters for differents cameras ?

MartyG-RealSense, 2022-11-22:
I have not seen a past case of IMU being enabled for rs_multi_camera_launch.py but as the launch file imports the parameters of rs_launch.py, I would assume that enabling the accel and gyro topics in the roslaunch instruction (which are disabled by default) would publish IMU topics for 'camera1' and 'camera2'.  Try adding the commands below to the end of the  rs_multi_camera_launch.py launch instruction please.

**enable_accel:=true enable_gyro:=true**

alvaropascuaI, 2022-11-22:
I'm trying to start cameras separately, but I can't rebuild to rtabmap with this command

**ros2 launch rtabmap_ros rtabmap.launch.py args:="-d" stereo:=true left_image_topic:=/camera_left/color/image_raw right_image_topic:=/camera_right/color/image_raw left_camera_info_topic:=/camera_left/color/camera_info right_camera_info_topic:=/camera_right/color/camera_info frame_id:=camera_link approx_sync:=true subscribe_rgb:=true rgb_topic:=/camera_right/color/image_raw**

MartyG-RealSense, 2022-11-23:
I hope that https://github.com/introlab/rtabmap_ros/issues/459 will be a useful reference for the problem of starting two cameras with rtabmap_ros.

closed 2022-11-23 by alvaropascuaI