realsenseai/realsense-ros #2542: No data published in /camera/imu even after setting unite_imu_method:=linear_interpolation

tsadarsh, 2022-11-12:
I have my Intel Realsense d435i connected to my Pop!_OS running **ROS2 Humble**. Here is the output from `lsb_release -a`:
```
Distributor ID:	Pop
Description:	Pop!_OS 22.04 LTS
Release:	22.04
Codename:	jammy
```

I am using the ros2-development branch and running the following command:
`ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true  enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation`

When I `ros2 topic echo /camera/imu` there is no data published. 

But I do get the output from `/camera/accel/sample` and `/camera/gyro/sample`. **Am I specifying the correct parameter value to _unite_imu_method_?**

Here is my `ros2 topic list`:
```
/camera/accel/imu_info
/camera/accel/metadata
/camera/accel/sample
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/aligned_depth_to_color/image_raw/compressed
/camera/aligned_depth_to_color/image_raw/compressedDepth
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressedDepth
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/metadata
/camera/extrinsics/depth_to_accel
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_gyro
/camera/gyro/imu_info
/camera/gyro/metadata
/camera/gyro/sample
/camera/imu
/parameter_events
/rosout
/tf_static
```

MartyG-RealSense, 2022-11-12:
Hi @tsadarsh  Your launch instruction looks fine.  Is data published from the IMU topic if you use **copy** instead of linear_interpolation?  The 'copy' state can provide more stable IMU data.

`unite_imu_method:=copy`

May I also confirm whether you are using ros2-development wrapper **4.51.1** and librealsense SDK version **2.51.1**, as wrapper 4.51.1 is the version in which support for ROS2 Humble and Ubuntu 22.04 was officially introduced.

https://github.com/IntelRealSense/realsense-ros/releases/tag/4.51.1

chivas1000, 2022-11-14:
> Hi @tsadarsh Your launch instruction looks fine. Is data published from the /imu topic if you use **copy** instead of linear_interpolation? The 'copy' state can provide more stable IMU data. [...]

Hi I'm using D435i and my branch is ROS2-beta
and I've also ran into this issue, the accel and gyro are working, but the /camera/imu doesn't show data
also, ros2 topic echo /camera/accel/imu_info does not show data but /camera/accel/metadata and /camera/accel/sample has data, so do the gyro.
Is that because I'm using ros2-beta branch wrapper since now ros2-development is available.

realsense.yaml:
device_type: ''
serial_no: ''
usb_port_id: ''

rgb_camera:
  profile: '640x480x15'
color_qos: "SENSOR_DATA"

depth_module:
  profile: '848x480x60'
  emitter_enabled: 1
  emitter_on_off: true
depth_qos: "SENSOR_DATA"
depth_info_qos: "SYSTEM_DEFAULT"

infra_qos: "SENSOR_DATA"

#enable_accel: false
enable_color: true
enable_depth: true
#enable_gyro: false
enable_infra1: true
enable_infra2: true

# imu
enable_gyro: true
enable_accel: true
unite_imu_method: copy
#enable_sync: true

# may need
# hold_back_imu_for_frames

pointcloud:
  enable: false

pointcloud_texture_index: 0
pointcloud_texture_stream: RS2_STREAM_ANY

enable_sync: false
align_depth: false

MartyG-RealSense, 2022-11-14:
Hi @chivas1000  **ros2-development** is just a renaming of the branch from its former name of 'ros2_beta' and not a change to the wrapper's code.  So it should not matter if you installed the wrapper when it was still called ros2_beta.

Can you confirm if it makes no difference if linear_interpolation is placed in quotation marks?

`unite_imu_method:="linear_interpolation"`

chivas1000, 2022-11-14:
> Hi @chivas1000 **ros2-development** is just a renaming of the branch from its former name of 'ros2_beta' and not a change to the wrapper's code. So it should not matter if you installed the wrapper [...]

no difference, still the accel and gyro has data while imu does not.

changed in yaml:
unite_imu_method: "linear_interpolation"

output:
nuc@nuc-NUC10i7FNK:~$ ros2 param get /camera/camera unite_imu_method
String value is: linear_interpolation

also tested in:
unite_imu_method: "copy"

results are the same

tsadarsh, 2022-11-14:
> Hi @tsadarsh Your launch instruction looks fine. Is data published from the /imu topic if you use **copy** instead of linear_interpolation? The 'copy' state can provide more stable IMU data. [...]

Sorry for the delayed response.

I tried to use `unite_imu_method:=copy` but there is still no data being published in the `/camera/imu` topic. 

How do I find out which version of the `ros2-development` wrapper and `librealsense SDK` I am using?

tsadarsh, 2022-11-14:
> I have my Intel Realsense d435i connected to my Pop!_OS running **ROS2 Humble**. Here is the output from `lsb_release -a`:
> 
> ```
> Distributor ID:	Pop
> Description:	Pop!_OS 22.04 LTS
> Release:	22.04
> Codename:	jammy
> ```

Like I mentioned before this is my OS and ROS version that I am running.

Here I writing down the steps that I followed to get started with ROS2 and Realsense integration

## Installation procedure that I took

I went through IntelRealsense/realsense-ros [README.md](https://github.com/IntelRealSense/realsense-ros#ros2-wrapper-for-intel-realsense-devices) 

Since I already have ROS2 Humble installed I proceeded to [Install the latest Intel® RealSense™ SDK 2.0](https://github.com/IntelRealSense/realsense-ros#step-2-install-the-latest-intel-realsense-sdk-20). The given option 1 is not applicable to me as:
> Option1 ... **(Not supported in Ubuntu 22.04 yet)**

So I proceeded to [Option 2](https://github.com/IntelRealSense/realsense-ros#option-2-build-from-source) to build from source. I was redirected to the [releases page](https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1). In the **Assets** section I could see only `*.exe` files. 

**So where do I install the SDK binary for Linux from?**

I then went to the **[Linux Installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)* page from the **Important Links** section. There, in the [Installing the packages](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) I registered the server's public key by running the given command:
```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
```
This is the output that I got:
```
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.8yij8MrBnM/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: keyserver receive failed: No route to host
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.rLxBuNs81n/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
```
Why does the **gpg: keyserver receive failed: No route to host**?

I then added the server to the list of repositories by:
```
sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
```
The next step says the **Install the binaries**. 
I am able to install `librealsense2-dkms` (1.3.19-0ubuntu1) but when I ` sudo apt-get install librealsense2-utils`, I am presented with:
```
E: Unable to locate package librealsense2-utils
```
The same happens with the following package installation:
```
E: Unable to locate package librealsense2-dev
E: Unable to locate package librealsense2-dbg
```
How do I proceed further to install the Realsense SDK? Should I open a separate Issue to address this Issue?

## Installing directly ROS2 realsense packages

I now directly did a `sudo apt install ros-humble-realsense2*` which installed the following packages:
```
ros-humble-realsense2-camera
ros-humble-realsense2-camera-dbgsym
ros-humble-realsense2-camera-msgs
ros-humble-realsense2-camera-msgs-dbgsym
ros-humble-realsense2-description
```
Now plugging in my Intel Realsense d435i and running `ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation` publishes all the topics mentioned in the [start of this thread](https://github.com/IntelRealSense/realsense-ros/issues/2542#issue-1446491415). And then the issue that this thread address is what I am facing.

At this point I badly need direction.

MartyG-RealSense, 2022-11-16:
@tsadarsh  The librealsense and ROS wrapper versions that you are using should be listed in the ROS launch log.  For example when using the **ros2-development** wrapper with librealsense 2.51.1 and wrapper 4.51.1:

```
INFO] [realsense2_camera_node-1]: process started with pid [7112]
[realsense2_camera_node-1] [INFO] [1668461851.324275521] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1668461851.324566109] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1668461851.324635910] [camera.camera]: Running with LibRealSense v2.51.1
```

I note that you are using **Pop!_OS** that is based on Ubuntu, rather than using Ubuntu directly. Under these circumstances, the librealsense SDK may be most stable if you build it from source code using the **RSUSB backend** installation method, which bypasses the Linux kernel and therefore is not dependent on Linux versions or kernel versions and does not require kernel patching.  However, RealSense ROS installation packages are based upon the RSUSB backend and so you may be achieving the same effect as an RSUSB source code build with your `sudo apt install ros-humble-realsense2*` package install.

In regard to the issues with the **/imu** topic reported by yourself and @chivas1000 - I will highlight it to the RealSense ROS developer team.

chivas1000, 2022-11-17:
hi @MartyG-RealSense 
I think this issue is similar
https://github.com/IntelRealSense/realsense-ros/issues/2470
is that the librealsense issue, I also not able to do RSUSB BACKEND due to my application.
so switching back to older version would be a choice?
that I'm using ubuntu 20.04 and ROS2 foxy, so which version would be recomended?
thanks

tsadarsh, 2022-11-17:
@MartyG-RealSense Thank you for the help. Yes, I am running, `librealsense 2.51.1` and `wrapper 4.51.1`. This showed up in my ROS logs during the launch sequence. 

> I will highlight it to the RealSense ROS developer team.

Thank you again. Waiting for your reply eagerly.

Nir-Az, 2022-11-17:
> @Nir-Az Thank you for the help. The fused data is now published in `/camera/imu` after setting `unite_imu_method:=2`.
> 
> This detail should be updated in the README.md. Can I make a PR regarding this?
Glad it works.
About the docs update, good idea, it will be appreciated,  thanks 😀

closed 2022-11-23 by tsadarsh