realsenseai/realsense-ros #2543: Ordering of the extrinsic rotation

geoeo, 2022-11-14:
The rotation in the extrinsic message is defined as float[9]. 
But it does not state explicitly if its row major or column major.

Which one does the array in the extrinsic message represent?

Best,
Marc

MartyG-RealSense, 2022-11-14:
Hi @geoeo  The SDK files at the links below define extrinsic rotation as **column-major** in their code notes. 

https://github.com/leggedrobotics/librealsense-rsl/blob/master/wrappers/python/python.cpp#L160

http://docs.ros.org/en/kinetic/api/librealsense2/html/Extrinsic_8java_source.html

geoeo, 2022-11-14:
Thanks @MartyG-RealSense !

Thats a pretty obscure link. A mention of this in the Readme.md would be appreciated!

MartyG-RealSense, 2022-11-16:
I will add a **Documentation** label to this case to highlight it in the future to Intel RealSense colleagues responsible for documentation.  This case should be kept open but you do not need to take any further action.  Thanks again!

SamerKhshiboun, 2023-04-13:
@MartyG-RealSense, the fix #2693 was merged. 
You can close this ticket

MartyG-RealSense, 2023-04-13:
Thanks very much @SamerKhshiboun  :)

closed 2023-04-13 by MartyG-RealSense