realsenseai/realsense-ros #2544: Noob-question: installation

ryanmaxwell96, 2022-11-14:
Hello,

Sorry for the newbie question, but I'm not familiar with how to install the package from a tar.gz file. I downloaded the Source.tar.gz file and extracted it to a folder I made so it looks like: ~/ros2_elequent_camera/src/realsense-ros-3.1.0

I saw online that I'm supposed to be able to install by running cmake ../realsense-ros-3.1.0 but I get this error when I run that:

CMake Error at CMakeLists.txt:94 (find_package):
  By not providing "Findrealsense_camera_msgs.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "realsense_camera_msgs", but CMake did not find one.

  Could not find a package configuration file provided by
  "realsense_camera_msgs" with any of the following names:

    realsense_camera_msgsConfig.cmake
    realsense_camera_msgs-config.cmake

  Add the installation prefix of "realsense_camera_msgs" to CMAKE_PREFIX_PATH
  or set "realsense_camera_msgs_DIR" to a directory containing one of the
  above files.  If "realsense_camera_msgs" provides a separate development
  package or SDK, be sure it has been installed.

I am using Ubuntu 18.04 and ros2-eloquent.

Thanks for any help anyone can give!

MartyG-RealSense, 2022-11-15:
Hi @ryanmaxwell96  The **3.1.0** Eloquent ROS2 wrapper is very old, dating back to September 2020, and is now obsolete.  It dates back to a time when there were two separate individual ROS2 wrappers for Eloquent and Foxy, and then the two wrappers were merged into one.  After that, this wrapper became obsolete as a new wrapper **ros2-development** (formerly ros2_beta) was introduced.  ros2-development is now the default RealSense ROS wrapper and is compatible with Eloquent if Ubuntu 18.04 is being used.

https://github.com/IntelRealSense/realsense-ros/releases/tag/4.51.1

I would not recommend using an old wrapper version such as 3.1.0 unless your project needs to use an old librealsense SDK version, such as the **2.38.1** SDK version that matches with 3.1.0 (it is important to match a particular ROS wrapper version with a particular SDK version listed in that wrapper version's release notes).

********

When building the RealSense ROS wrapper from source code, the librealsense SDK should be installed first and then the ROS wrapper installed secondly once librealsense has been confirmed to work successfully.  

With the current modern ros2-development wrapper (which should be matched to librealsense SDK version 2.51.1), once librealsense has been installed then you can build the wrapper by cloning it into your ROS2 workspace instead of using the tar.gz file.

https://github.com/IntelRealSense/realsense-ros/#step-3-install-intel-realsense-ros2-wrapper-from-sources

MartyG-RealSense, 2022-11-15:
Generally, a RealSense ROS wrapper installation involves either installing the librealsense SDK and ROS wrapper together at the same time from packages using **apt-get install**, or instead building the wrapper from source code by cloning it.  So unless you need to modify the source code of the ROS wrapper, you are not likely to need to use the zip or tar.gz files.

ryanmaxwell96, 2022-11-16:
Ok thanks for the all the info. I'm using a Jetson AGX. Previously, I had tried following all of the steps in this link, but it fails at colcon build.
https://github.com/IntelRealSense/realsense-ros/#step-3-install-intel-realsense-ros2-wrapper-from-sources

These are this errors I got at the build:

`/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:526:9: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const int get_data_size() const
         ^~~~~
In file included from /opt/ros/eloquent/include/librealsense2/hpp/rs_sensor.hpp:8:0,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_device.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_context.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/rs.hpp:9,
                 from /home/nvidia/ros2_ws/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/nvidia/ros2_ws/src/realsense-ros/realsense2_camera/src/parameters.cpp:4:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:526:9: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const int get_data_size() const
         ^~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1020:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1042:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1064:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1020:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1042:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
In file included from /opt/ros/eloquent/include/librealsense2/hpp/rs_sensor.hpp:8:0,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_device.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_context.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/rs.hpp:9,
                 from /home/nvidia/ros2_ws/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/nvidia/ros2_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:4:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:526:9: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const int get_data_size() const
         ^~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1064:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1020:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1042:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1064:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
[... cut ...]
/opt/ros/eloquent/include/librealsense2/rsutil.h:150:13: warning: ‘void rs2_fov(const rs2_intrinsics*, float*)’ defined but not used [-Wunused-function]
 static void rs2_fov(const struct rs2_intrinsics * intrin, float to_fov[2])
             ^~~~~~~
make[1]: *** [CMakeFiles/realsense2_camera.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< realsense2_camera [59.6s, exited with code 2]

Summary: 3 packages finished [1min 6s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera
`

MartyG-RealSense, 2022-11-16:
For a Jetson installation, it is recommendable to install the librealsense SDK first using a Jetson-specific installation procedure and then install the ROS wrapper afterwards from source code once you have confirmed that librealsense is working successfully.

You can install librealsense for Jetson from packages with `apt-get install `using the instructions at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

Then build the `ros2-development` ROS wrapper from source code for ROS2 using the colcon build procedure.

https://github.com/IntelRealSense/realsense-ros/#step-3-install-intel-realsense-ros2-wrapper-from-sources

ryanmaxwell96, 2022-11-16:
Ok I will try this when I can in a few days

ryanmaxwell96, 2022-11-22:
[image]

ryanmaxwell96, 2022-11-22:
So this is what I get when I run realsense-viewer so it seems to be working.

But when I follow the instructions on the link, https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

I get this issue:

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
Executing: /tmp/apt-key-gpghome.M6wHtmMduE/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1

ryanmaxwell96, 2022-11-22:
And it looks like I still basically get the same errors:

colcon build
[1.424s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'turtlesim' is in: /opt/ros/eloquent
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding turtlesim

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> realsense2_camera_msgs
Starting >>> turtlesim
Finished <<< realsense2_camera_msgs [2.20s]                          
Starting >>> realsense2_camera
Starting >>> realsense2_description                             
Finished <<< realsense2_description [0.57s]                     
Finished <<< turtlesim [3.43s]                                  
--- stderr: realsense2_camera                               
In file included from /opt/ros/eloquent/include/librealsense2/hpp/rs_sensor.hpp:8:0,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_device.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_context.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/rs.hpp:9,
                 from /home/nvidia/ros2_ws/src/realsense-ros/realsense2_camera/include/ros_sensor.h:6,
                 from /home/nvidia/ros2_ws/src/realsense-ros/realsense2_camera/src/ros_sensor.cpp:4:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:526:9: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const int get_data_size() const
         ^~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1020:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1042:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1064:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
In file included from /opt/ros/eloquent/include/librealsense2/hpp/rs_sensor.hpp:8:0,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_device.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_context.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/rs.hpp:9,
                 from /home/nvidia/ros2_ws/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/nvidia/ros2_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:4:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:526:9: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const int get_data_size() const
         ^~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1020:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1042:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/eloquent/include/librealsense2/hpp/rs_frame.hpp:1064:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
In file included from /opt/ros/eloquent/include/librealsense2/hpp/rs_sensor.hpp:8:0,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_device.hpp:8,
                 from /opt/ros/eloquent/include/librealsense2/hpp/rs_record_playback.hpp:8,
[... cut ...]
/opt/ros/eloquent/include/librealsense2/rsutil.h:150:13: warning: ‘void rs2_fov(const rs2_intrinsics*, float*)’ defined but not used [-Wunused-function]
 static void rs2_fov(const struct rs2_intrinsics * intrin, float to_fov[2])
             ^~~~~~~
make[2]: *** [CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/realsense2_camera.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< realsense2_camera [24.6s, exited with code 2]
                                 
Summary: 3 packages finished [27.6s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera

ryanmaxwell96, 2022-11-23:
Update: followed the instructions on https://askubuntu.com/questions/36507/how-do-i-import-a-public-key

using the key

F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE 

to get these lines copied into my key.txt file:

-----BEGIN PGP PUBLIC KEY BLOCK-----
Comment: Hostname: 
Version: Hockeypuck 2.1.0-184-g50f1108

xsBNBFpUqDEBCADuIoVeNfgr3owUxn0keHq9OPak+Phxk4ye/paDZ5JZ6D407GXt
QI0Ct6TVj2J9tWkcsK+G4DDxOQDPLIMf+mgjJYX0jwt6ruminSIhhp5fvh3Ol33e
X/2P01Iul/GhVlpg6MspnIiJOnv0okYNbOzSLdgxpmTk6dyxP49pHbLdzI7JugsS
tJljQgBeBP6pNlzBVTDvEvP2cXnqTVFxm2skiw9hyN2o2CXjGhGOz6Z5L6iOf/ee
8beYHomkEaaoNXc3Z+V5KzSg4uM1uYzqecxQ1InhrsYfb0ETjvAeJiNp1y+sSm+O
CDJWNxJq8tNSFrwvKPZ+e6cvv15yN6hChkkRABEBAAHNNyJDTiA9IEludGVsKFIp
IEludGVsKFIpIFJlYWxzZW5zZSIsIE89SW50ZWwgQ29ycG9yYXRpb27CwH4EEwEI
ACgFAlpUqDECGwMFCRLNVIAGCwkIBwMCBhUIAgkKCwQWAgMBAh4BAheAAAoJEMiz
pVpvPvzeGIAIAJiwFHVL/iqnDRVfzcI/Nm8gtoMDV2hU6QGcNkbwZn5JpoIyGGPh
zR01rWYX6hvt9BzzZljW7MF6LPnfi0IXzZ8ml/wFx0GwF9u0acCkxCZr4aIReO4W
t0Pe8u6uAIyhNdn9a4yacxFPdeU2W2w1xDTRUidiDO4V9vCllK4VGGHoBLi9ua6C
qIZrdnRvX3ire6WybLz33JyDbEK9SlKqs8kwyz4uq9a0sry79oTQiK4jAgLxKGq/
JA5ixemTnVrVIa+CZgUHqMW0SrgZeJDVMh7x2ZPGilSxmPDMNniEd5mxrFetFrzy
quh0EmUG2XwVM89VNH4Y45USyqal4RhISMw=
=oqan
-----END PGP PUBLIC KEY BLOCK-----

Then executed 

sudo apt-key add key.txt

and got the OK confirmation. 

But the first line still says 

gpg: Total number processed: 1
gpg:                      unchanged: 1

MartyG-RealSense, 2022-11-23:
Thanks very much for your patience.  I would recommend ensuring that the librealsense SDK is working correctly before proceeding onwards to building the ROS wrapper, as there is then less problems to diagnose.

Is librealsense's RealSense Viewer able to run correctly, please?

ryanmaxwell96, 2022-11-23:
You mean does Realsense Viewer run and show the camera output? Yes, it runs and it shows both the 2D RGB camera picture as well as the 3D picture

ryanmaxwell96, 2022-11-24:
Ok thanks for the terminal command. Unfortunately, the realsense2_camera package cannot be found

ryanmaxwell96, 2022-11-25:
Well shoot. I thought I had it figured out. I ran this exactly at first:
`sudo apt-get install python3-rosdep -y
sudo rosdep init # "sudo rosdep init --include-eol-distros" for Eloquent and earlier
rosdep update # "sudo rosdep update --include-eol-distros" for Eloquent and earlier
rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys=librealsense2 -y`

But read it closer and realized I needed to use the commented out parts for sudo rosdep init and sudo rosdep update since I'm using Eloquent. Did this and still no luck - same colcon error message :(

MartyG-RealSense, 2022-11-25:
So you used the Eloquent instructions like this?

```
sudo rosdep init --include-eol-distros
sudo rosdep update --include-eol-distros
rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys=librealsense2 -y
colcon build
```

ryanmaxwell96, 2022-11-25:
Yes

MartyG-RealSense, 2022-11-26:
I reviewed this case again from the beginning.  A notable factor on your particular computer is that you are using **turtlesim** and received a warning about it after inputting **colcon build**.

https://github.com/IntelRealSense/realsense-ros/issues/2544#issuecomment-1322969415

If you are still receiving those turtlesim related errors then it may be worth considering following the log messages' advice to add **--allow-overriding turtlesim** to the command line if you have not tried this already.

ryanmaxwell96, 2022-11-26:
Ok I added the terms to remove the warning about turtlesim, but still have the same issues

MartyG-RealSense, 2022-11-27:
It seems that turtlesim is not a factor in the problem.  Your use of a Jetson AGX is also likely not a factor, since installing the ROS wrapper from source code is appropriate for Jetson boards.

The next action that we could try is building from source code the previous generation of ROS2 wrapper that also supports Eloquent.  The final version of that wrapper was **3.2.3**.  It was designed for librealsense 2.50.0 but should work with 2.51.0 as 2.51.1 was mostly bug fixes rather than new features.

https://github.com/IntelRealSense/realsense-ros/releases/tag/3.2.3

The source code installation instructions for wrapper 3.2.3 are similar to 4.51.1 and also use colcon build but have a different git clone command.

https://github.com/IntelRealSense/realsense-ros/tree/ros2-legacy#step-3-install-intel-realsense-ros2-wrapper-from-sources

*********

If that wrapper does not work either and multiple previous installation attempts have been made, doing a complete wipe of the computer and a fresh reinstall of Ubuntu, librealsense and the ROS wrapper can suddenly make previously unsolvable problems disappear.

ryanmaxwell96, 2022-11-27:
Ok. I was actually considering upgrading to Jetpack 5.0.2 and Ubuntu 20.04 and starting over with Foxy instead.

ryanmaxwell96, 2022-11-28:
Ok now I have Ubuntu 20.04 freshly installed as well as Jetpack 5.0.2. Will those two links you shared still work? On one it assumes Ubuntu 16/18

MartyG-RealSense, 2022-11-29:
The librealsense SDK would not be a problem for Ubuntu 20.04.

Foxy or Galactic are recommended for the 3.2.3 wrapper when using 20.04.  

Is there any possibility of upgrading your project from Eloquent to at least Foxy, please?

ryanmaxwell96, 2022-11-29:
Ok thanks! I am now on Ubuntu 20.04, Jetpack 5.0.2, with Foxy ROS2 distro and the camera seems to be working, except for the Motion Module. Haven't looked into those errors yet.

For now, is there a way to not have to run the command ". install/local_setup.bash" inside ros2_ws on every new terminal? Is there a way to source it in .bashrc somehow?

ryanmaxwell96, 2022-11-29:
Nevermind. Got it with
# ROS 2 Sourcing
source /opt/ros/foxy/setup.bash
source ~/ros2_ws/install/setup.bash
source /usr/share/colcon_argcomplete/hook/colcon-argcomplete.bash

MartyG-RealSense, 2022-11-30:
It's excellent to hear that you have made significant progress!

MartyG-RealSense, 2022-12-08:
That's great to hear, @ryanmaxwell96 - thanks very much for the update!  As you suggested, I will close the issue.  Thanks again!

closed 2022-12-08 by MartyG-RealSense

MartyG-RealSense, 2023-06-07:
Hi @ryanmaxwell96  The new 2.54.1 version of librealsense with JetPack 5.0.2 support is now available.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1

Release notes:
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2541