realsenseai/realsense-ros #2545: Very bad performance happened in raspberry pi 4B when using D435i

Bread2018, 2022-11-14:
Hi,
Camera: D435i
OS: raspberry pi 4B arm64 : ubuntu 20.04 server 
      laptop : ubuntu 20.04 desktop
ROS version : both ROS2 Foxy 

serious problem:
1.  [realsense2_camera_node-1]  14/11 21:49:36,188 WARNING [281473055693216] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
2. very poor fps and serious screen delay

I have connected d435i on raspberry pi 4B and ran ros2 launch realsense2_camera rs_launch.py, then i've use laptop to subscribe the color stream topic (camera/color/image_raw)  and showed the color stream on my laptop, the video picture was only about 1FPS, with a delay of 7~8 seconds.

I have used this code for compilation of  librealsense2.0 sdk:
        sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
        sudo apt-get install git cmake libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
        sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
        git clone https://github.com/IntelRealSense/librealsense.git
        cd librealsense
        mkdir build && cd build
        cmake ../ -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DFORCE_RSUSB_BACKEND=ON -DBUILD_WITH_TM2=false -DIMPORT_DEPTH_CAM_FW=false
        sudo make uninstall && make clean && make && sudo make install
        sudo ./scripts/setup_udev_rules.sh

this code is based on https://github.com/datasith/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

**Some terminal output**

ubuntu@ubuntu:~/ros2_ws$ export ROS_DOMAIN_ID=1
ubuntu@ubuntu:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x15 rgb_camera.profile:=640x480x15
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2022-11-14-21-37-30-710797-ubuntu-7110
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [7112]
[realsense2_camera_node-1] [INFO] [1668461851.324275521] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1668461851.324566109] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1668461851.324635910] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1668461851.686930257] [camera.camera]: Device with serial number 920312072254 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1668461851.687168721] [camera.camera]: Device with physical ID 2-1-2 was found.
[realsense2_camera_node-1] [INFO] [1668461851.687229545] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1668461851.688102569] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1668461851.688224290] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1668461851.694748673] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1668461851.695835209] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1668461851.695968259] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1668461851.696023956] [camera.camera]: Device Serial No: 920312072254
[realsense2_camera_node-1] [INFO] [1668461851.696069675] [camera.camera]: Device physical port: 2-1-2
[realsense2_camera_node-1] [INFO] [1668461851.696125557] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1668461851.696168241] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1668461851.696209648] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1668461854.400802277] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1668461855.014413384] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1668461855.016924742] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1668461855.017254835] [camera.camera]: Set ROS param accel_fps to default: 63
[realsense2_camera_node-1] [INFO] [1668461855.047711795] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1668461855.105875040] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1]  14/11 21:37:35,216 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1668461855.267600507] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1668461855.268649038] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1668461855.275115958] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1668461855.316802966] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1668461855.318592388] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1668461855.402311380] [camera.camera]:
[realsense2_camera_node-1]  14/11 21:37:35,771 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/11 21:37:36,224 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/11 21:37:36,376 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/11 21:37:36,427 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/11 21:37:36,578 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/11 21:37:36,629 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/11 21:37:36,780 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/11 21:37:36,830 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/11 21:37:36,982 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/11 21:37:37,033 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/11 21:37:37,184 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/11 21:37:37,235 WARNING [281473399630240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2022-11-15:
Hi @Bread2018  Installing the librealsense SDK on a Raspberry Pi with Ubuntu can be complicated, though it looks as though you have developed an installation method that results in a working SDK and ROS wrapper combination.

The warning **control_transfer returned error, index: 768, error: Resource temporarily unavailable**, if generating continuously in the log, indicates that there is a serious communication problem.  Very low speeds (such as 3 Hz) can sometimes occur when passing RealSense data over a network connection between two computers, such as your Pi 4 and laptop.

May I first ask whether performance improves if you reset the camera at launch on the Pi by adding **initial_reset:=true** to the launch instruction?

`ros2 launch realsense2_camera rs_launch.py initial_reset:=true`

Bread2018, 2022-11-15:
Thanks for help~
I did the reset, but there is a new problem comes :
ubuntu@ubuntu:~$ ros2 launch realsense2_camera rs_launch.py initial_reset:=true
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2022-11-15-12-51-07-629801-ubuntu-5656
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [5658]
[realsense2_camera_node-1] [INFO] [1668516668.234563913] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1668516668.234857446] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1668516668.234925737] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1668516668.586557700] [camera.camera]: Device with serial number 920312072254 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1668516668.586737557] [camera.camera]: Device with physical ID 2-1-3 was found.
[realsense2_camera_node-1] [INFO] [1668516668.586794664] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1668516668.587673244] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1668516668.587834491] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1668516668.587894319] [camera.camera]: Resetting device...

How long the reset will be take?   I have waited for ten minutes, but it did not reset successfully, it still keep resetting, so i interrupted it. Am I too impatient? Or does this situation means that something went wrong?

About the serious delay, i fixed it by adding swap memory, it looks normal a bit.

MartyG-RealSense, 2022-11-16:
It's no trouble at all. :)

A reset of the camera should typically complete in a couple of seconds.  It is abnormal for the reset to repeat continuously, though I have seen it happen in some ROS launches that have problems.

Does UFW work if you use the following procedure?

```
sudo ufw enable
sudo ufw allow ssh
```

Bread2018, 2022-11-17:
sudo ufw enable sudo ufw allow ssh
I tried but this did not work for ROS topics, i will try to allow DDS port.

About the low fps, in addition to the communication problem, is this problem related to the number of topics published by the launch file? I'm trying to post less topics, but I don't know how to do, I just need to publish:

/camera/depth/image_rect_raw
/camera/depth/camera_info
/camera/color/image_raw

can you help me?

By the way, Do you know of any solutions about the problem of reset of the D435i?

MartyG-RealSense, 2022-11-17:
If you are enabling both depth and RGB in the ROS2 wrapper then it may be worth following the advice at https://github.com/IntelRealSense/realsense-ros/issues/2503#issuecomment-1290293354 to see if your FPS drop is related to the same problem.

MartyG-RealSense, 2022-11-24:
Hi @Bread2018  Do you require further assistance with this case, please?  Thanks!

Bread2018, 2022-11-27:
No
Thanks for your help!

MartyG-RealSense, 2022-11-27:
You are very welcome.  :)

As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-11-27 by MartyG-RealSense

NDL200201, 2023-09-30:
When I run d435 camera on raspberry pi 4b. ros2 launch realsense2_camera rs_launch.py
I have an issue:
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [1967]
[realsense2_camera_node-1] [INFO] [1696051397.932102598] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1696051397.932507366] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1696051397.932593212] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1]  30/09 12:23:17,948 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:17,948 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:17,948 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:17,948 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:17,948 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:17,948 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:17,948 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:17,949 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:17,949 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:17,949 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:17,949 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:17,949 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:17,949 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:18,004 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:18,004 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:18,005 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:18,005 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied
[realsense2_camera_node-1]  30/09 12:23:18,005 ERROR [281473474488544] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Permission denied

I dont see RGB-D camera open, i only see color camera. 
[realsense2_camera_node-1] [WARN] [1696051400.246223232] [camera.camera]: XXX Hardware Notification:Depth stream start failure,1.69605e+12,Error,Hardware Error
[realsense2_camera_node-1] [WARN] [1696051400.246490883] [camera.camera]: Hardware Notification:Depth stream start failure,1.69605e+12,Error,Hardware Error
Can you help me fix it? Thanks.

MartyG-RealSense, 2023-09-30:
Hi @NDL200201  What method did you use to install the librealsense 2.54.1 SDK on your Pi 4b before you installed the ROS wrapper, please?  On Pi boards it is recommendable to build librealsense from source code with CMake and include the flag` -DFORCE_RSUSB_BACKEND=TRUE` in the CMake build instruction.

NDL200201, 2023-10-01:
Hi @MartyG-RealSense I was uninstall librealsense 2.54.1 SDK on your Pi 4b then I reinstall librealsense from source code when I run this command "./scripts/setup_udev_rules.sh" in step 2 of install librealsense2 on this link "https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md#install-librealsense2". I have this issue

robot@robot-desktop:~/librealsense$ ./scripts/setup_udev_rules.sh
Setting-up permissions for RealSense devices

Remove all RealSense cameras attached. Hit any key when ready

armv8_cortex_a72: Failed to write 'change' to '/sys/devices/armv8_cortex_a72/uevent': Permission denied
breakpoint: Failed to write 'change' to '/sys/devices/breakpoint/uevent': Permission denied
kprobe: Failed to write 'change' to '/sys/devices/kprobe/uevent': Permission denied
3ef62f40.nvram: Failed to write 'change' to '/sys/devices/platform/3ef62f40.nvram/uevent': Permission denied
rmem0: Failed to write 'change' to '/sys/devices/platform/3ef62f40.nvram/rmem0/uevent': Permission denied
Fixed MDIO bus.0: Failed to write 'change' to '/sys/devices/platform/Fixed MDIO bus.0/uevent': Permission denied
fixed-0: Failed to write 'change' to '/sys/devices/platform/Fixed MDIO bus.0/mdio_bus/fixed-0/uevent': Permission denied
arm-pmu: Failed to write 'change' to '/sys/devices/platform/arm-pmu/uevent': Permission denied
cam1_regulator: Failed to write 'change' to '/sys/devices/platform/cam1_regulator/uevent': Permission denied
regulator.5: Failed to write 'change' to '/sys/devices/platform/cam1_regulator/regulator/regulator.5/uevent': Permission denied
cam_dummy_reg: Failed to write 'change' to '/sys/devices/platform/cam_dummy_reg/uevent': Permission denied
regulator.1: Failed to write 'change' to '/sys/devices/platform/cam_dummy_reg/regulator/regulator.1/uevent': Permission denied
cpufreq-dt: Failed to write 'change' to '/sys/devices/platform/cpufreq-dt/uevent': Permission denied
emmc2bus: Failed to write 'change' to '/sys/devices/platform/emmc2bus/uevent': Permission denied
fe340000.mmc: Failed to write 'change' to '/sys/devices/platform/emmc2bus/fe340000.mmc/uevent': Permission denied
mmc0::: Failed to write 'change' to '/sys/devices/platform/emmc2bus/fe340000.mmc/leds/mmc0::/uevent': Permission denied
mmc0: Failed to write 'change' to '/sys/devices/platform/emmc2bus/fe340000.mmc/mmc_host/mmc0/uevent': Permission denied
mmc0:5048: Failed to write 'change' to '/sys/devices/platform/emmc2bus/fe340000.mmc/mmc_host/mmc0/mmc0:5048/uevent': Permission denied
How can fix this issue? Thanks...

MartyG-RealSense, 2023-10-01:
@NDL200201  Installing librealsense on Raspberry Pi is usually difficult to achieve, unfortunately.  But it can be done.

Using the source code installation script **libuvc_installation.sh** at the link below has worked in the past for Pi users. 

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Before running that script, I would recommend going to the 'build' folder of the source code installation that you just performed and removing it by using the CMake command `sudo make uninstall && make clean`

NDL200201, 2023-10-01:
WOW!!!
Thanks for helping me to solve this issue.

MartyG-RealSense, 2023-10-01:
You are very welcome!  :)

thuanphat1501, 2023-10-02:
Hi @MartyG-RealSense I installed librealsense on Raspberry Pi 4B using Ubuntu 22.04 and installed realsense-ros. I saw the depth image on Rviz2 but when I enable pointcloud I don't see it. Can you help me to fix this issue? Thank you.

MartyG-RealSense, 2023-10-02:
Hi @thuanphat1501  Did you enable the pointcloud using the **pointcloud.enable:=true** instruction, please?

`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true`

thuanphat1501, 2023-10-02:
@MartyG-RealSense Yes, I was enable the pointcloud using the pointcloud.enable:=true
[image]
[image]

I dont see the pointcloud on Rviz2

MartyG-RealSense, 2023-10-02:
Does the pointcloud appear if you reduce FPS from 30 to 15 and use 640x480 for both depth and color?

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x15 rgb_camera.profile:=640x480x15 pointcloud.enable:=true`

NDL200201, 2023-10-04:
[image]
I cannot see the color of point cloud in rviz2. This is my Node on file launch
Node(
            package='realsense2_camera',
            executable='realsense2_camera_node',
            output='screen',
            parameters=[
                {'depth_module.profile': '640x480x15'},
                {'rgb_camera.profile': '640x480x15'},
                {'pointcloud.enable': True}
            ]
        )

MartyG-RealSense, 2023-10-04:
@NDL200201  It's good to hear that you can see the depth pointcloud now, even if the color data is not applied to it.

The pointcloud should be using aligned depth-color data without having to enable alignment, but let's check whether enabling alignment in the launch instruction makes a difference to the pointcloud.

`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true align_depth.enable:=true`

NDL200201, 2023-10-05:
[image]
Thank you for responding to my message. I tried to change my Node, but it doesn't work. Could you explain it for me? This is my Node to run camera D435 Node(
            package='realsense2_camera',
            executable='realsense2_camera_node',
            output='screen',
            parameters=[
                {'depth_module.profile': '640x480x15'},
                {'rgb_camera.profile': '640x480x15'},
                # {'enable_color': True},
                # {'enable_depth': True},
                {'align_depth.enable': True},
                {'pointcloud.enable': True}
            ]
        )

MartyG-RealSense, 2023-10-05:
You could also launch the realsense2_camera package with a **ros2 run** command instead of using your node:

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p align_depth.enable:=true -p pointcloud.enable:=true`

thuanphat1501, 2023-11-13:
hello, when I launch the camera with this command "ros2 launch realsense2_Camera rs_launch.py" on Raspberry Pi 4B, I get HARDWARE_CLOCK problem as shown in the picture, can you help me fix this `problem?
`[INFO] [1699864510.303900484] [camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video4 was found.
[INFO] [1699864510.303976706] [camera]: Device with name Intel RealSense D435 was found.
[INFO] [1699864510.304803651] [camera]: Device with port number 2-1 was found.
[INFO] [1699864510.304869132] [camera]: Device USB type: 3.2
[INFO] [1699864510.307976521] [camera]: getParameters...
[INFO] [1699864510.309027502] [camera]: JSON file is not provided
[INFO] [1699864510.309119576] [camera]: Device Name: Intel RealSense D435
[INFO] [1699864510.309168946] [camera]: Device Serial No: 944622072446
[INFO] [1699864510.309211632] [camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video4
[INFO] [1699864510.309254021] [camera]: Device FW version: 5.15.1
[INFO] [1699864510.309313835] [camera]: Device Product ID: 0x0B07
[INFO] [1699864510.309358224] [camera]: Sync Mode: Off
[WARN] [1699864510.361320460] [camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[INFO] [1699864510.415643603] [camera]: Stopping Sensor: Depth Module
[INFO] [1699864510.429841194] [camera]: Starting Sensor: Depth Module
 13/11 15:35:15,660 ERROR [281472772073696] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[INFO] [1699864515.661581430] [camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1699864515.662003393] [camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1699864515.662574060] [camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[INFO] [1699864515.666585874] [camera]: Stopping Sensor: RGB Camera
[WARN] [1699864515.672522781] [camera]: 
[WARN] [1699864515.672844077] [camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.`

MartyG-RealSense, 2023-11-13:
Hi @thuanphat1501  Unless performance is being negatively affected then you can ignore the hardware clock warning

thuanphat1501, 2023-11-13:
But nothing shows up on rviz2 and then I have some notifications 
```
[realsense2_camera_node-1]  13/11 17:01:20,652 ERROR [281473349183712] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1]  13/11 17:01:20,652 ERROR [281473349183712] (error-handling.cpp:96) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1]  13/11 17:01:25,771 ERROR [281473577314528] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  13/11 17:01:25,772 ERROR [281473577314528] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  13/11 17:01:30,892 ERROR [281473349183712] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1]  13/11 17:01:30,892 ERROR [281473349183712] (error-handling.cpp:96) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1]  13/11 17:01:36,011 ERROR [281473585768672] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1] [ERROR] [1699869696.012923827] [camera.camera]: ./src/rs_node_setup.cpp:370:An exception has been thrown: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
```

MartyG-RealSense, 2023-11-13:
**xioctl(UVCIOC_CTRL_QUERY** messages indicate that there is likely a conflict between the librealsense SDK and the computer's Linux kernel.  Which Linux version and kernel version are you using, please?

Also, I note that your camera firmware driver version is **5.15.1.0**, which is designed for use only with librealsense version **2.54.2**.  Which librealsense version are you using?

thuanphat1501, 2023-11-15:
My raspberry pi using librealsense version 2.54.2 got the error as I said. I tried downloading version 2.54.1 but the error still appeared. raspberry pi kernel version is 5.15.0 can you help me fix it?

MartyG-RealSense, 2023-11-15:
When installing librealsense, did you compile it from source code with CMake?  If you did then did you include in the CMake build instruction the flag **-DFORCE_RSUSB_BACKEND=TRUE** to build librealsense in RSUSB mode to bypass the kernel so that librealsense is not dependent on it?  RSUSB = true is usually the best way to compile librealsense on Raspberry Pi.

thuanphat1501, 2023-11-16:
I build librealsense package using this command on Raspberry Pi
`cmake ../ -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false -DFORCE_RSUSB_BACKEND=ON -DBUILD_WITH_CUDA;bool=false `

MartyG-RealSense, 2023-11-16:
Is performance still bad if you reset the camera at launch with the **initial_reset** command?

`ros2 launch realsense2_camera rs_launch.py initial_reset:=true`

thuanphat1501, 2023-11-17:
when I run your command. performance is still bad and this is the message on my terminal
[image]

MartyG-RealSense, 2023-11-17:
In https://github.com/IntelRealSense/realsense-ros/issues/2545#issuecomment-1812005271 you said that you were using SDK 2.54.2 and then downloaded 2.54.1.  If you have more than one librealsense version installed on the same computer then it is not recommended to do so due to the possibility of the two versions conflicting with each other.

Sometimes the ROS wrapper will start working normally after everything else tried has failed if the computer is wiped and everything (including Ubuntu) is installed again fresh from the beginning, making sure to only install librealsense once.

thuanphat1501, 2023-11-17:
I reinstalled Ubuntu 22.04 and ROS2 Humble. Should I use SDK 2.54.1 or 2.54.2 and choose to install with installation.md or libuvc_installation.md for my Raspberry Pi?

MartyG-RealSense, 2023-11-17:
There is a ROS2 wrapper for 2.54.1 but not for 2.54.2 (which was a minor SDK release focused only on bug fixes).  So 2.54.2 should be fine to use with the current 4.54.1 ROS2 wrapper.

Raspberry Pi works best with a libuvc installation of librealsense.