realsenseai/realsense-ros #2546: How to record a depth video with D415

Aziz1565, 2022-11-15:
I'm looking to record a depth video of an object where I want to measure how this depth will vary with time, hence I need to record a video for a couple of minutes while having the depth information. Not sure if that is done through python or from the software itself.

Your help is greatly appreciated 

Regards,
Abdulaziz

MartyG-RealSense, 2022-11-15:
Hi @Aziz1565  You could use the RealSense SDK's pre-made **RealSense Viewer** tool's 'Record' button to record a video-like sequence to a **bag** format file and play it back in the Viewer.  A bag file is like a video recording of camera data.

[image]

Aziz1565, 2022-11-15:
> Hi @Aziz1565 You could use the RealSense SDK's pre-made **RealSense Viewer** tool's 'Record' button to record a video-like sequence to a **bag** format file and play it back in the Viewer or with a program script. A bag file is like a video recording of camera data.
> 
> [image]

Thank you for your reply.
Yes but in that case, is it possible to get the value of the depth? as I can't extract them manually since it will be a 2 min video with multiple variation in the depth of the object.

MartyG-RealSense, 2022-11-15:
If you move the mouse cursor over an area of the depth image then the depth value in meters of the coordinate that the mouse is hovered over will be displayed in the bottom corner of the Viewer window.  The depth value updates in real-time as the bag plays.

[image]

MartyG-RealSense, 2022-11-15:
It's no trouble at all.  You could convert the bag file to a textual **csv** file (a format compatible with databases and spreadsheets) using the SDK's **rs-convert** bag frame extraction tool.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

Aziz1565, 2022-11-15:
> It's no trouble at all. You could convert the bag file to a textual **csv** file (a format compatible with databases and spreadsheets) using the SDK's **rs-convert** bag frame extraction tool.
> 
> https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

Thank you so much for your help.

Is there a video that can help me in understanding how to work with these tools? 

Again, Thank you for your help

MartyG-RealSense, 2022-11-15:
You are very welcome!

A wide range of videos are available at the official RealSense YouTube channel.

https://www.youtube.com/c/IntelRealSense/videos

On that channel, an introductory video to the RealSense Viewer tool is here:

https://www.youtube.com/watch?v=BVM1wxE9x0E

Aziz1565, 2022-11-15:
> You are very welcome!
> 
> A wide range of videos are available at the official RealSense YouTube channel.
> 
> https://www.youtube.com/c/IntelRealSense/videos
> 
> On that channel, an introductory video for the RealSense Viewer tool is here:
> 
> https://www.youtube.com/watch?v=BVM1wxE9x0E

Thank you. I'll check it out

closed 2022-11-15 by Aziz1565