realsenseai/realsense-ros #2547: Manual Depth Alignment vs On-Chip

geoeo, 2022-11-16:
Hello,

I have a question regarding the depth alignment and the extrinsics of the relasense.
When I set the flag `align_depth` the published depth map on the topic `/camera/aligned_depth_to_color/image_raw` is aligned with the rgb cam; great!

When I try to align the rgb cam with the topic "/camera/depth/image_rect_raw' the results are nowhere near the on chip alignment. The extrinsics I use are the ones published by the realsense.

In general I reproject the pixels of the `/camera/depth/image_rect_raw` image and then apply the depth_to_color transform published by the realsense node. 

Are there any other steps happening with the on-chip alginment?

MartyG-RealSense, 2022-11-16:
Hi @geoeo  Alignment is not performed 'on-chip' on the camera hardware but is instead processed in the computer's CPU.  You may be thinking of **on-chip calibration**, which is not related to depth-color alignment.

If you perform a manual depth alignment instead of making use of the RealSense SDK's **align processing block** (which happens automatically when using align_depth) then the alignment will not be taking account of differences in resolution or in the sizes of the depth and color field of view (FOV).  

For example, on the D435 / D435i camera models the color sensor FOV size is smaller than the depth FOV size.  When performing depth to color alignment with align_depth = true, the align processing block automatically resizes the depth FOV to match the color FOV size.

The **Overview** section of the documentation for the 'rs-align' example program provides information about adjustments that the align processing block makes.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align#overview

geoeo, 2022-11-16:
Thanks for that. What about the D455 FoV. I cant find any mention of the FoV in https://www.intelrealsense.com/wp-content/uploads/2020/06/Intel-RealSense-D400-Series-Datasheet-June-2020.pdf

geoeo, 2022-11-16:
Regardless I checked out the code and its definitely the FoV mapping missing on my end. Thanks @MartyG-RealSense

closed 2022-11-16 by geoeo

MartyG-RealSense, 2022-11-16:
Thnks very much @geoeo for the update!

D455 depth sensor FOV size is 87° × 58° and RGB sensor FOV is 90° × 65°

You can compare the FOVs of different RealSense camera models side by side at the link below.

https://www.intelrealsense.com/compare-depth-cameras/