realsenseai/realsense-ros #2549: ROS Noetic multi-camera set up not working

cambel, 2022-11-17:
Hi!

We have a robotic system with 4x Intel `RealSense SR305 cameras`. 
Previously, in `ROS Melodic`, loading all the cameras worked fine. However, since updating the system to `ROS Noetic`, it does not work at all. Even trying to load one camera doesn't work all the time.

Our current set up:
The system run on a Docker container with Ubuntu 20.04 (kernel 5.15.0-53-generic).
librealsense v2.51.1

Running `realsense-viewer` recognizes all the cameras and can connect to them simultaneously.
But connecting through realsense-ros does not work.

## Log when trying to load 2 cameras:
``` shell
SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 0
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -1.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/confidence_fps: 30
 * /camera1/realsense2_camera/confidence_height: 480
 * /camera1/realsense2_camera/confidence_width: 640
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_confidence: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_infra: False
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_pose: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters: 
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: 30
 * /camera1/realsense2_camera/fisheye_frame_id: camera1_fisheye_f...
 * /camera1/realsense2_camera/fisheye_height: 0
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera1_fisheye_o...
 * /camera1/realsense2_camera/fisheye_width: 0
[... 199 lines cut ...]
[ INFO] [1668666782.870953467]: num_filters: 0
[ INFO] [1668666782.870964306]: Setting Dynamic reconfig parameters.
[FATAL] [1668666791.343507512]: ASSERTION FAILED
	file = /opt/ros/noetic/include/ros/publisher.h
	line = 107
	cond = false
	message = 
[FATAL] [1668666791.343534076]: Call to publish() on an invalid Publisher
[FATAL] [1668666791.343539795]: 

[camera1/realsense2_camera_manager-1] process has died [pid 165827, exit code -5, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/root/.ros/log/4836ea42-6631-11ed-8025-88aedd2295bc/camera1-realsense2_camera_manager-1.log].
log file: /root/.ros/log/4836ea42-6631-11ed-8025-88aedd2295bc/camera1-realsense2_camera_manager-1*.log
[FATAL] [1668666792.152469642]: ASSERTION FAILED
	file = /opt/ros/noetic/include/ros/publisher.h
	line = 107
	cond = false
	message = 
[FATAL] [1668666792.152494745]: Call to publish() on an invalid Publisher
[FATAL] [1668666792.152500613]: 

[camera2/realsense2_camera_manager-3] process has died [pid 165829, exit code -5, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/root/.ros/log/4836ea42-6631-11ed-8025-88aedd2295bc/camera2-realsense2_camera_manager-3.log].
log file: /root/.ros/log/4836ea42-6631-11ed-8025-88aedd2295bc/camera2-realsense2_camera_manager-3*.log
```
## Note
Launching `realsense-ros` does not only fails, but somehow it breaks the connection to the cameras. Before launching `realsense-ros`, realsense-viewer works fine. After failing to launch `realsense-ros`, realsense-viewer fails with a `Segmentation fault`.

MartyG-RealSense, 2022-11-17:
Hi @cambel  May I ask the following questions, please.

1.  Are you able to launch the cameras in RealSense ROS if the RealSense Viewer is not active (the Viewer is not required in order to use RealSense cameras with RealSense ROS).

2.  Have you tried rebuilding the ROS wrapper after the change from Melodic to Noetic?

3.  Are the cameras able to launch if you use **rsmultiple_devices.launch**

`roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=636201000324 serial_no_camera2:=636201001178
`

cambel, 2022-11-17:
Hi @MartyG-RealSense 

1. Yes, but not always. Half of the attempts to launch the cameras failed with a very similar log. Sometimes it does connect.
2. Yes, the ROS wrapper has been build from scratch. Not mixing anything with the Melodic version.
3. No, it still does not work. Actually, the log I posted above is from launching `rs_multiple_devices.launch`

MartyG-RealSense, 2022-11-17:
Thank you very much.  How does it behave if you add **initial_reset:=true** onto the end of the multiple_devices.launch instruction to reset the cameras at launch?

`roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=636201000324 serial_no_camera2:=636201001178 initial_reset:=true`

cambel, 2022-11-17:
Thank you for the quick responses.

Adding the `initial_reset=:true` makes one of the cameras to successfully connect while the other fails.

```shell
[ INFO] [1668671610.854048227]: Initializing nodelet with 20 worker threads.
[ INFO] [1668671610.864040018]: Initializing nodelet with 20 worker threads.
[ INFO] [1668671610.914670944]: RealSense ROS v2.3.2
[ INFO] [1668671610.914692474]: Built with LibRealSense v2.50.0
[ INFO] [1668671610.914700189]: Running with LibRealSense v2.50.0
[ INFO] [1668671610.923913452]: RealSense ROS v2.3.2
[ INFO] [1668671610.923931695]: Built with LibRealSense v2.50.0
[ INFO] [1668671610.923938234]: Running with LibRealSense v2.50.0
[ INFO] [1668671610.933404994]:  
[ INFO] [1668671611.029780582]:  
 17/11 07:53:31,109 ERROR [140658769655552] (handle-libusb.h:100) failed to claim usb interface: 2, error: RS2_USB_STATUS_BUSY
[ WARN] [1668671611.111772713]: Device 1/2 failed with exception: failed to set power state
 17/11 07:53:31,110 ERROR [140658994902784] (sensor.cpp:572) acquire_power failed: failed to set power state
 17/11 07:53:31,111 WARNING [140658994902784] (rs.cpp:310) null pointer passed for argument "device"
[ INFO] [1668671611.261859993]: Device with serial number 636201000324 was found.

[ INFO] [1668671611.261895936]: Device with physical ID 2-8-5 was found.
[ INFO] [1668671611.261910932]: Device with name Intel RealSense SR305 was found.
[ INFO] [1668671611.262244180]: Device with port number 2-8 was found.
[ INFO] [1668671611.262275721]: Resetting device...
[ INFO] [1668671611.494588614]: Device with serial number 636201001178 was found.

[ INFO] [1668671611.494611149]: Device with physical ID 2-9-6 was found.
[ INFO] [1668671611.494616807]: Device with name Intel RealSense SR305 was found.
[ INFO] [1668671611.494823768]: Device with port number 2-9 was found.
[ INFO] [1668671611.494836808]: Resetting device...
[ INFO] [1668671617.424443369]:  
[ INFO] [1668671617.707586377]:  
[ INFO] [1668671617.859418966]: Device with serial number 636201000324 was found.

[ INFO] [1668671617.859505625]: Device with physical ID 2-8-7 was found.
[ INFO] [1668671617.859549530]: Device with name Intel RealSense SR305 was found.
[ INFO] [1668671617.860522120]: Device with port number 2-8 was found.
[ INFO] [1668671617.863932940]: getParameters...
[ INFO] [1668671617.902863632]: setupDevice...
[ INFO] [1668671617.902877705]: JSON file is not provided
[ INFO] [1668671617.902882874]: ROS Node Namespace: camera1
[ INFO] [1668671617.902888777]: Device Name: Intel RealSense SR305
[ INFO] [1668671617.902893272]: Device Serial No: 636201000324
[ INFO] [1668671617.902897638]: Device physical port: 2-8-7
[ INFO] [1668671617.902901745]: Device FW version: 3.27.3.0
[ INFO] [1668671617.902905951]: Device Product ID: 0x0B48
[ INFO] [1668671617.902909997]: Enable PointCloud: Off
[ INFO] [1668671617.902914207]: Align Depth: Off
[ INFO] [1668671617.902918214]: Sync Mode: Off
[ INFO] [1668671617.902935757]: Device Sensors: 
[ERROR] [1668671617.934531469]: An exception has been thrown: failed to set power state
[ERROR] [1668671617.934617594]: Exception: failed to set power state
 17/11 07:53:37,933 ERROR [140381291255552] (handle-libusb.h:100) failed to claim usb interface: 2, error: RS2_USB_STATUS_BUSY
 17/11 07:53:37,934 ERROR [140381567657728] (sensor.cpp:572) acquire_power failed: failed to set power state
 17/11 07:53:37,934 WARNING [140381567657728] (rs.cpp:376) null pointer passed for argument "list"
[ INFO] [1668671618.055536111]: Device with serial number 636201000324 was found.

[ INFO] [1668671618.055590179]: Device with physical ID 2-8-7 was found.
[ INFO] [1668671618.055605743]: Device with name Intel RealSense SR305 was found.
[ INFO] [1668671618.056082161]: Device with port number 2-8 was found.
[ INFO] [1668671618.400845399]: Device with serial number 636201001178 was found.

[ INFO] [1668671618.400891191]: Device with physical ID 2-9-8 was found.
[ INFO] [1668671618.400903954]: Device with name Intel RealSense SR305 was found.
[ INFO] [1668671618.401285657]: Device with port number 2-9 was found.
[ INFO] [1668671618.403261282]: getParameters...
[ INFO] [1668671618.435159717]: setupDevice...
[ INFO] [1668671618.435188705]: JSON file is not provided
[ INFO] [1668671618.435202717]: ROS Node Namespace: camera2
[... 3 lines cut ...]
[ INFO] [1668671618.435254353]: Device FW version: 3.27.3.0
[ INFO] [1668671618.435268878]: Device Product ID: 0x0B48
[ INFO] [1668671618.435281306]: Enable PointCloud: Off
[ INFO] [1668671618.435293363]: Align Depth: Off
[ INFO] [1668671618.435304964]: Sync Mode: Off
[ INFO] [1668671618.435340453]: Device Sensors: 
[ INFO] [1668671618.505945962]: Coded-Light Depth Sensor was found.
[ INFO] [1668671618.599632804]: RGB Camera was found.
[ INFO] [1668671618.599727473]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1668671618.599776458]: num_filters: 0
[ INFO] [1668671618.599800742]: Setting Dynamic reconfig parameters.
[ INFO] [1668671625.249311518]: Done Setting Dynamic reconfig parameters.
[ INFO] [1668671625.250409082]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1668671625.251664991]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1668671625.251750714]: setupPublishers...
[ INFO] [1668671625.257370357]: Expected frequency for depth = 30.00000
[ INFO] [1668671625.319157077]: Expected frequency for color = 30.00000
[ INFO] [1668671625.332334992]: setupStreams...
[ INFO] [1668671625.567002905]: SELECTED BASE:Depth, 0
[ INFO] [1668671625.571507555]: RealSense Node Is Up!
[ WARN] [1668671625.857921322]: 
[ WARN] [1668671625.858027482]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[ERROR] [1668671626.068412393]: Incomplete bulk usb transfer!

```

cambel, 2022-11-17:
If I launch them from different terminals.

1. If the other has not finished loading correctly, one or both fails.
2. If I wait for the first one to successfully complete (retrying when necessary), and then launch the second one. It seems to work. Both are running.

The following log is from case (1):
```shell
[ERROR] [1668672217.808873395]: Skipped loading plugin with error: XML Document '/root/o2ac-ur/catkin_ws/src/o2ac_visualization/nodelet_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..
[ INFO] [1668672217.811634376]: Initializing nodelet with 20 worker threads.
[ INFO] [1668672217.945558309]: RealSense ROS v2.3.2
[ INFO] [1668672217.945579972]: Built with LibRealSense v2.50.0
[ INFO] [1668672217.945588181]: Running with LibRealSense v2.50.0
[ INFO] [1668672218.005559526]:  
Import ur_ikfast not available, IKFAST would not be supported without it
[ INFO] [1668672218.485589569]: Device with serial number 636201000324 was found.

[ INFO] [1668672218.485613367]: Device with physical ID 2-8-9 was found.
[ INFO] [1668672218.485619428]: Device with name Intel RealSense SR305 was found.
[ INFO] [1668672218.485820759]: Device with port number 2-8 was found.
[ INFO] [1668672218.486877927]: getParameters...
[ INFO] [1668672218.503124261]: setupDevice...
[ INFO] [1668672218.503137888]: JSON file is not provided
[ INFO] [1668672218.503144455]: ROS Node Namespace: a_bot_inside_camera
[ INFO] [1668672218.503152083]: Device Name: Intel RealSense SR305
[ INFO] [1668672218.503157590]: Device Serial No: 636201000324
[ INFO] [1668672218.503162977]: Device physical port: 2-8-9
[ INFO] [1668672218.503172123]: Device FW version: 3.27.3.0
[ INFO] [1668672218.503177760]: Device Product ID: 0x0B48
[ INFO] [1668672218.503184064]: Enable PointCloud: On
[ INFO] [1668672218.503189742]: Align Depth: On
[ INFO] [1668672218.503195384]: Sync Mode: On
[ INFO] [1668672218.503217137]: Device Sensors: 
[ INFO] [1668672218.685979913]: Coded-Light Depth Sensor was found.
[ INFO] [1668672218.921357032]: RGB Camera was found.
[ INFO] [1668672218.921463663]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1668672218.922245387]: Add Filter: pointcloud
[ INFO] [1668672218.923589483]: num_filters: 2
[ INFO] [1668672218.923632356]: Setting Dynamic reconfig parameters.
[FATAL] [1668672227.259875260]: ASSERTION FAILED
	file = /opt/ros/noetic/include/ros/publisher.h
	line = 107
	cond = false
	message = 
[FATAL] [1668672227.259968713]: Call to publish() on an invalid Publisher
[FATAL] [1668672227.260001572]: 

[a_bot_inside_camera/realsense2_camera_manager-1] process has died [pid 258080, exit code -5, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/root/.ros/log/566b31c8-664e-11ed-8514-88aedd2295bc/a_bot_inside_camera-realsense2_camera_manager-1.log].
log file: /root/.ros/log/566b31c8-664e-11ed-8514-88aedd2295bc/a_bot_inside_camera-realsense2_camera_manager-1*.log
```

MartyG-RealSense, 2022-11-17:
In past cases where two cameras have been launched at the same time and one succeeds but the other fails, a solution has been to set up a 'sleep' delay on one of the cameras so that it launches after the first one.  An example of such a mechanism is **timed_roslaunch**

http://wiki.ros.org/timed_roslaunch

The solution that was implemented in the RealSense ROS wrapper for this issue was to create an instruction called **usb_port_id** that allows a camera to be identified by the ID of the port that it is plugged into instead of the camera serial number.  usb_port_id: will attach to the device with the given USB port (usb_port_id). i.e 4-1, 4-2 etc.

cambel, 2022-11-17:
I tried both approaches.
1. Using a delayed launch with `timed_roslaunch` seems to work fine.
2. Setting the specific `usb_port_id` for each camera does not seem to work either. 

In any case, I can keep working with the method (1). 
Thank you!

MartyG-RealSense, 2022-11-17:
You are very welcome.  It's great that timed_roslaunch worked for you.  :)

closed 2022-11-17 by cambel