realsenseai/realsense-ros #2552: [ROS2] Weird bug when enable pointcloud

gachiemchiep, 2022-11-18:
Hello RealSensen team
@MartyG-RealSense 

I switch into ROS2 and a weird bug happen.
1. When enable pointcloud, the color and depth images are blurred. Look like it was processed by some filter
[image]

2. When disable pointcloud, the color and depth images are fine. 
[image]

Maybe some pointer using for creating pointcloud are messing up. 
Would you mind take a look at this and show me how to fix this problem.

Thank you.

MartyG-RealSense, 2022-11-18:
Hi @gachiemchiep  What computing device / computer are you using, please (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc).  Thanks!

gachiemchiep, 2022-11-19:
@MartyG-RealSense 
Sorry. realsense is totally fine,  It is a bug in rviz. 
I stop displaying pointcloud then everything become normal again.

closed 2022-11-19 by gachiemchiep

MartyG-RealSense, 2022-11-19:
It's no trouble at all, @gachiemchiep - thanks very much for the update!