realsenseai/realsense-ros #2553: L500 series with realsense-ros(ros2-development)

huangjiancong1, 2022-11-18:
How can I used L515 sensor with the branch: ros2-development?

I have installed:
librealsense-2.51.1

But when I launch this: ` ros2 launch realsense2_camera rs_launch.py`
I get this warning: 

> [INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [21253]
[realsense2_camera_node-1] 1668741941.935035 [12] realsense2: using network interface enp3s0 (udp/10.11.10.14) selected arbitrarily from: enp3s0, docker0
[realsense2_camera_node-1] [INFO] [1668741941.947049821] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1668741941.947178717] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1668741941.947196619] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1668741942.231081339] [camera.camera]: Device with serial number f1150576 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1668741942.231340762] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1668741942.231471905] [camera.camera]: Device with name Intel RealSense L515 was found.
[realsense2_camera_node-1] [INFO] [1668741942.233007897] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1668741942.233191402] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1668741942.236220182] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1668741942.237305166] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1668741942.237326935] [camera.camera]: Device Name: Intel RealSense L515
[realsense2_camera_node-1] [INFO] [1668741942.237339608] [camera.camera]: Device Serial No: f1150576
[realsense2_camera_node-1] [INFO] [1668741942.237371372] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1668741942.237405493] [camera.camera]: Device FW version: 01.05.08.01
[realsense2_camera_node-1] [INFO] [1668741942.237423882] [camera.camera]: Device Product ID: 0x0B64
[realsense2_camera_node-1] [INFO] [1668741942.237437176] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1668741942.244132721] [camera.camera]: Set ROS param depth_module.profile to default: 640x480x30
[realsense2_camera_node-1]  18/11 11:25:42,243 WARNING [140421741147904] (backend-v4l2.cpp:1444) Pixel format 20202043-0000-0010-8000-00aa003 likely requires patch for fourcc code C   !
[realsense2_camera_node-1] [INFO] [1668741942.248579500] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1668741942.249167242] [camera.camera]: Set ROS param gyro_fps to default: 100
[realsense2_camera_node-1] [INFO] [1668741942.249388010] [camera.camera]: Set ROS param accel_fps to default: 200
[realsense2_camera_node-1] [WARN] [1668741942.256687173] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1668741942.257977053] [camera.camera]: Stopping Sensor: L500 Depth Sensor
[realsense2_camera_node-1] [INFO] [1668741942.265163693] [camera.camera]: Starting Sensor: L500 Depth Sensor
[realsense2_camera_node-1] [ERROR] [1668741942.268284395] [camera.camera]: /home/bot/WS/realsense2_camera/src/rs_node_setup.cpp:344:An exception has been thrown: The requested pixel format 'C   ' is not natively supported by the running Linux kernel and likely requires a patch Last Error: Invalid argument
[realsense2_camera_node-1] Failed to start device: The requested pixel format 'C   ' is not natively supported by the running Linux kernel and likely requires a patch Last Error: Invalid argument
[realsense2_camera_node-1]  18/11 11:25:42,267 WARNING [140421741147904] (backend-v4l2.cpp:1444) Pixel format 20202043-0000-0010-8000-00aa003 likely requires patch for fourcc code C   !
[realsense2_camera_node-1]  18/11 11:25:42,268 ERROR [140421741147904] (librealsense-exception.h:52) The requested pixel format 'C   ' is not natively supported by the running Linux kernel and likely requires a patch Last Error: Invalid argument
[realsense2_camera_node-1] [INFO] [1668741942.268450648] [camera.camera]: Stopping Sensor: L500 Depth Sensor
[realsense2_camera_node-1] [INFO] [1668741942.272141296] [camera.camera]: Starting Sensor: L500 Depth Sensor
[realsense2_camera_node-1]  18/11 11:25:42,268 ERROR [140421741147904] (l500-depth.cpp:574) Exception caught in l500_depth_sensor::open
[realsense2_camera_node-1] [ERROR] [1668741942.274945858] [camera.camera]: /home/bot/WS/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:344:An exception has been thrown: The requested pixel format 'C   ' is not natively supported by the running Linux kernel and likely requires a patch Last Error: Invalid argument
[realsense2_camera_node-1] [ERROR] [1668741942.274999931] [camera.camera]: Error updating the sensors: The requested pixel format 'C   ' is not natively supported by the running Linux kernel and likely requires a patch Last Error: Invalid argument

huangjiancong1, 2022-11-18:
ros-development branch is working with using D435

MartyG-RealSense, 2022-11-18:
Hi @huangjiancong1  Could you first try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing this make any difference?

huangjiancong1, 2022-11-21:
> Hi @huangjiancong1 Could you first try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera, [...]

Hi @MartyG-RealSense , thank you for your reply.
I reversed the micro-sized end of the USB cable, and it still the `[ERROR]` as show above.

I also try to re-install `ros2-development` branch in a novel linux environment. The same issue was occured.

MartyG-RealSense, 2022-11-22:
The logs contain repeated references to the warning **likely requires patch for fourcc code C**.  A fourcc code warning typically occurs if the librealsense SDK has been built from source code but the kernel has not had a patch applied to it before librealsense was built.

Did you install librealsense by building it from source code?  And if you did, was a kernel patch applied by running a patch script?

huangjiancong1, 2022-11-23:
> The logs contain repeated references to the warning **likely requires patch for fourcc code C**. A fourcc code warning typically occurs if the librealsense SDK has been built from source code but [...]

Hi, @MartyG-RealSense . Thank you for your reply again.

The librealsense SDK was built from source and I wihtout running a patch script.

There are many patches that I can run. It is great that if you can tell me which kernel patch that I can run?

The system I am using is:
Linux 4.19.0-22-amd64 x86_64

MartyG-RealSense, 2022-11-23:
The patch script for amd64 computers with Ubuntu 14, 16, 18 or 20 is **./scripts/patch-realsense-ubuntu-lts.sh**

If you are using Debian instead of Ubuntu then I recommend building the SDK from source code with the **RSUSB backend** installation method.  An RSUSB build bypasses the kernel and so is not dependent on Linux versions of kernel versions and does not require kernel patching.  Instructions can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

MartyG-RealSense, 2022-11-29:
Hi @huangjiancong1  Do you require further assistance with this case, please?  Thanks!

huangjiancong1, 2022-12-01:
> Hi @huangjiancong1 Do you require further assistance with this case, please? Thanks!

Hello @MartyG-RealSense.
Thank you for your outstanding technical assistance!
You can close this issue now.

MartyG-RealSense, 2022-12-01:
You are very welcome  :)

As you suggest, I will close the issue as you are are satisfied.  Thanks again!

closed 2022-12-01 by MartyG-RealSense