realsenseai/realsense-ros #2554: tf frames not being published in a multi camera setup with D415 cameras

Pran-Seven, 2022-11-18:
Hello,
I am using a dual d415 setup and Ihave tried both of the following while using an external manager for the nodelet and not realsense manager (though I vaguely remember this issue popping up occasionally with realsense manager as well).

**ISSUE:** 
The tf frames of one of the cameras are not being published and the behaviour is random- sometimes camera 1 is being published sometimes camera 2.

**SETUP:**
1. Using a modified version of dual_cameras.launch 
Launching both cameras using multiple devices with changes to a account for processing similar to rgbd.launch for both cameras

2. Launching both cameras seperately using two instances of rs_rgbd.launch

In both the cases, frames for only one of the cameras is being published. Is this a common issue or what could be possibly causing this and how to fix it? Thank you!

MartyG-RealSense, 2022-11-18:
Hi @Pran-Seven  Is the two-camera external manager system the same one that you were using in your earlier case at https://github.com/IntelRealSense/realsense-ros/issues/2466 please?

Would it be possible to set a delay for one of the two launches using the method described at https://github.com/IntelRealSense/realsense-ros/issues/2549#issuecomment-1318258399 to avoid a conflict between the cameras when launching them simultaneously?

Pran-Seven, 2022-11-18:
Yup, the same system.

Will try adding a delay and update, thank you!
But the same problem exists when I launch using multiple_cameras.launch, so I am not sure if the delay would make a difference.

Pran-Seven, 2022-11-20:
I did try adding a delay but no changes to the result. It still publishes only one set of frames.

MartyG-RealSense, 2022-11-21:
It is difficult to diagnose an external manager case as https://github.com/IntelRealSense/realsense-ros/issues/1069 is the only other case except for your own https://github.com/IntelRealSense/realsense-ros/issues/2466 where an external manager was used with RealSense ROS.  The RealSense user in https://github.com/IntelRealSense/realsense-ros/issues/1069 had no data being published.

I note though that you have also tried launching without the external manager and were not successful in streaming both cameras.   From your description "sometimes camera 1 is being published sometimes camera 2", it does sound like whichever camera that is first to be successfully activated blocks the second camera.

Pran-Seven, 2022-11-21:
Hey, I am able to get the data and see both the pointclouds on rviz, the issue is that only one set of tf cameras is being published. Yes that does seem to be the case because the issue is independent of the order of the cameras in the lunch file.

MartyG-RealSense, 2022-11-22:
There was a similar case in 2016 with the original RealSense ROS wrapper at https://github.com/IntelRealSense/realsense-ros/issues/120 where two cameras were being used but only one of the camera echoed TFs.

The solution reported in that particular case at https://github.com/IntelRealSense/realsense-ros/issues/120#issuecomment-269029732 was to enable Dynamic Transforms by fixing a bug in the realsense_camera wrapper with a PR at https://github.com/IntelRealSense/realsense-ros/pull/169

It is not clear how that solution could be applied to the modern **realsense2_camera** wrapper though.

Pran-Seven, 2022-11-22:
Thank you for the links, I was able to go through them and yes like you mention I don't see how that will fix the issue. Unfortunately with my limited knowledge I don't think I will be able to dig deeper and identify what is causing this. Thank you for your help!

MartyG-RealSense, 2022-11-23:
Let's go back to the beginning of the case and re-review the details.

You said that you had tried launching with two separate instances of **rs_rgbd.launch** and only one camera was publishing frames.

1.  Did you launch the two cameras in separate ROS terminals?

2.  Have you tried launching with **rs_camera.launch** instead of rs_rgbd.launch as suggested in the ROS wrapper documentation?

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#work-with-multiple-cameras

Pran-Seven, 2022-11-24:
Hey, yeah sure.

1. I am not launching it via terminals, rather from a launch file of the robot's perception pipeline, where I specified two separate instances of cameras and passed the names and serial numbers accordingly.

2. Yes, earlier we were using rs_camera.launch and even the problem would occur sporadically.

MartyG-RealSense, 2022-11-24:
I am not familiar with using the perception pipeline, unfortunately.  The best resource that I could find regarding using multiple sensors with a perception pipeline was the one at the link below that makes use of **MoveIt!**

http://docs.ros.org/en/kinetic/api/moveit_tutorials/html/doc/perception_pipeline/perception_pipeline_tutorial.html

Pran-Seven, 2022-11-25:
Oh it is not a common pipeline, just something we created for the tasks we needed. But I can provide my launch file, if necessary for analysing, if it would help.

MartyG-RealSense, 2022-11-25:
Yes, please do provide your launch file.  Thanks!

Pran-Seven, 2022-11-25:
```
<launch>
 <arg name="realsense_enabled" value="$(env SEVEN_REALSENSE_ENABLED)"/>
 <node pkg="nodelet" type="nodelet" name="mpsManager" args="manager" output="screen"/>

  <group  if="$(arg realsense_enabled)">
    <include file="$(find realsense2_camera)/launch/rs_multiple_devices.launch">
      <group ns="$(arg camera1)">
        <arg name="serial_no_camera1"    			value="746112061710"/>
	      <arg name="camera1"              			value="sensor_d415_right"/>
        <arg name="initial_reset"             value="true"/>
        <arg name="align_depth"               value="true"/>
      </group>
      <group ns="$(arg camera2)">
        <arg name="serial_no_camera2"    			value="746112060570"/>
        <arg name="camera2"              			value="sensor_d415_left"/>
        <arg name="initial_reset"             value="true"/>
        <arg name="align_depth"               value="true"/>
      </group>
    </include>
  </group> 
  <!-- <group  if="$(arg realsense_enabled)">
    <include file="$(find realsense2_camera)/launch/rs_multiple_devices.launch">
        <arg name="serial_no_camera1"    			value="802212060737"/>
	      <arg name="camera1"              			value="sensor_d415_right"/>
        <arg name="initial_reset"             value="true"/>
        <arg name="serial_no_camera2"    			value="746112061094"/>
        <arg name="camera2"              			value="sensor_d415_left"/>
    </include>
  </group> -->

  <!-- <group if="$(arg realsense_enabled)">
	 <include file="$(find realsense2_camera)/launch/rs_rgbd.launch">
	   <arg name="camera" value="sensor_d415_left"/>
	   <arg name="serial_no" value="746112060570"/>
	   <arg name="enable_pointcloud" value="true"/>
	   <arg name="enable_depth" value="true"/>
	   <arg name="depth_height" value="480"/>
	   <arg name="depth_width" value="640"/>
	   <arg name="color_width" value="640"/>
	   <arg name="color_height" value="480"/>
     <arg name="initial_reset"             value="true"/>
     <arg name="external_manager"    value="true"/>
     <arg name="manager"             value="/mpsManager"/>
     <arg name="depth_fps"           default="30"/>
     <arg name="color_fps"           default="30"/>
	 </include>
	 <include file="$(find realsense2_camera)/launch/rs_rgbd.launch">
	   <arg name="camera" value="sensor_d415_right"/>
	   <arg name="serial_no" value=" 746112061710"/>
	   <arg name="enable_pointcloud" value="true"/>
	   <arg name="enable_depth" value="true"/>
	   <arg name="depth_height" value="640"/>
	   <arg name="depth_width" value="480"/>
	   <arg name="color_width" value="640"/>
	   <arg name="color_height" value="480"/>
     <arg name="initial_reset"             value="true"/>
     <arg name="external_manager"    value="true"/>
     <arg name="manager"             value="/mpsManager"/>
     <arg name="depth_fps"           default="30"/>
     <arg name="color_fps"           default="30"/>
	 </include>
	</group> -->

 <include file="launch/description.launch">
   <arg name="simulation" default="false"/>
 </include>

 <node pkg="nodelet" type="nodelet" name="mpsNodelet" args="load OrganizedMultiPlaneSegmentation mpsManager">
  <!-- <param name="inputTopic" value="/$(arg camera1)/depth_registered/points"/>  -->
 </node>
 
 <node pkg="nodelet" type="nodelet" name="PointCloudMergerNodelet" args="load PointCloudMergerNodelet mpsManager"/> 

 <node pkg="nodelet" type="nodelet" name="peopleDetection" args="load peopleDetection mpsManager" output="screen">
  <!-- <remap from="~input" to="$(arg FILTERED_CLOUD)"/> -->
  <!-- <param name="inputTopic" value="/$(arg camera1)/depth_registered/points"/>
  <param name="cameraInfoTopic" value="/$(arg camera1)/aligned_depth_to_color/camera_info"/>
  <param name="groundPlaneTopic" value="/mpsNodelet/groundPlaneCoefficients"/> -->

 </node>
</launch>
```

Pran-Seven, 2022-11-25:
MULTIPLE_DEVICES_LAUNCH
```
<launch>
  <!-- <arg name="serial_no_camera1"    			default="746112061710"/> 			
  <arg name="serial_no_camera2"    			default="746112060570"/> 			  
  <arg name="camera1"              			default="sensor_d415_right"/>		
  <arg name="camera2"              			default="sensor_d415_left"/>		 -->
  <arg name="serial_no_camera1"    			default="746112060570"/> 			
  <arg name="serial_no_camera2"    			default="746112061710"/> 			
  <arg name="camera1"              			default="sensor_d415_right"/>		
  <arg name="camera2"              			default="sensor_d415_left"/>		
  <arg name="tf_prefix_camera1"         default="$(arg camera1)"/>
  <arg name="tf_prefix_camera2"         default="$(arg camera2)"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="reconnect_timeout"         default="6.0"/>

  <group ns="$(arg camera1)">
    <!-- Camera device specific arguments -->
    <arg name="external_manager"    default="true"/>
    <arg name="manager"             default="/mpsManager"/>
    <arg name="usb_port_id"         default=""/>
    <arg name="device_type"         default=""/>
    <arg name="json_file_path"      default=""/>

    <arg name="fisheye_width"       default="-1"/>
    <arg name="fisheye_height"      default="-1"/>
    <arg name="enable_fisheye"      default="false"/>

    <arg name="depth_width"         default="640"/>
    <arg name="depth_height"        default="480"/>
    <arg name="enable_depth"        default="true"/>

    <arg name="infra_width"         default="640"/>
    <arg name="infra_height"        default="480"/>
    <arg name="enable_infra1"       default="true"/>
    <arg name="enable_infra2"       default="true"/>

    <arg name="color_width"         default="640"/>
    <arg name="color_height"        default="480"/>
    <arg name="enable_color"        default="true"/>

    <arg name="fisheye_fps"         default="-1"/>
    <arg name="depth_fps"           default="30"/>
    <arg name="infra_fps"           default="30"/>
    <arg name="color_fps"           default="30"/>
    <arg name="gyro_fps"            default="-1"/>
    <arg name="accel_fps"           default="-1"/>
    <arg name="enable_gyro"         default="false"/>
    <arg name="enable_accel"        default="false"/>

    <arg name="enable_pointcloud"   default="false"/>
    <arg name="enable_sync"         default="true"/>
    <arg name="align_depth"         default="true"/>
    <arg name="filters"             default=""/>

    <arg name="publish_tf"          default="true"/>
    <arg name="tf_publish_rate"     default="0"/>

    <!-- rgbd_launch specific arguments -->

    <!-- Arguments for remapping all device namespaces -->
    <arg name="rgb"                             default="color" />
    <arg name="ir"                              default="infra1" />
    <arg name="depth"                           default="depth" />
    <arg name="depth_registered_pub"            default="depth_registered" />
    <arg name="depth_registered"                default="depth_registered" unless="$(arg align_depth)" />
    <arg name="depth_registered"                default="aligned_depth_to_color" if="$(arg align_depth)" />
    <arg name="depth_registered_filtered"       default="$(arg depth_registered)" />
    <arg name="projector"                       default="projector" />

[... 241 lines cut ...]
      <!-- Publish registered XYZRGB point cloud with software registered input -->
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_sw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg depth_registered_filtered)/sw_registered/image_rect_raw" />
        <remap from="depth_registered/points"     to="$(arg depth_registered)/points" />
      </node>
    </group>

    <group if="$(eval depth_registered_processing and hw_registered_processing)">
      <!-- Publish registered XYZRGB point cloud with hardware registered input (ROS Realsense depth alignment) -->
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_hw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg depth_registered)/image_raw" />
        <remap from="depth_registered/points"     to="$(arg depth_registered_pub)/points" />
      </node>
    </group>

  </group>

</launch>
```

MartyG-RealSense, 2022-11-25:
If a launch is based on rs_rgbd then it requires installation of additional RGBD support that the ROS wrapper does not have by default.  Have you installed the appropriate support from https://github.com/IntelRealSense/librealsense/issues/10223#issuecomment-1152864261 for your particular ROS1 version (Kinetic, Melodic or Noetic)

Pran-Seven, 2022-11-25:
Yes, I have made sure to install the additional support, and I believe the installation is correct because the launch does not give me an error and I ma able to visualise the pointclouds coming from the camera.

MartyG-RealSense, 2022-11-25:
If you have launched both cameras separately using two instances of the original unchanged rs_rgbd launch file and TF has only been published for one camera then it could be concluded that the problem is not with the additions made to your launch file.

If the problem was caused by something within the code of the RealSense ROS1 wrapper then it could not be fixed with an update as development of the ROS1 wrapper has unfortunately ended and the focus of the RealSense ROS developer team is now on the ROS2 wrapper.

Pran-Seven, 2022-11-26:
Oh that is unfortunate to hear. Thank you for help though, really appreciate it!

closed 2022-11-26 by Pran-Seven

MartyG-RealSense, 2022-11-26:
You are very welcome!