realsenseai/realsense-ros #2555: cannot find topic in rviz

shaoxh, 2022-11-18:
Rosrun realsense successfully(roslaunch realsense2_camera rs_camera.launch align_depth:=true) 
and I could find all the topics by "rostopic list". 
But when use rviz, there is no other [topics] but only depth and rgb image. Where are the others? 

The realsense terminal has no error. The outputs are like follow:
[ INFO] [1668775028.937170488]: Initializing nodelet with 16 worker threads.
[ INFO] [1668775029.121886171]: RealSense ROS v2.3.2
[ INFO] [1668775029.121923112]: Built with LibRealSense v2.51.1
[ INFO] [1668775029.121937526]: Running with LibRealSense v2.51.1
[ INFO] [1668775029.146283039]:  
[ INFO] [1668775029.253110688]: Device with serial number 037322251191 was found.

[ INFO] [1668775029.253172373]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1:1.0/video4linux/video2 was found.
[ INFO] [1668775029.253194978]: Device with name Intel RealSense D455 was found.
[ INFO] [1668775029.253846748]: Device with port number 4-1 was found.
[ INFO] [1668775029.253882736]: Device USB type: 3.2
[ INFO] [1668775029.256432386]: getParameters...
[ INFO] [1668775029.291308486]: setupDevice...
[ INFO] [1668775029.291322365]: JSON file is not provided
[ INFO] [1668775029.291327981]: ROS Node Namespace: camera
[ INFO] [1668775029.291335961]: Device Name: Intel RealSense D455
[ INFO] [1668775029.291340374]: Device Serial No: 037322251191
[ INFO] [1668775029.291345320]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1:1.0/video4linux/video2
[ INFO] [1668775029.291349897]: Device FW version: 05.13.00.50
[ INFO] [1668775029.291354461]: Device Product ID: 0x0B5C
[ INFO] [1668775029.291359972]: Enable PointCloud: Off
[ INFO] [1668775029.291365405]: Align Depth: On
[ INFO] [1668775029.291369437]: Sync Mode: On
[ INFO] [1668775029.291396541]: Device Sensors: 
[ INFO] [1668775029.295056386]: Stereo Module was found.
 18/11 20:37:09,295 WARNING [140332796925696] (backend-v4l2.cpp:1444) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ INFO] [1668775029.298396764]: RGB Camera was found.
[ INFO] [1668775029.298490501]: Motion Module was found.
[ INFO] [1668775029.298507641]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1668775029.298684008]: num_filters: 1
[ INFO] [1668775029.298694577]: Setting Dynamic reconfig parameters.
[ INFO] [1668775029.544119074]: Done Setting Dynamic reconfig parameters.
[ INFO] [1668775029.544506506]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1668775029.545178194]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1668775029.545953522]: setupPublishers...
[ INFO] [1668775029.546911775]: Expected frequency for depth = 30.00000
[ INFO] [1668775029.557142872]: Expected frequency for color = 30.00000
[ INFO] [1668775029.563145980]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1668775029.568959170]: setupStreams...
[ INFO] [1668775029.654382161]: SELECTED BASE:Depth, 0
[ INFO] [1668775029.673068686]: RealSense Node Is Up!
[ WARN] [1668775029.926685624]: 
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
 18/11 21:52:10,226 ERROR [140332553467648] (ds5-thermal-monitor.cpp:94) Error during thermal compensation handling: query is available during streaming only
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

shaoxh, 2022-11-18:
[image]
[image]
[image]

shaoxh, 2022-11-18:
[image]

MartyG-RealSense, 2022-11-18:
Hi @shaoxh  In the RealSense ROS wrapper the infrared and IMU topics are disabled by default and should be enabled in the launch in order for their topics to be published.  For example:

`roslaunch realsense2_camera rs_camera.launch align_depth:=true enable_infra1:=true enable_infra2:=true enable_gyro:=true enable_accel:=true`

If you wish to combine the accel and gyro topics into a single **/imu** topic then add the command below to the end of the roslaunch instruction:

`unite_imu_method:=linear_interpolation`

MartyG-RealSense, 2022-12-02:
Case closed due to no further comments received

closed 2022-12-02 by MartyG-RealSense