realsenseai/realsense-ros #2559: ROS Humble: binary instalation for arm64

renzobazan96, 2022-11-22:
Hello,

I want to install the package over sudo apt install ros-humble-realsense2-camera. Currently there exist a package for amd64 but no for arm64: http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-humble-realsense2-camera/
For now I can build it from source

Is there any plan when the binaries will be released?

MartyG-RealSense, 2022-11-23:
Thanks very much @renzobazan96 for your question.  I will refer it to the RealSense ROS development team.

@Nir-Az  Could you advise @renzobazan96 about their Humble binary question above, please?  Thanks very much l

Nir-Az, 2022-11-24:
@renzobazan96 things happened faster than I imagined.
Please refresh your first link 😎

closed 2022-11-24 by renzobazan96

MartyG-RealSense, 2022-11-24:
Thanks again @Nir-Az  :)

It's great to hear that you were successful, @renzobazan96 - thanks very much for the update!

AndreV84, 2022-12-13:
so ros2 node supports humble release ? Right? @MartyG-RealSense

AndreV84, 2022-12-13:
what is the solution for arm64 focal ros humble implementation for d400? if any?

MartyG-RealSense, 2022-12-13:
Hi @AndreV84  The RealSense ROS2 wrapper supports Humble but for **Ubuntu 22.04 Jammy** rather than 20.04 Focal.

[image]

Building the wrapper for Humble on arm64 has had some problems though with the buildfarm, as described at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1280318586 and the link below.

https://discourse.ros.org/t/humble-build-farm-job-on-arm-fails-on-first-release/27413