realsenseai/realsense-ros #2563: How to visualise IMU data coming from d435i?

tsadarsh, 2022-11-27:
I am receiving the odometry information at `/camera/imu` topic after setting the `unit_imu_method:=2`. But I am unable to visualise them in Rviz. Does the realsense_ros library publish the required `tf` or am I supposed to set up the static transformations? 

[image]

The console from which I start Rviz, spams with the following INFO once I add the IMU topic in RViz for visualization:
> [INFO] [1669524588.898890607] [rviz]: Message Filter dropping message: frame 'camera_imu_optical_frame' at time 1669524588.549 for reason 'discarding message because the queue is full'

Is there any parameter that I need to `enable` for this? I went through the [documentation](https://dev.intelrealsense.com/docs/ros-wrapper) and made sure that `publish_tf` and `publish_odom_tf` is set to **True**.

[Here](https://www.youtube.com/watch?v=a-mfCeykmYw) is a sample video demonstration of what I am trying to visualise.

MartyG-RealSense, 2022-11-27:
Hi @tsadarsh  Does setting the topic in RViz as **/camera/gyro/sample** (the individual gyro topic) instead of /camera/imu (the combined IMU topic from unite_imu_method) cause IMU gyro data to be displayed in RViz?

tsadarsh, 2022-11-28:
> setting the topic in RViz as **/camera/gyro/sample** (the individual gyro topic) instead of /camera/imu

gives the following WARN message in my terminal and I do not see any change reflected in Rviz:
 > [WARN] [1669626320.995131908] [rviz]: New publisher discovered on topic '/camera/gyro/sample', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY

MartyG-RealSense, 2022-11-28:
And you still do not see odometry information if you use the **/camera/imu** topic and set **unite_imu_method:=1** (Copy instead of Linear_Interpolation)?

tsadarsh, 2022-11-29:
I set the topic to `/camera/imu` and `unite_imu_method:=1`. 
> Message Filter dropping message: frame 'camera_imu_optical_frame' at time 1669732749.843 for reason 'discarding message because the queue is full'

This is the output from the Rviz console.

MartyG-RealSense, 2022-11-30:
May I confirm that you have enabled the gyro and accel topics in your ROS launch, as they are disabled by default in the RealSense ROS wrapper.  They can be enabled by adding **enable_gyro:=true enable_accel:=true** to the launch instruction.

tsadarsh, 2022-11-30:
Hi @MartyG-RealSense. I am using the following launch command to start the realsense2_camera node:
```ros2 launch realsense2_camera rs_launch.py enable_accel:=true enable_gyro:=true unite_imu_method:=1 pointcloud.enable:=true```

MartyG-RealSense, 2022-12-02:
@tsadarsh  Your TF problem with ROS2 also sounds like a recent case at https://github.com/IntelRealSense/realsense-ros/issues/2560

tsadarsh, 2022-12-02:
@MartyG-RealSense Thank you for the help. I am currently checking out `imu_tools`. I am trying to create a launch file for it in ROS2. I will update you on my progress in some time.

MartyG-RealSense, 2022-12-14:
Hi @tsadarsh  Do you have an update about this case that you can provide, please?  Thanks!

tsadarsh, 2022-12-14:
Hi @MartyG-RealSense. I am extremely sorry for the delay. I had my University Exams going and they got over only yesterday. I will be working on this issue today and will try to give you updates in a few hours.

MartyG-RealSense, 2022-12-14:
It's no trouble at all.  Thanks very much for the update!

tsadarsh, 2022-12-14:
**Yes, I got some updates!**

Running the `imu_filter_madgwick` node does the magic :magic_wand: 

[rs_imu_rviz_demo.webm](https://user-images.githubusercontent.com/66778010/207669772-2bf9162f-8fb6-43e0-a2e7-baac56dad316.webm)

# Here are the commands that I used (summary):

## Step 1: Launch Realsense node :rocket: 
Launch the realsense camera node with accel and gyro enabled. 
```
ros2 launch realsense2_camera rs_launch.py enable_accel:=true enable_gyro:=true unite_imu_method:=1
```

## Step 2: Install `imu_tools` package :package: 

```sudo apt-get install ros-<YOUR_ROSDISTO>-imu-tools```

Details on source installation can be found [here](https://github.com/CCNYRoboticsLab/imu_tools/tree/humble#from-source-ros2)

## Step 3: Run imu_filter_madgwick node :magic_wand: 
Run the *imu_filter_madgwick node* and remap the */camera/imu* topic to */imu/data_raw*:
```
ros2 run imu_filter_madgwick imu_filter_madgwick_node --ros-args -p use_mag:=false -r /imu/data_raw:=/camera/imu
```

## Visulaize :world_map: 
Open up Rviz2 and set **Fixed Frame** to *odom*. Add the */imu/data/Imu* visualization from *By topic tab*:
```
ros2 run rviz2 rviz2
```

MartyG-RealSense, 2022-12-14:
Thanks so much @tsadarsh for sharing with the RealSense ROS community the method that worked for you!

closed 2022-12-16 by tsadarsh

Kaiser-Wang, 2023-09-12:
Hi @tsadarsh Did you receive any yaw drifts over time in your test?