realsenseai/realsense-ros #2564: D435i: aligned_depth_to_color topic flickering between aligned and not aligned depth

matlabbe, 2022-11-28:
When I launch my D435i camera with:
```
ros2 launch realsense2_camera rs_launch.py \
    align_depth.enable:=true \
    enable_gyro:=true \
    enable_accel:=true \
    unite_imu_method:=1 \
    enable_sync:=true \
    pointcloud.enable:=true
```
The topic `/camera/aligned_depth_to_color/image_raw` is flickering between the aligned version and the not aligned version (`/camera/depth/image_rect_raw`). Below is what I can see in rviz2. On left, from top to bottom: `color`, `aligned_depth_to_color` and `depth` (not aligned with color). However, the colored point cloud looks fine:
[image]

Why is `/camera/aligned_depth_to_color/image_raw` flickering?

My config (I am using the realsense binaries from ros2 galactic):
```
ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true enable_gyro:=true enable_accel:=true unite_imu_method:=1 enable_sync:=true pointcloud.enable:=true

[INFO] [launch]: All log files can be found below /home/mathieu/.ros/log/2022-11-27-17-08-56-992341-mathieu-R12-43217
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [43219]
[realsense2_camera_node-1] [INFO] [1669597737.214204593] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1669597737.214412595] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1669597737.214460729] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1669597737.276378519] [camera.camera]: Device with serial number 846112070739 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1669597737.276504247] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1669597737.276530112] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1669597737.277015787] [camera.camera]: Device with port number 2-9 was found.
[realsense2_camera_node-1] [INFO] [1669597737.277043690] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1669597737.279898060] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1669597737.280634887] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1669597737.280681389] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1669597737.280718272] [camera.camera]: Device Serial No: 846112070739
[realsense2_camera_node-1] [INFO] [1669597737.280742626] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1669597737.280765422] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1669597737.280786434] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1669597737.280806241] [camera.camera]: Sync Mode: On
[realsense2_camera_node-1] [INFO] [1669597737.347355203] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1669597737.348348101] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1]is  27/11 17:08:57,348 WARNING [140629258508032] (backend-v4l2.cpp:1444) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[realsense2_camera_node-1] [INFO] [1669597737.351599395] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1669597737.351917621] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1669597737.351951605] [camera.camera]: Set ROS param accel_fps to default: 63
[realsense2_camera_node-1] [WARN] [1669597737.355614391] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1669597737.356107793] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1669597737.358498997] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1669597737.367099993] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1669597737.367151372] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1669597737.368230316] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1669597737.379001229] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1669597737.379054987] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1669597737.442335746] [camera.camera]: Starting Sensor: Motion Module
[realsense2_camera_node-1]  27/11 17:08:57,439 WARNING [140629258508032] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[realsense2_camera_node-1] [INFO] [1669597737.446404897] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 63
[realsense2_camera_node-1] [INFO] [1669597737.446479531] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[realsense2_camera_node-1] [INFO] [1669597737.447709535] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1669597737.530489463] [camera.camera]: 
[realsense2_camera_node-1] [WARN] [1669598035.490490287] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
```

MartyG-RealSense, 2022-11-28:
Hi @matlabbe  You could first try removing **enable_sync:=true** from your launch instruction to eliminate it as a possible cause of the alternating image.  The ROS wrapper documentation states that enable_sync gathers the closest frames of different sensors, such as infrared, color and depth, to be sent with the same timetag.  But this happens automatically when the pointcloud filter is enabled, so you should not need to set enable_sync to true.

matlabbe, 2022-11-28:
With:
```
ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true
```
Same result (also tested latest `realsense2_camera` from source, same flickering issue than on galactic binaries):
[image]

Here is the firmware used on the camera, I think it is the latest (as realsense-viewer v2.51.1 doesn't propose me to upgrade it):

[image]

On ROS1/noetic realsense binaries, it works correctly (no flikering).
```
roslaunch realsense2_camera rs_camera.launch align_depth:=true
```
[image]

matlabbe, 2022-11-29:
There is the log corresponding to previous test:
```
$ ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true

[INFO] [launch]: All log files can be found below /home/mathieu/.ros/log/2022-11-28-18-05-07-855926-mathieu-R12-4710
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [4712]
[realsense2_camera_node-1] [INFO] [1669687508.147881173] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1669687508.148051729] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1669687508.148086056] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1669687508.261712000] [camera.camera]: Device with serial number 846112070739 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1669687508.261854147] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1669687508.261882903] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1669687508.262381143] [camera.camera]: Device with port number 2-9 was found.
[realsense2_camera_node-1] [INFO] [1669687508.262414244] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1669687508.267296588] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1669687508.268088325] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1669687508.268137371] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1669687508.268161398] [camera.camera]: Device Serial No: 846112070739
[realsense2_camera_node-1] [INFO] [1669687508.268181803] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1669687508.268205376] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1669687508.268225312] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1669687508.268245310] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1669687508.334006272] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1669687508.336776426] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1]  28/11 18:05:08,336 WARNING [139748370806528] (backend-v4l2.cpp:1444) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[realsense2_camera_node-1] [INFO] [1669687508.339928132] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1669687508.340175243] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1669687508.340194895] [camera.camera]: Set ROS param accel_fps to default: 63
[realsense2_camera_node-1] [WARN] [1669687508.343322706] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1669687508.344018592] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1669687508.400144238] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1669687508.455596491] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1669687508.455879697] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1669687508.459313535] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1669687508.466610450] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1669687508.467072004] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1669687508.522827501] [camera.camera]:
```

For the `No stream match for pointcloud chosen texture Process - Color` warning, I tried the original bringup line, but I didn't reproduce it, not sure what happened at that time.

MartyG-RealSense, 2022-11-29:
So the **No stream match for pointcloud chosen texture Process - Color** didn't repeat in the last test but the image is still flickering?

The warning **likely requires patch for fourcc code RW16!** has occurred twice now in subsequent tests.  A fourcc code warning typically suggests that librealsense has been built from source code but a patch has not been applied to the kernel before librealsense was built.  It does not usually occur if librealsense was built from packages, as the kernel patch is bundled within the packages.

Can you confirm which method you used to install the librealsense SDK and the ROS2 Galactic wrapper, please?

matlabbe, 2022-11-30:
I saw that I had also `librealsense2` binaries installed along `ros-galactic-librealsense2`. I uninstalled all packages mentionning realsense or librealsense, then reinstalled only `ros-galatic-realsense2-camera`.

```
sudo apt install ros-galactic-realsense2-camera
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  [...]
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  ros-galactic-librealsense2 ros-galactic-realsense2-camera-msgs
The following NEW packages will be installed:
  ros-galactic-librealsense2 ros-galactic-realsense2-camera ros-galactic-realsense2-camera-msgs
0 upgraded, 3 newly installed, 0 to remove and 153 not upgraded.
Need to get 13.8 MB of archives.
After this operation, 42.1 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://packages.ros.org/ros2/ubuntu focal/main amd64 ros-galactic-librealsense2 amd64 2.51.1-1focal.20220912.180123 [13.3 MB]
Get:2 http://packages.ros.org/ros2/ubuntu focal/main amd64 ros-galactic-realsense2-camera-msgs amd64 4.51.1-1focal.20220913.170304 [65.0 kB]
Get:3 http://packages.ros.org/ros2/ubuntu focal/main amd64 ros-galactic-realsense2-camera amd64 4.51.1-1focal.20221016.135406 [489 kB]
Fetched 13.8 MB in 4s (3,235 kB/s)                      
Selecting previously unselected package ros-galactic-librealsense2.
(Reading database ... 348155 files and directories currently installed.)
Preparing to unpack .../ros-galactic-librealsense2_2.51.1-1focal.20220912.180123_amd64.deb ...
Unpacking ros-galactic-librealsense2 (2.51.1-1focal.20220912.180123) ...
Selecting previously unselected package ros-galactic-realsense2-camera-msgs.
Preparing to unpack .../ros-galactic-realsense2-camera-msgs_4.51.1-1focal.20220913.170304_amd64.deb ...
Unpacking ros-galactic-realsense2-camera-msgs (4.51.1-1focal.20220913.170304) ...
Selecting previously unselected package ros-galactic-realsense2-camera.
Preparing to unpack .../ros-galactic-realsense2-camera_4.51.1-1focal.20221016.135406_amd64.deb ...
Unpacking ros-galactic-realsense2-camera (4.51.1-1focal.20221016.135406) ...
Setting up ros-galactic-realsense2-camera-msgs (4.51.1-1focal.20220913.170304) ...
Setting up ros-galactic-librealsense2 (2.51.1-1focal.20220912.180123) ...
Setting up ros-galactic-realsense2-camera (4.51.1-1focal.20221016.135406) ...
Processing triggers for libc-bin (2.31-0ubuntu9.9) ...
```

Here is the log, I still see the flickering issue:
```
ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true
[INFO] [launch]: All log files can be found below /home/mathieu/.ros/log/2022-11-29-18-57-23-897893-mathieu-R12-38248
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [38250]
[realsense2_camera_node-1] [INFO] [1669777044.119039031] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1669777044.119201669] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1669777044.119231565] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1669777044.184296501] [camera.camera]: Device with serial number 846112070739 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1669777044.184413396] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1669777044.184440448] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1669777044.184944449] [camera.camera]: Device with port number 2-9 was found.
[realsense2_camera_node-1] [INFO] [1669777044.184976382] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1669777044.187739275] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1669777044.188291957] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1669777044.188331811] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1669777044.188352368] [camera.camera]: Device Serial No: 846112070739
[realsense2_camera_node-1] [INFO] [1669777044.188369826] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1669777044.188390852] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1669777044.188409950] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1669777044.188428128] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1669777044.250033724] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1669777044.252793402] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1669777044.256019443] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1669777044.256302178] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1669777044.256329367] [camera.camera]: Set ROS param accel_fps to default: 63
[... cut ...]
	librealsense2_camera_msgs__rosidl_typesupport_cpp.so => /home/mathieu/ros2_ws/install/realsense2_camera_msgs/lib/librealsense2_camera_msgs__rosidl_typesupport_cpp.so (0x00007f7e40f3f000)
```

I'll try in the next days to debug why on ros1 it works and why on ros2 it is flickering. For reference:
ros1:
https://github.com/IntelRealSense/realsense-ros/blob/44a703ed0ce7570668fae47e354a0b8d62ed1c05/realsense2_camera/src/base_realsense_node.cpp#L1704-L1720

ros2:
https://github.com/IntelRealSense/realsense-ros/blob/af65233a85e72bc137a71bbd6c9b7d8f7de62c9c/realsense2_camera/src/base_realsense_node.cpp#L589-L602

Maybe frameset is different for some reasons between ros1 and ros2.

MartyG-RealSense, 2022-11-30:
Thanks very much for the further information.  I look forward to your next report after your testing.  Good luck!

matlabbe, 2022-12-04:
Ok I've found a patch to make it work on ros2. As shown in my previous post, the biggest difference between ros1 and ros2 is that in ros1 there is a variable `is_color_frame` that makes sure that only the aligned depth frame is published by the the aligned depth publisher. Here is the diff to make it work for ROS2:
```diff
diff --git a/realsense2_camera/src/base_realsense_node.cpp b/realsense2_camera/src/base_realsense_node.cpp
index 328eb53..dc0dfb7 100755
--- a/realsense2_camera/src/base_realsense_node.cpp
+++ b/realsense2_camera/src/base_realsense_node.cpp
@@ -590,7 +590,7 @@ void BaseRealSenseNode::frame_callback(rs2::frame frame)
             {
                 if (sent_depth_frame) continue;
                 sent_depth_frame = true;
-                if (_align_depth_filter->is_enabled())
+                if (_align_depth_filter->is_enabled() && frameset.get_color_frame())
                 {
                     publishFrame(f, t, COLOR,
                             _depth_aligned_image,
```

Here is the debug log, we can set that not aligned and aligned depth frames appear in the `frameset`, so we should have a way to know which one is aligned to color:
```bash
# Aligned depth and color:
[realsense2_camera_node-1] [DEBUG] [1670113933.512288535] [camera.camera]: Frameset arrived.
[realsense2_camera_node-1] [DEBUG] [1670113933.512363310] [camera.camera]: List of frameset before applying filters: size: 2
[realsense2_camera_node-1] [DEBUG] [1670113933.512382308] [camera.camera]: Frameset contain (Color, 0, RGB8 3) frame. frame_number: 150 ; frame_TS: 1670113933463.059326 ; ros_TS(NSec): 1670113933463059200
[realsense2_camera_node-1] [DEBUG] [1670113933.512396766] [camera.camera]: Frameset contain (Depth, 0, Z16 0) frame. frame_number: 150 ; frame_TS: 1670113933463.059326 ; ros_TS(NSec): 1670113933463059200
[realsense2_camera_node-1] [DEBUG] [1670113933.512408763] [camera.camera]: num_filters: 11
[realsense2_camera_node-1] [DEBUG] [1670113933.521614939] [camera.camera]: List of frameset after applying filters: size: 2
[realsense2_camera_node-1] [DEBUG] [1670113933.521664253] [camera.camera]: Frameset contain (Depth, 0, Z16) frame. frame_number: 196 ; frame_TS: 1670113933463.059326 ; ros_TS(NSec): 1670113933463059200
[realsense2_camera_node-1] [DEBUG] [1670113933.521691582] [camera.camera]: frame: 1280 x 720
[realsense2_camera_node-1] [DEBUG] [1670113933.521707045] [camera.camera]: publishFrame(...)
[realsense2_camera_node-1] [DEBUG] [1670113933.521731678] [camera.camera]: Frameset contain (Color, 0, RGB8) frame. frame_number: 150 ; frame_TS: 1670113933463.059326 ; ros_TS(NSec): 1670113933463059200
[realsense2_camera_node-1] [DEBUG] [1670113933.521777745] [camera.camera]: frame: 1280 x 720
[realsense2_camera_node-1] [DEBUG] [1670113933.521792522] [camera.camera]: publishFrame(...)
[realsense2_camera_node-1] [DEBUG] [1670113933.521819804] [camera.camera]: publishFrame(...)

# This depth image is not aligned, it should not be published in aligned depth publisher:
[realsense2_camera_node-1] [DEBUG] [1670113933.611862731] [camera.camera]: Frameset arrived.
[realsense2_camera_node-1] [DEBUG] [1670113933.611924539] [camera.camera]: List of frameset before applying filters: size: 1
[realsense2_camera_node-1] [DEBUG] [1670113933.611931072] [camera.camera]: Frameset contain (Depth, 0, Z16 0) frame. frame_number: 197 ; frame_TS: 1670113933496.470703 ; ros_TS(NSec): 1670113933496470784
[realsense2_camera_node-1] [DEBUG] [1670113933.611938894] [camera.camera]: num_filters: 11
[realsense2_camera_node-1] [DEBUG] [1670113933.611947648] [camera.camera]: List of frameset after applying filters: size: 1
[realsense2_camera_node-1] [DEBUG] [1670113933.611952028] [camera.camera]: Frameset contain (Depth, 0, Z16) frame. frame_number: 197 ; frame_TS: 1670113933496.470703 ; ros_TS(NSec): 1670113933496470784
[realsense2_camera_node-1] [DEBUG] [1670113933.611959128] [camera.camera]: frame: 848 x 480
[realsense2_camera_node-1] [DEBUG] [1670113933.611964599] [camera.camera]: publishFrame(...)
[realsense2_camera_node-1] [DEBUG] [1670113933.611982775] [camera.camera]: publishFrame(...)

# Aligned depth and color:
[realsense2_camera_node-1] [DEBUG] [1670113933.611997193] [camera.camera]: Frameset arrived.
[realsense2_camera_node-1] [DEBUG] [1670113933.612001986] [camera.camera]: List of frameset before applying filters: size: 2
[realsense2_camera_node-1] [DEBUG] [1670113933.612006722] [camera.camera]: Frameset contain (Color, 0, RGB8 3) frame. frame_number: 151 ; frame_TS: 1670113933563.359619 ; ros_TS(NSec): 1670113933563359744
[realsense2_camera_node-1] [DEBUG] [1670113933.612012860] [camera.camera]: Frameset contain (Depth, 0, Z16 0) frame. frame_number: 151 ; frame_TS: 1670113933563.359619 ; ros_TS(NSec): 1670113933563359744
[realsense2_camera_node-1] [DEBUG] [1670113933.612018508] [camera.camera]: num_filters: 11
[realsense2_camera_node-1] [DEBUG] [1670113933.615805606] [camera.camera]: List of frameset after applying filters: size: 2
[realsense2_camera_node-1] [DEBUG] [1670113933.615839934] [camera.camera]: Frameset contain (Depth, 0, Z16) frame. frame_number: 198 ; frame_TS: 1670113933563.359619 ; ros_TS(NSec): 1670113933563359744
[realsense2_camera_node-1] [DEBUG] [1670113933.615866709] [camera.camera]: frame: 1280 x 720
[realsense2_camera_node-1] [DEBUG] [1670113933.615873498] [camera.camera]: publishFrame(...)
[... cut ...]
[realsense2_camera_node-1] [DEBUG] [1670113933.725368689] [camera.camera]: publishFrame(...)
[realsense2_camera_node-1] [DEBUG] [1670113933.725400517] [camera.camera]: Frameset contain (Color, 0, RGB8) frame. frame_number: 154 ; frame_TS: 1670113933663.696777 ; ros_TS(NSec): 1670113933663696896
[realsense2_camera_node-1] [DEBUG] [1670113933.725426158] [camera.camera]: frame: 1280 x 720
[realsense2_camera_node-1] [DEBUG] [1670113933.725442239] [camera.camera]: publishFrame(...)
[realsense2_camera_node-1] [DEBUG] [1670113933.725475936] [camera.camera]: publishFrame(...)
```

MartyG-RealSense, 2022-12-04:
Thanks so much @matlabbe for sharing the patch with the RealSense ROS community!

MartyG-RealSense, 2022-12-10:
Hi @matlabbe  Do you require further assistance with this case, please?  Thanks!

matlabbe, 2022-12-11:
On my part it is fine with that patch, though it would be more convenient if it could be fixed directly in `ros2-developpment` branch.

MartyG-RealSense, 2022-12-11:
Thanks very much @matlabbe for the update.  I have added a **wait to close** label to this case to keep it open to see whether other RealSense users report the same flickering issue.  Thanks again!

MartyG-RealSense, 2023-02-18:
As there have not been further reports of the flickering issue from other RealSense ROS users after 2 months, I will close this issue.  Thanks again @matlabbe

closed 2023-02-18 by MartyG-RealSense

jemoster, 2023-03-27:
I saw the same issue as matlabbe and was able to fix it using an updated patch.
[aligned_depth.patch](https://github.com/IntelRealSense/realsense-ros/files/11073521/aligned_depth.patch)

MartyG-RealSense, 2023-03-27:
Thanks so much @jemoster for sharing your solution with the RealSense community!