realsenseai/realsense-ros #2565: ROS2 Dynamic Reconfigure Image Transport

xkaraman, 2022-11-28:
Currently only `base` topic parameters can be configured.

Image transport plugins like `compressed`, `compressedDepth` are not available for reconfigure from `ros2 run rqt_reconfigure rqt_reconfigure`
[image]

Trying to configure them manually with
`ros2 param set /camera/camera .aligned_depth_to_color.image_raw.compressed.format png` results to an error of

`Setting parameter failed: parameter '.aligned_depth_to_color.image_raw.compressed.format' cannot be set because it was not declared`

MartyG-RealSense, 2022-11-30:
Hi @xkaraman  Your **ros2 param set** instruction looks unusual to me in regard to how it has full-stop marks instead of / symbols.  

`ros2 param set /camera/camera .aligned_depth_to_color.image_raw.compressed.format png`

My expectation would be that the instruction would look more like this:

`ros2 param set /camera/aligned_depth_to_color/image_raw/compressed`

Also, `aligned` topics would only be published if **align_depth** was set to true.  Can you confirm if align_depth is set to true in your launch please?  Thanks!

xkaraman, 2022-11-30:
Yeah `.param` it's seems unusual to me coming from ros1 but apparently that's how it's done in ros2. Also `ros2 param set node_name param_name value` is the way to go for setting them in ROS2.

Yes `aligned_depth` was set to true and topics where visible and able to visualize through rviz

MartyG-RealSense, 2022-12-01:
@xkaraman  I will highlight your issue to one of my Intel RealSense colleagues.  Thanks very much for your patience!

@Nir-Az  In this issue, @xkaraman has compressed topics listed in ROS2 when using **ros2 topic list** but cannot interact with those compressed topics in the rqt_reconfigure window during runtime after launch or configure them with `ros2 param set` instructions.  Is that normal for ROS2 please?  Thanks!

[image]

Nir-Az, 2022-12-04:
Hi,
I am not familiar with image transport parameters, but are you sure you see the new compressed topic in the topic list?

xkaraman, 2022-12-05:
Yes @Nir-Az, i have also confirmed through `ros2 topic echo /camera/color/image_raw/compressed --once --no-arr` that yields:
```
header:
  stamp:
    sec: 1670235648
    nanosec: 56710912
  frame_id: camera_color_optical_frame
format: rgb8; jpeg compressed bgr8
data: '<sequence type: uint8, length: 285846>'
---
```

I would like to also report that when i open `ros2 run rqt_reconfigure rqt_reconfigure` and navigate to `camera/camera` paremters, when i scroll down to see the pameters, `colorizer.stream_format_filter` is the last one i can see before `rqt` crashing with an expception of `[ros2run]: Floating point exception`.

Nir-Az, 2022-12-05:
OK, we will need to investigate those 2 issues on our side
Thanks

xkaraman, 2022-12-05:
I noticed that if i remove `namespace` argument when launching the node from ROS2 launch file `rs_launch.py` params do NOT have the `.` in from of the param that has been odd to me as well as @MartyG-RealSense. Maybe i should open a new issue for it?
Still no compressed parameters are shown though.

MartyG-RealSense, 2022-12-14:
Hi @xkaraman  Let's await the outcome of the investigation by @Nir-Az before creating another issue.  Thanks!

xkaraman, 2023-01-03:
Any news on this @MartyG-RealSense ?

MartyG-RealSense, 2023-01-03:
Hi @xkaraman  I will follow up with @Nir-Az about this issue.  Thanks very much for your patience.

MartyG-RealSense, 2023-01-03:
Hi @xkaraman  I checked with the RealSense ROS team about this issue.  They advised that it's in their pipeline but not a main focus at the moment and they will investigate it once they are able to do so.

MartyG-RealSense, 2023-01-09:
Hi @xkaraman  Bearing in mind the information in the comment above, do you require further assistance with this case please?  Thanks!

xkaraman, 2023-01-10:
Ok thanks for the update @MartyG-RealSense. Personally i don't need further assistance but maybe the issue must remain opened until a fix is proposed?

MartyG-RealSense, 2023-01-10:
I put an **Enhancement** label in the case to signify to keep it open whilst the feature request remains active.

SamerKhshiboun, 2023-03-19:
Hi @xkaraman 

My name is Samer and I'm a developer in the RealSense SW Team.

Regarding the scroll down issue in the rqt_reconfigure and that the last parameter you see is colorizer.stream_format_filter:
The problem is that the next paramter that rqt_reconfigure trying to pull and view, is "colorizer.stream_index_filter" which defined as an INT value in a range of [-1, INT_MAX].
LibRealSense is returning this range, but ROS is not accepting the INT_MAX due to conversion issues, which turns to be negative number

This is the message I get from rqt_reconfigure:
`[WARN] [1679264563.906326759] [rqt_reconfigure]: Failed to retrieve parameters from node: argument 2 overflowed: value must be in the range -2147483648 to 2147483647
`

I will continue investigating this issue and will look for a proper way to define the parameter range.
Regarding the other issues, I need to look more deeper in order to understand it.

Will update asap.

xkaraman, 2023-03-29:
Thanks @SamerKhshiboun for the update! If you have any other updates or want me to help somehow, you are more than welcome to ask! :)

johannesWen, 2023-05-15:
Hi, @SamerKhshiboun 
Do you have an update concerning the integer overflow?

MartyG-RealSense, 2023-05-25:
Hi @SamerKhshiboun  Could you kindly assist with the question of @johannesWen above, please?  Thanks!

SamerKhshiboun, 2023-05-31:
Hi @MartyG-RealSense , @johannesWen 
We opened a PR on librealsense repo
https://github.com/IntelRealSense/librealsense/pull/11858

When this PR will be merged, you can use the latest code of librealsense (development branch) and recompile your librealsense library, and then when using rqt_reconfigure the problem will disappear.

Also, this fix will be included in the future releases of Librealsense

Thanks,
Samer

MartyG-RealSense, 2023-07-11:
Case closed due to merging of a fix into the librealsense SDK.

closed 2023-07-11 by MartyG-RealSense