realsenseai/realsense-ros #2567: can't able to open L515 camera

Daya-AWR1642, 2022-12-02:
I followed installation guideline for Ubuntu 20.04  from:
https://github.com/IntelRealSense/realsense-ros#installation-instructions
when I run the "ros2 launch realsense2_camera rs_launch.py" this line in terminal I'm getting like this:

dayananda@dayananda-Inspiron-3501:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py
[INFO] [launch]: All log files can be found below /home/dayananda/.ros/log/2022-12-02-16-39-04-582379-dayananda-Inspiron-3501-5443
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [5445]
[realsense2_camera_node-1] [INFO] [1669979344.917238612] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1669979344.917467097] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1669979344.917513965] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1669979345.157759419] [camera.camera]: Device with serial number f1321376 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1669979345.157884946] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video2 was found.
[realsense2_camera_node-1] [INFO] [1669979345.157911647] [camera.camera]: Device with name Intel RealSense L515 was found.
[realsense2_camera_node-1] [INFO] [1669979345.158423800] [camera.camera]: Device with port number 2-2 was found.
[realsense2_camera_node-1] [INFO] [1669979345.158452760] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1669979345.160756414] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1669979345.161212455] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1669979345.161248280] [camera.camera]: Device Name: Intel RealSense L515
[realsense2_camera_node-1] [INFO] [1669979345.161273075] [camera.camera]: Device Serial No: f1321376
[realsense2_camera_node-1] [INFO] [1669979345.161291523] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video2
[realsense2_camera_node-1] [INFO] [1669979345.161311069] [camera.camera]: Device FW version: 01.05.04.01
[realsense2_camera_node-1] [INFO] [1669979345.161331631] [camera.camera]: Device Product ID: 0x0B64
[realsense2_camera_node-1] [INFO] [1669979345.161350253] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1669979345.168578669] [camera.camera]: Set ROS param depth_module.profile to default: 640x480x30
[realsense2_camera_node-1] [INFO] [1669979345.170549777] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1669979345.170735904] [camera.camera]: Set ROS param gyro_fps to default: 100
[realsense2_camera_node-1] [INFO] [1669979345.170768514] [camera.camera]: Set ROS param accel_fps to default: 200
[realsense2_camera_node-1] [WARN] [1669979345.174431029] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1669979345.174856031] [camera.camera]: Stopping Sensor: L500 Depth Sensor
[realsense2_camera_node-1] [INFO] [1669979345.176856616] [camera.camera]: Starting Sensor: L500 Depth Sensor
[realsense2_camera_node-1] [INFO] [1669979345.195438080] [camera.camera]: Open profile: stream_type: Confidence(0), Format: RAW8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1669979345.195519052] [camera.camera]: Open profile: stream_type: Infra(0), Format: Y8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1669979345.195532473] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1669979345.195733064] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1669979345.388084822] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1669979345.394859170] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1669979345.395600183] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1669979346.187190035] [camera.camera]: 
[realsense2_camera_node-1] [WARN] [1669979346.187275039] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.

But camera is not opening

Daya-AWR1642, 2022-12-03:
I got this 
dayananda@dayananda-Inspiron-3501:~/ros2_ws$ ros2 topic list
/parameter_events
/rosout

MartyG-RealSense, 2022-12-03:
It is unusual that no camera topics are listed as you have no significant warning messages after the completion of launch with the **RealSense Node Is Up!** message, indicating that the launch was successful.

Can the camera display streams correctly in the RealSense Viewer tool?

Daya-AWR1642, 2022-12-03:
yes, I can

MartyG-RealSense, 2022-12-03:
Do the topics publish if you add **initial_reset:=true** to your launch instruction to reset the camera at launch?

`ros2 launch realsense2_camera rs_launch.py initial_reset:=true`

Daya-AWR1642, 2022-12-03:
[image]

But still not getting

MartyG-RealSense, 2022-12-03:
Global time can be enabled by adding **global_time_enabled:=true** to your launch instruction.

Your camera's firmware driver is also somewhat old at version **1.5.4.1** and so I would recommend updating to version **1.5.8.1**, the newest version available for L515.  The update can be performed in the RealSense Viewer.

Daya-AWR1642, 2022-12-03:
[image]

still no

MartyG-RealSense, 2022-12-03:
I cannot see anything in the logs that would indicate a problem with publishing the topics, and the camera works normally with the RealSense Viewer.

When you perform the ROS launch, is the ROS terminal the only program that you have open and you do not have any other RealSense programs such as the RealSense Viewer already running?

Did you also test reversing the USB connector as suggested earlier at https://github.com/IntelRealSense/realsense-ros/issues/2567#issuecomment-1335285691

Daya-AWR1642, 2022-12-05:
Hi
My issue is resolved, I can launch camera through ros2
Thank you

MartyG-RealSense, 2022-12-05:
You are very welcome.  Thanks very much for the confirmation that you can launch the camera in ROS2.

MartyG-RealSense, 2022-12-10:
Hi @Daya-AWR1642  Do you require further assistance with this case, please?  Thanks!

Daya-AWR1642, 2022-12-10:
No, Thanks