realsenseai/realsense-ros #2571: Camera's frame id should be configurable

nakai-omer, 2022-12-07:
Hello,

After struggling to figure out why we are not getting correct transformations, we found that this lib forces the camera's link to be named `camera_link`, otherwise transformations aren't valid. In our case the camera's link is called `base_camera_link`, as we have multiple cameras. 
It will be very helpful for the camera's link to be configurable, so existing robot descriptions don't need to be re-written just because we changed our camera to a realsense one. Re-writing the urdf, also affects having to re-import the robot to tools like Isaac Sim, and redo all the sensor configurations there etc. This might also fix issues with multiples cameras and their transformations.

This is the relevant part of our tf tree:
[image]

MartyG-RealSense, 2022-12-07:
Hi @nakai-omer  As you are using multiple cameras, you may find it useful to use the ROS2 wrapper's **rs_multi_camera_launch.py** launch file for launching more than one camera so that the launch file takes care of naming the cameras.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_multi_camera_launch.py

The ROS1 wrapper's equivalent multicam launch file is **rs_multiple_devices.launch**

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#work-with-multiple-cameras

nakai-omer, 2022-12-12:
@MartyG-RealSense We are using a single camera, we just gave it a different link name, which is not `camera_link`, and we had to look through the code to figure out what to change the frame names in this package.

MartyG-RealSense, 2022-12-12:
Whilst I am not aware of any current plans to create such a feature, I have set a **wait to close** label on this issue to keep it open.  Other RealSense ROS users are welcome to comment in support of adding such a feature if they believe that it would be advantageous to them.

nakai-omer, 2022-12-12:
@MartyG-RealSense I think there is a misunderstanding. See my second comment please. It does exist, it is just not documented. There is `camera_name` param, that doesn't exist in the documentation.

MartyG-RealSense, 2022-12-12:
Thanks very much for the clarification.  You are correct that there is a short reference to `camera_name` in the ROS2 wrapper documentation section about moving from ros2_legacy.

[image]

I will highlight to the RealSense ROS team your request to make camera_name more prominent in the wrapper instructions.

Nir-Az, 2022-12-13:
https://dev.intelrealsense.com/docs/ros-wrapper currently holds a list of examples and parameters for ROS1,
We are about to update this list for ROS2, with that we will address this issue as well.
Thanks for notifying

MartyG-RealSense, 2022-12-22:
Hi @nakai-omer  Bearing in mind the advice from @Nir-Az in the above comment, do you require further assistance with this case please?  Thanks!

nakai-omer, 2022-12-22:
@MartyG-RealSense no thanks a alot

closed 2022-12-22 by nakai-omer