realsenseai/realsense-ros #2573: RGB

jwson97, 2022-12-08:
Hi, I am using 435i.
I'm trying to recognize an object and figure out the distance. 
I perform object detection on the rgb image and get the distance from the depth image. 
But RGB image is not aligned with Depth image. 

[image]
[image]

The first image is the result of using ros2 with realsense2_camera rs_launch.py, and the second image from pyrealsense2. 

In the first image, the location of the red dot is on the chair, but in the depth image, points with the same coordinates is not on the chair.
The result for the second image is similar. A point in RGB is different from a point in depth. 

From this problem, I cannot find the distance of the object from depth image after recognizing the object from the RGB image. How can I solve this problem. Or is there any other way to recognize the object and get the distance?

jwson97, 2022-12-08:
Sorry I solved it with rs.align.

jwson97, 2022-12-08:
It published on different topic. It wokrs.

closed 2022-12-08 by jwson97

MartyG-RealSense, 2022-12-08:
Thanks very much @jwson97 for the updates - it's excellent news that you succeeded!