realsenseai/realsense-ros #2574: No image with realsense_d435i_color.launch.py

ryanmaxwell96, 2022-12-09:
I'm having issues getting an image to appear when running the following command:

ros2 launch rtabmap_ros realsense_d435i_color.launch.py

My setup is:

Jetson AGX Xavier
Jetpack 5.0.2
Ubuntu 20.04
Foxy ROS2 distro

The terminal output that comes is this:

`[INFO] [launch]: All log files can be found below /home/nvidia/.ros/log/2022-12-09-13-51-30-676381-nvidia-desktop-12755
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rgbd_odometry-1]: process started with pid [12757]
[INFO] [rtabmap-2]: process started with pid [12759]
[INFO] [rtabmapviz-3]: process started with pid [12761]
[INFO] [point_cloud_xyz-4]: process started with pid [12763]
[INFO] [pointcloud_to_depthimage-5]: process started with pid [12765]
[INFO] [imu_filter_madgwick_node-6]: process started with pid [12767]
[INFO] [static_transform_publisher-7]: process started with pid [12769]
[static_transform_publisher-7] [INFO] [1670626292.087608462] [static_transform_publisher_2Thqt8CnRIyfq9Ie]: Spinning until killed publishing transform from 'camera_gyro_optical_frame' to 'camera_imu_optical_frame'
[imu_filter_madgwick_node-6] [INFO] [1670626292.105999728] [imu_filter_madgwick]: Starting ImuFilter
[imu_filter_madgwick_node-6] [INFO] [1670626292.121180900] [imu_filter_madgwick]: Using dt computed from message headers
[imu_filter_madgwick_node-6] [INFO] [1670626292.122127639] [imu_filter_madgwick]: The gravity vector is kept in the IMU message.
[imu_filter_madgwick_node-6] [INFO] [1670626292.127043169] [imu_filter_madgwick]: Imu filter gain set to 0.100000
[imu_filter_madgwick_node-6] [INFO] [1670626292.127545916] [imu_filter_madgwick]: Gyro drift bias set to 0.000000
[imu_filter_madgwick_node-6] [INFO] [1670626292.127791625] [imu_filter_madgwick]: Magnetometer bias values: 0.000000 0.000000 0.000000
[pointcloud_to_depthimage-5] [INFO] [1670626294.879037345] [pointcloud_to_depthimage]: Params:
[pointcloud_to_depthimage-5] [INFO] [1670626294.891783595] [pointcloud_to_depthimage]:   approx=true
[point_cloud_xyz-4] [INFO] [1670626294.895261189] [point_cloud_xyz]: Approximate time sync = false
[pointcloud_to_depthimage-5] [INFO] [1670626294.897965558] [pointcloud_to_depthimage]:   queue_size=10
[pointcloud_to_depthimage-5] [INFO] [1670626294.898277159] [pointcloud_to_depthimage]:   fixed_frame_id=camera_link
[pointcloud_to_depthimage-5] [INFO] [1670626294.898594616] [pointcloud_to_depthimage]:   wait_for_transform=0.100000s
[pointcloud_to_depthimage-5] [INFO] [1670626294.898806243] [pointcloud_to_depthimage]:   fill_holes_size=1 pixels (0=disabled)
[pointcloud_to_depthimage-5] [INFO] [1670626294.898955499] [pointcloud_to_depthimage]:   fill_holes_error=0.100000
[pointcloud_to_depthimage-5] [INFO] [1670626294.899070865] [pointcloud_to_depthimage]:   fill_iterations=1
[pointcloud_to_depthimage-5] [INFO] [1670626294.899161110] [pointcloud_to_depthimage]:   decimation=2
[pointcloud_to_depthimage-5] [INFO] [1670626294.899256923] [pointcloud_to_depthimage]:   upscale=false (upscale_depth_error_ratio=0.020000)
[pointcloud_to_depthimage-5] [WARN] [1670626294.903263090] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rgbd_odometry-1] [INFO] [1670626295.114063503] [rgbd_odometry]: Odometry: frame_id               = camera_link
[rgbd_odometry-1] [INFO] [1670626295.125191521] [rgbd_odometry]: Odometry: odom_frame_id          = odom
[rgbd_odometry-1] [INFO] [1670626295.125549012] [rgbd_odometry]: Odometry: publish_tf             = true
[rgbd_odometry-1] [INFO] [1670626295.125669306] [rgbd_odometry]: Odometry: wait_for_transform     = 0.100000
[rgbd_odometry-1] [INFO] [1670626295.126133747] [rgbd_odometry]: Odometry: initial_pose           = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
[rgbd_odometry-1] [INFO] [1670626295.126238712] [rgbd_odometry]: Odometry: ground_truth_frame_id  = 
[rgbd_odometry-1] [INFO] [1670626295.127046851] [rgbd_odometry]: Odometry: ground_truth_base_frame_id = camera_link
[rgbd_odometry-1] [INFO] [1670626295.127182539] [rgbd_odometry]: Odometry: config_path            = 
[rgbd_odometry-1] [INFO] [1670626295.127254639] [rgbd_odometry]: Odometry: publish_null_when_lost = true
[rgbd_odometry-1] [INFO] [1670626295.127324722] [rgbd_odometry]: Odometry: guess_frame_id         = 
[rgbd_odometry-1] [INFO] [1670626295.127596929] [rgbd_odometry]: Odometry: guess_min_translation  = 0.000000
[rgbd_odometry-1] [INFO] [1670626295.127701382] [rgbd_odometry]: Odometry: guess_min_rotation     = 0.000000
[rgbd_odometry-1] [INFO] [1670626295.127774378] [rgbd_odometry]: Odometry: guess_min_time         = 0.000000
[rgbd_odometry-1] [INFO] [1670626295.127843566] [rgbd_odometry]: Odometry: expected_update_rate   = 0.000000 Hz
[rgbd_odometry-1] [INFO] [1670626295.127916530] [rgbd_odometry]: Odometry: max_update_rate        = 0.000000 Hz
[rgbd_odometry-1] [INFO] [1670626295.130965364] [rgbd_odometry]: Odometry: wait_imu_to_init       = true
[rgbd_odometry-1] [INFO] [1670626295.131415020] [rgbd_odometry]: Odometry: stereoParams_=0 visParams_=1 icpParams_=0
[rtabmap-2] [INFO] [1670626295.261323610] [rtabmap]: rtabmap(maps): map_filter_radius          = 0.000000
[rtabmap-2] [INFO] [1670626295.264712046] [rtabmap]: rtabmap(maps): map_filter_angle           = 30.000000
[rtabmap-2] [INFO] [1670626295.265086466] [rtabmap]: rtabmap(maps): map_cleanup                = true
[rtabmap-2] [INFO] [1670626295.265442965] [rtabmap]: rtabmap(maps): map_always_update          = false
[rtabmap-2] [INFO] [1670626295.265581789] [rtabmap]: rtabmap(maps): map_empty_ray_tracing      = true
[rtabmap-2] [INFO] [1670626295.265679138] [rtabmap]: rtabmap(maps): cloud_output_voxelized     = true
[rtabmap-2] [INFO] [1670626295.265763558] [rtabmap]: rtabmap(maps): cloud_subtract_filtering   = false
[rtabmap-2] [INFO] [1670626295.265956721] [rtabmap]: rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[rtabmap-2] [INFO] [1670626295.266587218] [rtabmap]: rtabmap(maps): octomap_tree_depth         = 16
[rgbd_odometry-1] [INFO] [1670626295.435866994] [rgbd_odometry]: odometry: Subscribing to IMU topic /imu/data
[rgbd_odometry-1] [INFO] [1670626295.436400143] [rgbd_odometry]: odometry: qos_imu = 0
[rgbd_odometry-1] [INFO] [1670626295.472253798] [rgbd_odometry]: RGBDOdometry: approx_sync    = false
[... 124 lines cut ...]
[rtabmapviz-3]    /odom \
[rtabmapviz-3]    /camera/color/image_raw \
[rtabmapviz-3]    /camera/realigned_depth_to_color/image_raw \
[rtabmapviz-3]    /camera/color/camera_info
[rgbd_odometry-1] [WARN] [1670626315.561941084] [rgbd_odometry]: rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rgbd_odometry-1] rgbd_odometry subscribed to (exact sync):
[rgbd_odometry-1]    /camera/color/image_raw,
[rgbd_odometry-1]    /camera/realigned_depth_to_color/image_raw,
[rgbd_odometry-1]    /camera/color/camera_info
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
^X[static_transform_publisher-7] [INFO] [1670626315.770534634] [rclcpp]: signal_handler(signal_value=2)
[rtabmap-2] [INFO] [1670626315.770580236] [rclcpp]: signal_handler(signal_value=2)
[imu_filter_madgwick_node-6] [INFO] [1670626315.770678449] [rclcpp]: signal_handler(signal_value=2)
[rgbd_odometry-1] [INFO] [1670626315.770725587] [rclcpp]: signal_handler(signal_value=2)
[rgbd_odometry-1] [WARN] [1670626315.774815131] [rgbd_odometry]: rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.

`

When I run ros2 topic info /camera/color/image_raw, I get:
Publisher count: 0

I'm not sure why it is not publishing. (realsense_d400.launch.py and realsense_d435i_infra.launch.py behave the same)

[image]
[image]

ryanmaxwell96, 2022-12-10:
I'm trying to do SLAM with the d435i realsense. Maybe there is a better package to do this? Like Isaac-Ros?

MartyG-RealSense, 2022-12-10:
Hi @ryanmaxwell96  There is a known issue with the RealSense IMU not working with JetPack 5, as this JetPack version is not yet officially supported by the librealsense SDK at the time of writing this (depth and color does work with it).  Any RealSense-compatible SLAM solution that makes use of the IMU is therefore likely to experience the same issue when used with JetPack 5 if it requires topics to be published by the RealSense ROS wrapper.

I consulted my Intel RealSense colleagues about this yesterday and they informed me that JetPack 5 support is being worked on and will be in a future SDK release, though I do not have a specific time estimate to provide unfortunately.

MartyG-RealSense, 2022-12-10:
An rtabmap_ros user of ROS2 Foxy at https://github.com/introlab/rtabmap_ros/issues/527 who was using **realsense_d400.launch.py** had the same set of errors.  There was not a solution reached in that case though.

My research did not find a solution elsewhere unfortunately.  A more typical approach to using rtabmap_ros with RealSense cameras is for the RealSense ROS wrapper to be roslaunched first and then the rtabma_ros launch instruction inputted secondly after the ROS wrapper launch has completed.

MartyG-RealSense, 2022-12-17:
Hi @ryanmaxwell96  Have you made any progress with your problem, please?  Thanks!

ryanmaxwell96, 2022-12-18:
I'm waiting on getting a Samsung nvme 256gb to expand memory. I think I'm unsure at the moment what ROS, what Jetpack, and what Ubuntu version to go with as I want SLAM with the d435i and Isaac ROS looks good, thus needing to expand memory from 32GB to 256gb. 

I don't suppose you'd have a suggestion as for which packages I should download, to do SLAM via ROS with Jetson Xavier AGX and D435i?

MartyG-RealSense, 2022-12-18:
A RealSense user with a D435i, ROS2 Foxy and an Xavier Nx at https://github.com/IntelRealSense/realsense-ros/issues/2387 used **slam_toolbox**

https://github.com/SteveMacenski/slam_toolbox

ryanmaxwell96, 2022-12-22:
Ok so on the ros rtabmap tutorial website you shared, what commands should I use?

I tried to run the command:

ros2 launch rtabmap_ros rtabmap.launch.py \ ... everything else as written in tutorial for D435i

But I still get a blank Rtabmap screen as shown

MartyG-RealSense, 2022-12-23:
If you launched the camera with **ros2 launch rtabmap_ros rtabmap.launch.py** and added "everything else as written in tutorial for D435i", do you mean the D435i instructions from the **HandHeldMapping** rtabmap_ros tutorial please?

[image]

If so then using **unite_imu_method:=linear_interpolation** does not work with the **ros2_development** wrapper.  Instead, unite_imu_method's modes are set with a number.  The linear_interpolation mode is '2'.

**unite_imu_method:=2**

And align_depth:=true in the ros2_development wrapper is now **align_depth.enable:=true**

ryanmaxwell96, 2022-12-23:
Thanks again for being responsive.

I ran these in order in 3 separate terminals

ros2 launch realsense2_camera rs_launch.py \
    align_depth.enable:=true \
    unite_imu_method:=2 \
    enable_gyro:=true \
    enable_accel:=true
    
    
ros2 run imu_filter_madgwick imu_filter_madgwick_node     
	--ros-args --remap _use_mag:=false \    
	--ros-args --remap _publish_tf:=false \    
	--ros-args --remap _world_frame:=enu \    
	--ros-args --remap /imu/data_raw:=/camera/imu \    
	--ros-args --remap /imu/data:=/rtabmap/imu
	
ros2 launch rtabmap_ros rtabmap.launch.py

imu_filter_madgwick_node gives errors as expected. rs_launch.py gives a warning that the D435i is running as if it was on a USB 2 but depth and rgb work fine in realsense viewer.
And rtabmap.launch.py runs the same as before with nothing showing up including the image.

When I run RVIZ2, at the very least i can get the image.

MartyG-RealSense, 2022-12-23:
Thanks very much for the update.  Does the USB 2 behaviour become USB 3 if **initial_reset:=true** is added to the rs_launch.py launch instruction?

ryanmaxwell96, 2022-12-23:
No unfortunately it stays as usb2.1

MartyG-RealSense, 2022-12-23:
Do you still have the original USB3 USB-C cable supplied with the D435i?

ryanmaxwell96, 2022-12-23:
> Do you still have the original USB3 USB-C cable supplied with the D435i?

No unfortunately it did not come with the camera when I bought it

MartyG-RealSense, 2022-12-23:
Are you using a USB hub?  If the cable is USB 3 but the hub is a USB 2 hub then the camera would be treated as a USB 2 device.

ryanmaxwell96, 2022-12-23:
> Are you using a USB hub and have the camera plugged into it?  If the cable is USB 3 but the hub is a USB 2 hub then the camera would be treated as a USB 2 device.

My camera is plugged directly into the Jetson USB 3.0 port

MartyG-RealSense, 2022-12-23:
Assuming that the stock cable is USB 3 since the ports are USB 3, if the cable is not damaged (e.g through excessive bending) then that would shift the suspicion towards a compatibility issue with JetPack 5.

ryanmaxwell96, 2022-12-23:
> Assuming that the stock cable supplied with the Jetson AGX Xavier H01 developer kit is USB 3 since the ports are USB 3, if the cable is not damaged (e.g through excessive bending) then that would shift the suspicion towards a compatibility issue with JetPack 5.

I tried with a standard Apple USB to USB C cable (unsure whether 3.0 or 2.0), but it gave same results in the viewer.

At any rate, if it gives depth and rgb info just fine in the viewer, I find it hard to believe this is the reason why rtabmap will not show anything.

Perhaps there is a transform issue? In RVIZ, it does not give any other options other than map for the base link and global is not set

MartyG-RealSense, 2022-12-24:
My knowledge of rtabmap_ros is admittedly limited.  My research of your question though found a number of references that advised that in order to visualize a 3D point cloud in RViz, the **MapCloud** plugin for RViz should be added if you have not done so already, as advised at the top of Section**2.  Mapping Mode** of the rtabmap_ros guide.

http://wiki.ros.org/rtabmap_ros#Map_Cloud_Display

I also recall that in order to render the data in RViz, **rtabmapviz** should be set to false and **rviz** to true, like in the opernsource_tracking launch file.

https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/launch/opensource_tracking.launch#L23-L24

ryanmaxwell96, 2022-12-27:
Ok thanks I got MapCloud up and running. 

I still have the transform issues. Map is the only frame that is listed but it gives errors as not being found. I think I need something about needing a static transform right? I will need to look further into this.

[image]

MartyG-RealSense, 2022-12-27:
I am not familiar with this subject in ROS2.  I do apologize.  Please do update me if you make progress in your investigation.

ryanmaxwell96, 2022-12-27:
I was playing around with the tutorial

https://github.com/introlab/rtabmap/wiki/Kinect-mapping

And after changing frame to 640x480 at 30Hz, I finally got my first map!

The blue dots represent how the camera was moving, but doesn't this mean that the IMU is working?
[image]

ryanmaxwell96, 2022-12-28:
I think I will close out this issue as it has gone in many different rabbit trails, sorry about that!  At least I have a map now. But if I find out any more about ROS2 transform issue, I will let you know. Thanks for your prompt communication!

closed 2022-12-28 by ryanmaxwell96

MartyG-RealSense, 2022-12-28:
It's no trouble at all.  It's great to hear that you made significant progress!  Thanks very much for the update, and please do share any new ROS2 transform insights that you discover.  :)

jsb9945, 2023-01-28:
Hello, I want to create a 3d map using d435i. My current environment is ubuntu 20.04 and ros2 foxy.

I ran this command

ros2 launch realsense2_camera rs_launch.py
align_depth.enable:=true
unite_imu_method:=2
enable_gyro:=true
enable_accel:=true

i got error

[INFO] [launch]: All log files can be found below /home/csrl/.ros/log/2023-01-28-16-34-13-313569-csrl-System-Product-Name-33149
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [33151]
[realsense2_camera_node-1] [INFO] [1674891253.617941112] [RealSenseCameraNode]: RealSense ROS v3.2.2
[realsense2_camera_node-1] [INFO] [1674891253.618003537] [RealSenseCameraNode]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1674891253.618015590] [RealSenseCameraNode]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1674891253.661201263] [RealSenseCameraNode]: Device with serial number 943222072377 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1674891253.661252468] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb2/2-3/2-3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1674891253.661269339] [RealSenseCameraNode]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1674891253.661834474] [RealSenseCameraNode]: Device with port number 2-3 was found.
[realsense2_camera_node-1] [INFO] [1674891253.661858348] [RealSenseCameraNode]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1674891253.663831455] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-1] [ERROR] [1674891253.666404722] [RealSenseCameraNode]: Failed to set parameter: unite_imu_method. expected [string] got [integer]
[realsense2_camera_node-1] expected [string] got [integer]
[realsense2_camera_node-1] terminate called after throwing an instance of 'rclcpp::ParameterTypeException'
[realsense2_camera_node-1] what(): expected [string] got [integer]
[ERROR] [realsense2_camera_node-1]: process has died [pid 33151, exit code -6, cmd '/home/csrl/ros3_ws/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_baofn84g'].

How can i solve it?

MartyG-RealSense, 2023-01-28:
Hi @jsb9945  It appears that you are using some instructions that only work in the **ros2-development** ROS2 wrapper, whilst wrapper 3.2.2 belongs to a previous generation of ROS2 wrapper that has different commands and is now obsolete.  The launch commands for these functions in 3.2.2 should be:

```
ros2 launch realsense2_camera rs_launch.py align_depth:=true enable_gyro:=true
enable_accel:=true unite_imu_method:=linear_interpolation
```

jsb9945, 2023-01-29:
Hello. Thank you for your quick reply. Which version of the wrapper should I use?

MartyG-RealSense, 2023-01-29:
The version of the current generation of the ROS2 wrapper that should be used with librealsense 2.51.1 is **4.51.1**.

https://github.com/IntelRealSense/realsense-ros/releases/tag/4.51.1

However, if the previous generation wrapper 3.2.2 works for you and provides the results that you need then you may not need to change wrappers.  You will have to move to the current wrapper generation if you use a librealsense version newer than 2.51.1 in future though.

jsb9945, 2023-01-29:
Thank you very much. I solved problem!!

MartyG-RealSense, 2023-01-29:
That's great to hear, @jsb9945 - thanks very much for the update!

MrDarknessWolf, 2023-05-09:
question where did either realsense_d400.launch.py and the rtabmap.launch.py for rtabmap_ros went ? every time when executing the command it says its missing from the folder

MartyG-RealSense, 2023-05-10:
Hi @MrDarknessWolf  This launch file is still present in the 'rtabmap_examples' directory of the **ros2** branch of rtabmap_ros but not in the same folder on the main **master** branch.

https://github.com/introlab/rtabmap_ros/tree/ros2/rtabmap_examples/launch