realsenseai/realsense-ros #2575: Pointcloud is not generating on Jetson Orin 5.0.2

naitiknakrani, 2022-12-12:
<html>
<body>

Required Info |  
-- | --
Camera Model | D435
Operating System & Version | Jetson Linux with Jetpack 5.0.2 Ubuntu 20.04
Kernel Version (Linux Only) | 5.10.104-tegra
Platform | Jetson Orin
SDK Version | 2.53.1
ros version | Galactic

</body>
</html>

Hi, We are facing issue generating Pointcloud from D435 using https://github.com/IntelRealSense/realsense-ros (ros2-developement branch)

We have launched https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_launch.py with following parameters set

depth_module.profile = 640x480x30
rgb_camera.profile = 640x480x30
pointcloud.enable = true
pointcloud.ordered_pc =true

These are the "_ros2 launch realsense2_camera rs_launch.py_" command logs

[INFO] [launch]: All log files can be found below /home/orin-2/.ros/log/2022-12-09-21-49-35-969045-orin-2-12585
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [12587]
[realsense2_camera_node-1] [INFO] [1670602776.706797176] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1670602776.707053240] [camera.camera]: Built with LibRealSense v2.53.1
[realsense2_camera_node-1] [INFO] [1670602776.707096216] [camera.camera]: Running with LibRealSense v2.53.1
[realsense2_camera_node-1] [INFO] [1670602776.737093976] [camera.camera]: Device with serial number 140122075906 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1670602776.737246552] [camera.camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-3/2-3.2/2-3.2:1.0/video4linux/video6 was found.
[realsense2_camera_node-1] [INFO] [1670602776.737279608] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1670602776.737726264] [camera.camera]: Device with port number 2-3.2 was found.
[realsense2_camera_node-1] [INFO] [1670602776.737763544] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1670602776.743259256] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1670602776.745089816] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1670602776.745149048] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1670602776.745189816] [camera.camera]: Device Serial No: 140122075906
[realsense2_camera_node-1] [INFO] [1670602776.745215288] [camera.camera]: Device physical port: /sys/devices/platform/3610000.xhci/usb2/2-3/2-3.2/2-3.2:1.0/video4linux/video6
[realsense2_camera_node-1] [INFO] [1670602776.745238168] [camera.camera]: Device FW version: 05.14.00.00
[realsense2_camera_node-1] [INFO] [1670602776.745258872] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1670602776.745277752] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1670602776.834070936] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1670602776.844143160] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [WARN] [1670602776.856120120] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1670602776.879124568] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1670602776.892496696] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1670602776.969457432] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1670602776.970660280] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1670602776.977486136] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1670602776.989767384] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1670602776.994559544] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1670602777.025822808] [camera.camera]:
[realsense2_camera_node-1] [WARN] [1670602777.333902392] [camera.camera]: No stream match for pointcloud chosen texture Process - Color

We observed that, by enabling pointcloud either RGB or depth frame stopped publishing. When we set pointcloud.enable =false, both RGB and Depth frame started publishing.

@MartyG-RealSense Kindly require your assistance.

MartyG-RealSense, 2022-12-12:
Hi @naitiknakrani  Problems with generating a pointcloud on Jetson devices specifically in the RealSense ROS wrapper is a known issue (the problem does not occur on desktop and laptop PCs).  Symptoms can include very low FPS or missing color.  Like in your case, the problem disappears when the pointcloud is disabled.  This issue is described in detail at https://github.com/IntelRealSense/realsense-ros/issues/1967

There are a couple of workarounds on ROS1 but the options for resolving it are more limited on ROS2 unfortunately.  The last known configuration for Jetson that was able to generate a pointcloud on ROS2 was librealsense **2.48.0** and ROS2 wrapper **3.2.2**.

naitiknakrani-eic, 2022-12-15:
Thanks @MartyG-RealSense  for the update.

MartyG-RealSense, 2022-12-22:
Hi @naitiknakrani  Do you require further assistance with this case, please?  Thanks!

naitiknakrani-eic, 2022-12-22:
No Thank you.

MartyG-RealSense, 2022-12-22:
Thanks very much @naitiknakrani-eic for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-12-22 by MartyG-RealSense

mesmerizingengineers, 2023-01-12:
Hello, I am experiencing a similar issue on ROS1 Noetic for the Jetson Orin. I previously was able to get the realsense2_camera on the Orin but re flashed it recently and it no longer works. 

This is the process I followed for the installation of realsense2_camera and the realsense SDK:

1. Installed librealsense SDK by following steps in [Ubuntu installation guide for 20.04](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) except for kernel patching procedure.
2. Tested Realsense-viewer. all works well.
3. installed ros-noetic-realsense2-camera with method 1.

When I try to run `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

it seems to throw some warnings and when I try to observe the nodes' images in rviz it will only display the depth image while the RGB image will very occasionally display for a short period and then cut to "NO IMAGE". I can't seem to get a point cloud to display which seems to indicate that these nodes aren't functioning properly. Is there any known solution to this? If not are there any steps I could follow to diagnose the issue?

[image]

Output:
`SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
[... 86 lines cut ...]
[ INFO] [1673545039.675543312]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1673545039.675666896]: setupPublishers...
[ INFO] [1673545039.680751344]: Expected frequency for depth = 30.00000
[ INFO] [1673545039.735390448]: Expected frequency for color = 30.00000
[ INFO] [1673545039.768211376]: setupStreams...
 12/01 09:37:20,005 WARNING [281472326144368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1673545040.056821712]: SELECTED BASE:Depth, 0
[ INFO] [1673545040.073499632]: RealSense Node Is Up!
[ WARN] [1673545040.124059536]: 
 12/01 09:37:20,126 WARNING [281472326144368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/01 09:37:20,176 WARNING [281472326144368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/01 09:37:20,229 WARNING [281472326144368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/01 09:37:20,427 WARNING [281472326144368] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[... cut ...]
 12/01 09:37:20,989 WARNING [281472326144368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/01 09:37:21,548 WARNING [281472326144368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/01 09:37:51,207 ERROR [281471804473712] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
[ INFO] [1673545246.107805936]: publishing unordered pointcloud.
[ WARN] [1673545246.464449488]: No stream match for pointcloud chosen texture Process - Color
`

MartyG-RealSense, 2023-01-12:
Hi @mesmerizingengineers  If you use Method 1 then it installs both librealsense and the ROS wrapper from packages at the same time.  If you have already installed librealsense before performing Method 1 then you will then have two librealsense installations on the same computer that may conflict with each other.

Also, the **installation.md** guide for source code install is not suitable for use with Jetson boards as the guide is intended for x86 / x64 architecture computers such as desktop and laptop PCs instead of Jetson's Arm architecture. There is a dedicated official librealsense installation guide for Jetson boards at the link below.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

I would recommend installing librealsense first from packages or source code using the Jetson-specific instructions in the above guide, and then secondly building the ROS wrapper from source code using the wrapper's **Method 2** instructions.

If you wish to consider patching the Jetson's kernel then there is a dedicated patch script for Jetson called **./scripts/patch-realsense-ubuntu-L4T.sh**

mesmerizingengineers, 2023-01-12:
> Hi @mesmerizingengineers If you use Method 1 then it installs both librealsense and the ROS wrapper from packages at the same time. If you have already installed librealsense before performing [...]

Thanks so much for the reply.

My apologies I actually misremembered the initial procedure I took in installing librealsense. I actually did follow the instructions on that page. It would not allow me to install it through debian packages when I last tried. I assumed that it just hadn't been updated for Jetson Orin yet so I opted for building from source using native backend. After looking [at this posted issue](https://github.com/IntelRealSense/librealsense/issues/10722#issuecomment-1230250762) I did what other users suggested and skipped the kernel patching procedure.

What I did was:

1. Follow instructions for building from source using native backend minus the kernel patching.
2. Went to https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md to install missing components (Steps 1, 2, 3 and 5 is what I assumed this meant) as per the instructions suggestions.
3. Continued with instructions until installation was complete
4. Ran realsense-viewer

Is it safe to assume that I have correctly built librealsense from source at this point? I assumed it was ok since I got realsense-viewer working on it.

I see your point about the two installations possibly conflicting with one another. So is it the case that following method 2 for realsense-ros (ROS1 branch) may fix the weird issues with the nodes described? Is purging librealsense2 and realsense2-camera all I need to do to insure that there will be no further conflicts when I try to install these packages from the source?

MartyG-RealSense, 2023-01-13:
There is a discussion at https://github.com/IntelRealSense/librealsense/issues/10722 about installing on Orin.  One RealSense user was able to install from source code using JetPack 5 and building the source code with the **RSUSB backend** installation method to bypass the kernel so that patching is not necessary, and two other users were able to successfully build from source code 'natively' (non-RSUSB) without patching the kernel.

One of the users who built without kernel patching was also able to sucessfully use the RealSense ROS wrapper with ROS2 Humble.

If librealsense is built from source code natively then it is always ideal to apply a kernel patch if possible, though it is possible to run the SDK without patching.

It is a positive sign for your installation that realsense-viewer is working correctly.  Sometimes though, a build may work fine with realsense-viewer but have problems with a RealSense ROS wrapper launch.  So a problem-free experience with realsense-viewer does not guarantee that ROS will work fine too.

If you are going to build librealsense and the ROS wrapper from source code then I would recommend purging all RealSense related packages from your computer, yes, using the command below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

If both package and source code builds have been performed on the same computer and there are still problems even after having purged the packages, I would recommend completely deleting the **catkin_ws** catkin workspace folder and creating a new one, then building the wrapper again from source code.

mesmerizingengineers, 2023-01-14:
> There is a discussion at [IntelRealSense/librealsense#10722](https://github.com/IntelRealSense/librealsense/issues/10722) about installing on Orin. One RealSense user was able to install from source code using JetPack 5 and building the source code with the **RSUSB backend** installation method to bypass the kernel so that patching is not necessary, and two other users were able to successfully build from source code 'natively' (non-RSUSB) without patching the kernel.
>
> One of the users who built without kernel patching reported at [IntelRealSense/librealsense#10722 (comment)](https://github.com/IntelRealSense/librealsense/issues/10722#issuecomment-1230250762) that they were also able to sucessfully use the RealSense ROS wrapper with ROS2 Humble.
> 
> As well as successfully installing without kernel patching, that Orin user was also able to achieve a working build using the alternative **libuvc backend** installation method at the link below.
> 

I'm familiar with this discussion. I opted for the "native" instructions since it seemed easier at the time and I had actually gotten it to work at one point on this device before I had to re flash it. I find it rather odd that it didn't work this time even though I'm relatively sure I followed the same steps as last time. 

> https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md
> 
> If librealsense is built from source code natively then it is always ideal to apply a kernel patch if possible, though it is possible to run the SDK without patching.
> 

It seems that currently there is no way to successfully patch the kernel on the Jetson Orin unless I have missed something. So not sure what to do about this besides use the RSUSB backend if that's the problem.

Thanks so much for the help again. I really appreciate it as the person posting this is a rather novice ROS user.

I will try to purge all real-sense related packages and uninstall the current librealsense. I'm likely going to try just installing through debian packages to see if that works ok after purging. If it doesn't, then I will try to purge once again and try the three different methods of installing librealsense that people had luck with in that discussion. 

I will report on if the issue is resolved after I try these things.

mesmerizingengineers, 2023-01-14:
Hello, I documented the steps I took in attempting to resolve this issue. Luckily I have a restore point for this device that can retract all steps I took here. The easiest solution seems to be a full system reset and just install through RSUSB method and Method 2 for realsense-ros. However, resetting the system isn't really an option for my team currently.

What I tried was:

1. Uninstalled librealsense and purged all realsense packages
`sudo make uninstall && sudo make clean` in librealsense directory
`sudo apt purge ros-noetic-realsense2-camera ros-noetic-librealsense2` to purge packages installed through debian
`sudo apt purge ros-noetic-realsense2-description`
When trying `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`
an error is produced:
`dpkg: error: --purge needs at least one package name argument` Not sure why that is. Any comment on this is appreciated.

2. Tried installing with Method 1 again. Same result as before i.e depth nodes display properly but RGB nodes do not.

3. Purged once again 

4. Installed librealsense using RSUSB-backend using [these instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md). Unfortunately I forgot to put `-DBUILD_WITH_CUDA=true` but installed successfully and recognized devices.

5.  Followed instructions for realsense-ros installation Method 2 in my current catkin workspace with all my packages installed.

6. `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud` Recieved an error:
`[ INFO] [1673673112.011980646]: Setting Dynamic reconfig parameters.
[ INFO] [1673673116.627763477]: Done Setting Dynamic reconfig parameters.
/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /home/hawkeyeiii/catkin_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1Ev
[camera/realsense2_camera_manager-2] process has died [pid 25047, exit code 127, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/hawkeyeiii/.ros/log/f396b3d4-93c9-11ed-a2ec-90e868bc7c5f/camera-realsense2_camera_manager-2.log].
log file: /home/hawkeyeiii/.ros/log/f396b3d4-93c9-11ed-a2ec-90e868bc7c5f/camera-realsense2_camera_manager-2*.log
`
7. Searched if any one has solved this error... Eventually read your comment in [this discussion](https://github.com/IntelRealSense/realsense-ros/issues/2354#issuecomment-1140189910) about this error being rare and unsolved. 

Since I'm working with a team a full system reset isn't something my teammates would really appreciate.... and it would also be better not to tamper with the catkin workspace as there's a bit of work in there that I can't delete currently. 

Unfortunately I haven't gotten a chance to go past this point but will tomorrow. Do you know if just making a separate catkin ws with only realsense-ros would suffice for testing if the catkin ws is the issue? Or is will the other catkin ws interfere with it still? 

Once again, I really appreciate your patience and help.

MartyG-RealSense, 2023-01-14:
Regarding the message **dpkg: error: --purge needs at least one package name argument**, my understanding is that it means that it found no RealSense packages to remove.

Yes, the error **undefined symbol: _ZN2cv3MatC1Ev** does not have a clearly defined cause and solution.

My recollection is that the Intel RealSense team member who created the RealSense ROS1 wrapper advised that only RealSense files should be in the ** catkin_ws/src** workspace and not non-RealSense files.  So making a separate catkin_ws at a different path with only realsense-ros for testing purposes sounds like a good strategy.

mesmerizingengineers, 2023-01-16:
Hello. I'm happy to report that I solved the Issue by following [this gentleman's solution](https://github.com/IntelRealSense/realsense-ros/issues/2467#issuecomment-1268417031) exactly. making another catkin_ws was not enough to resolve the error. The editing of the CMakeList.txt was what I missed.

Thanks you so much for your patience and help. I'm currently not experiencing any errors or problems viewing the pointcloud in rviz.

MartyG-RealSense, 2023-06-07:
Hi @naitiknakrani  The new 2.54.1 version of librealsense with JetPack 5.0.2 support is now available.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1

Release notes:
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2541

naitiknakrani-eic, 2023-06-07:
Thanks for the update 👍 @MartyG-RealSense