realsenseai/realsense-ros #2576: Opensource Tracking

alvaropascuaI, 2022-12-12:
Hi, I'm trying  to launch file opensource_tracking.launch, but i have error in ros topics i dont know if with only launch file all ready works or i need to launch something more? I follow next video but dont work in viewer, i change some topic but i dont know what i need to change more

[image]

MartyG-RealSense, 2022-12-12:
Hi @alvaropascuaI  Are you using a RealSense camera model that is equipped with an IMU (D435i or D455).

If you have one of these cameras, have you enabled the IMU topics please, as these are disabled by default in the RealSense ROS wrapper.

`roslaunch realsense2_camera opensource_tracking.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation`

alvaropascuaI, 2022-12-12:
I am using d435i, but if i launch with this command i have the same problem in topics i dont know what i need to put in configuration, i follow this [tutorial](https://www.youtube.com/watch?v=mguXtybmZv4), but dont generate map

MartyG-RealSense, 2022-12-12:
Your image shows that a color image and a pointcloud are being generated in Rviz.  What are the ROS topic errors that you are receiving, please?

In the **Frames** section of the RViz options, did you untick all settings except **camera_link** and **map**

[image]

MartyG-RealSense, 2022-12-12:
Assuming that the video is based on Intel's D435i SLAM guide at the link below, did you install the prerequisite modules **imu_filter_madgwick** and **robot_localization**

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

As mentioned above, when using this guide you should also untick everything under Frames except camera_link and map

alvaropascuaI, 2022-12-12:
I install the modules of rtabmap_ros, imu_filter_madgwick and robot_localization and work worse than before in TF Frames i can check the same topics, and i have the next error in terminal, i try to install another time the realsense-ros-noetic but nothing

[image]

MartyG-RealSense, 2022-12-12:
A Noetic user of the SLAM guide at https://github.com/IntelRealSense/realsense-ros/issues/2494 experienced the same errors.  I suggested using the Noetic version of the modules instead of the original Kinetic ones in the guide.

sudo apt-get install ros-noetic-robot-localization

sudo apt-get install ros-noetic-rtabmap-ros

sudo apt install ros-noetic-imu-filter-madgwick

MartyG-RealSense, 2022-12-12:
The **Viewing Map Data** section of the guide at the link below states that when running RTABmap in mapping mode, you can open rviz and add the MapCloud panel.  Then, configure the correct listening topic as /rtabmap/mapData, and activate the checkbox **Download map**.

https://admantium.medium.com/ros-simultaneous-mapping-and-localization-with-rtabmap-e527e6a6716

MartyG-RealSense, 2022-12-13:
Could you provide more information please about what you mean about activating stereo with ROS?  Thanks!

alvaropascuaI, 2022-12-13:
I want to launch this [file](https://github.com/introlab/rtabmap_ros/blob/ros2/launch/demo/demo_stereo_outdoor.launch) with camera D455 because i read [here](https://www.intel.es/content/www/es/es/products/sku/205847/intel-realsense-depth-camera-d455/specifications.html) he used a tecnology stereo but i think im confused
[image]

MartyG-RealSense, 2022-12-13:
Thanks very much for the clarification.  This rtabmap_ros launch file can be launched with the instruction below.

`roslaunch rtabmap_ros demo_stereo_outdoor.launch`

This may be a situation though where you would need to launch the RealSense ROS wrapper first.

`roslaunch realsense2_camera opensource_tracking.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation`

And then launch the rtabmap_ros launch file secondly after the RealSense ROS wrapper has completed launching.

`roslaunch rtabmap_ros demo_stereo_outdoor.launch`

MartyG-RealSense, 2022-12-19:
Hi @alvaropascuaI  Do you require further assistance with tis case, please?  Thanks!

alvaropascuaI, 2022-12-19:
Sorry, I thought that I answered you, I deleted everything and then I installed everything again, but in rviz I see this
In tf frames I have the following, the map frame is missing
[image]
[image]

MartyG-RealSense, 2022-12-19:
Fixed Frames is set to 'camera_link' instead of **map** like it was in your earlier image.

[image]

Once Fixed Frames is changed to 'map', it should appear in the Frames list.  Untick everything in that list except **camera_link** and **map**.

alvaropascuaI, 2022-12-19:
I change fixed frame for map and generate map but in tf frames no appeard, and how i can do that with 2 cameras ?
[image]

MartyG-RealSense, 2022-12-19:
Are you able to launch with two cameras if you have two seperate ROS terminals open (one for each camera) with the two cameras distinguished by their serial numbers and a defined camera name?

**Terminal 1**
`roslaunch realsense2_camera opensource_tracking.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation camera:=cam_1 serial_no:= camera:=cam_1 serial_no:=<serial number of the first camera>`

**Terminal 2**
`roslaunch realsense2_camera rs_opensource_tracking.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation camera:=cam_1 serial_no:= camera:=cam_2 serial_no:=<serial number of the second camera>`

alvaropascuaI, 2022-12-19:
If I do that, I can start both cameras, but I don't know how I can use the opensource file to rebuild with 2 cameras, because I thought that if I send information from 2 cameras for the same themes, rviz can do the rebuild, because I don't know how. You can do the reconstruction with 2 cameras and save the map cloud, I can't think of anything else because I tried with RTABMAP but it doesn't work

MartyG-RealSense, 2022-12-19:
In Intel's guide for creating a combined pointcloud from two cameras and displaying it in RViz, they also launch the cameras in separate ROS terminals.  In Step 4 of the guide, they set Fixed Frame to refer to camera 1 and add a topic from each camera in RViz.

```
Set “Fixed Frame” to “cam_1_link”
Add -> By topic -> /cam_1/depth/color/points/PointCloud2
Add -> By topic -> /cam_2/depth/color/points/PointCloud2
```

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

alvaropascuaI, 2022-12-19:
Yes, i do the guide u told me but is only viewer and i need to generate a map cloud with 2 cameras, you understand me ?

MartyG-RealSense, 2022-12-19:
I recall that a RealSense ROS user asked the Intel RealSense team member who was the developer of the RealSense ROS1 wrapper about the possibility of obtaining the orientations of two cameras using an edited version of opensource_tracking.launch for handling two cameras instead of one.  The wrapper creator responded to the question at https://github.com/IntelRealSense/realsense-ros/issues/1609#issuecomment-755169040

There was another RealSense ROS user who attempted to adapt the D435i SLAM guide to use opensource_tracking.launch for two cameras and they did not succeed.

alvaropascuaI, 2022-12-20:
I want to launch the open source file with saved topics in ROS 2 and convert them to ROS 1 in order to use the open source file with the rosbags-convert library but I get the following error

[image]

MartyG-RealSense, 2022-12-20:
The ROS2 to ROS1 / ROS1 to ROS2 message bridging project at https://github.com/ros2/ros1_bridge may be helpful for cross-communication between ROS2 and ROS1.

alvaropascuaI, 2022-12-21:
In my ROS1 if I launch the file like this it works for me I wanted to leave the most basic to try to replicate it in ROS 2, I also launch the camera separately with the following command

ROS 1
`roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_pointcloud:=true align_depth:=true`
ROS 2
`ros2 launch realsense2_camera rs_launch.py initial_reset:=true enable_pointcloud:=true serial_no:=_153222070827 align_depth:=true`
[image]

and in ROS 2 for try to replicate the opensource file i launch next

`ros2 launch rtabmap_ros rtabmap.launch.py args:="--delete_db_on_start" rgb_topic:=/camera/color/image_raw depth_topic:=/camera/aligned_depth_to_color/image_raw camera_info_topic:=/camera/color/camera_info depth_camera_info_topic:=/camera/depth/camera_info rtabmapviz:=false rviz:=true`

when rviz open dont see nothing and i have the next 
[image]

MartyG-RealSense, 2022-12-21:
I recalled that we have previously discussed the subject of replicating opensource_tracking in ROS2 in August 2021 at https://github.com/IntelRealSense/realsense-ros/issues/2434#issuecomment-1205115900 and you also had problems with RViz then too.

A few days ago at https://github.com/IntelRealSense/realsense-ros/issues/2574#issuecomment-1356723219 I suggested to another RealSense ROS user to use **slam_toolbox** for SLAM in ROS2.

alvaropascuaI, 2022-12-21:
Yes, but in August i wanted launch with 1 camera to do reconstruction 3d and i found RTABMAP and works, but i need do it a reconstruction with 2 cameras and with RTABMAP i cant works

Regarding the slam_toolbox repository, it generates a 2D map for you and what I want is to generate a 3D map, even so, thank you very much

MartyG-RealSense, 2022-12-21:
A discussion at https://github.com/introlab/rtabmap_ros/issues/459 suggests that rtabmap_ros does not have dual camera support by default but can use multiple cameras if it is installed by cloning its source and catkin-make building it with the term **-DRTABMAP_SYNC_MULTI_RGBD=ON** included.

MartyG-RealSense, 2022-12-28:
Hi @alvaropascuaI  Was the advice in the above comment helpful, please?  Thanks!

MartyG-RealSense, 2023-01-09:
Case closed due to no further comments received.

closed 2023-01-09 by MartyG-RealSense

alvaropascuaI, 2023-01-09:
Hi @MartyG-RealSense, I was on vacation, I finally used NerfStudio for the rebuild, I am able to generate cloud maps as the rgb camera images work fine, thanks for everything

MartyG-RealSense, 2023-01-09:
That's great to hear that you achieved a solution, @alvaropascuaI - thanks very much for the update!