realsenseai/realsense-ros #2577: Realsense ROS node interfering with TIS ethernet camera

Lazar-FP, 2022-12-12:
Hi, we are having trouble running realsense l515 camera and a The Imaging Source ethernet camera on the same PC. We have Ubuntu 20.04 PC with Intel i7 and there we are running Ubuntu 20.04 docker image with ROS Noetic. Inside the docker, we installed realsense-ros package with apt install ros-noetic-realsense2-camera and driver for the ethernet camera https://github.com/TheImagingSource/tiscamera/releases/download/v-tiscamera-1.0.0/tiscamera_1.0.0.4005_amd64_ubuntu_2004.deb. We start realsense node with realsense2_camera/launch/includes/nodelet.launch.xml, but since we do not want to use it right away, we call service REALSENSE_CAMERA_NAME/enable with argument False. After that we start a gstreamer pipeline for our ethernet camera, but every 5 minutes to 1 hour we get SIGSEGV. For testing we are using fakesink:

```
$ gst-launch-1.0 tcamsrc serial="12345678" ! fakesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Caught SIGSEGV
#0  0x00007f7f8387999f in __GI___poll (fds=0x56409916a890, nfds=2, timeout=-1)
#1  0x00007f7f839ce36e in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007f7f839ce6f3 in g_main_loop_run ()
#3  0x00007f7f83b4eab3 in gst_bus_poll ()
#4  0x000056409811e6e5 in  ()
#5  0x000056409811d5fe in  ()
#6  0x00007f7f8378b083 in __libc_start_main (main=
#7  0x000056409811dc0e in  ()
```

When we turn off the realsense node or call service REALSENSE_CAMERA_NAME/enable with argument True, gstreamer pipeline for the ethernet camera does not throw errors, so it seems that the realsense node in disabled state is causing the issue.

Could you help us with this issue and suggest how to solve it?

[realsense_startup.txt](https://github.com/IntelRealSense/realsense-ros/files/10208658/realsense_startup.txt)

MartyG-RealSense, 2022-12-12:
Hi @Lazar-FP  Looking at your realsense_startup.txt log, I note that there are multiple **control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11** warning messages after launch has completed at the line 'RealSense Node is Up!'  If this warning is generating repeatedly in the log then it can indicate that there is a serious communication problem with the camera.  I wonder if these occur after the realsense2_camera node is set to false.

As discussed at https://github.com/IntelRealSense/realsense-ros/issues/1195 - if you would prefer not to stop the node by killing it with **rosnode kill realsense2_camera** then an alternative for saving processing may be to **unsubscribe** from topics and then resubscribe when you require them.

Another approach discussed in that case at https://github.com/IntelRealSense/realsense-ros/issues/1195#issuecomment-626750400 is to create two publishers, one that sends a topic's data and another that doesn't, and switch between the publishers to turn topic publishing on and off.

MartyG-RealSense, 2022-12-19:
Hi @Lazar-FP  Do you require further assistance with this case, please? Thanks!

Lazar-FP, 2022-12-19:
Hi @MartyG-RealSense, I apologize for the late reply and thank you for your suggestions. After launching the node, I get the control_transfer error every second or so in the console. It disappears if I call the enable service with argument false. Could you explain me what kind of communication problem could this be or what could I check to determine that?

Since Christmas holidays are coming, I would need a bit more time for testing that might continue in January, so please keep this issue open. Thanks!

MartyG-RealSense, 2022-12-19:
'Communications problems' can cover a range of possibilities.  This could include the stability of the USB connection.  You could check whether the cable might be 'bad' or damaged (for example, due to bending of the cable) but it does appear that it is related mainly to the true / false status of enable service.

It's fine to keep the issue open through January.  I have added a **wait to close** label to this issue to signify that it should be kept open.

MartyG-RealSense, 2023-01-16:
I do not have further comments to add.  It's great to hear that you were successful.  Thanks so much for sharing the details of what worked for you.  As you suggest, I will close the issue.  Thanks again!