realsenseai/realsense-ros #2580: D435i with Jetson Nano

CJdev99, 2022-12-15:
I upgraded a jetson nano to ubuntu 20.04 and am running ROS noetic. I installed the driver for the camera and the node runs, but is constantly sending
` control_transfer returned error, index: 768, error: resource temporarily available, number: 11`

Any ideas how I can get this to work with my setup?

MartyG-RealSense, 2022-12-15:
Hi @CJdev99  This **resource temporarily unavailable** warning, when generating continuously, can indicate that there is a serious communication problem with the camera.

 
Are you able to enable the depth stream or RGB stream on their own in the RealSense Viewer instead of enabling all three streams simultaneously?

Is there the possibility that the USB cable might have been damaged since the last time that you used the camera?

Has the camera been stored unused for the past couple of months in a place such as a box or drawer?

CJdev99, 2022-12-15:
Hi thanks for the response.
1.  I have tried all modes separately and both camera modes show “no frames received”. But, it appears the IMU is working as the orientation view does update.

 2.  With the jetson nano, there is no usbc port so I use usb-c to usb cable. With my windows laptop, I use the usbc port. Originally before use on the nano, I had always used the usbc port and it worked fine both on windows and linux. Now it doesn’t work on both linux and windows with either cable. I have checked both cables to make sure they work.

3. It has been sitting around on a desk/in a bag for about 3 months.

MartyG-RealSense, 2022-12-15:
There have been a few past cases where a 400 Series camera has stopped working correctly when used for the first time after a period of box or drawer storage for weeks / months.  This is a hardware issue rather than a software one.  https://github.com/IntelRealSense/librealsense/issues/10735 is an example of a case with a period when the camera was stored unused and then had a missing RGB stream and control_transfer returned errors.  The outcome of that case was that the RealSense user replaced the camera.

CJdev99, 2022-12-16:
Has there been any sort of fix for that I can try? Any chance to get a replacement sent in, or send it in and get it repaired? It's for a university robotics project and don't have much of a budget to buy another one.

MartyG-RealSense, 2022-12-16:
I will discuss your case with my Intel RealSense colleagues.  Thanks very much for your patience!

MartyG-RealSense, 2023-01-04:
Case closed due to resolution reached.

closed 2023-01-04 by MartyG-RealSense

dnovischi, 2023-01-08:
Hi @MartyG-RealSense,

Regarding the discussion in #812. I have succeed to install the D435i on both ubuntu 18.04 and 20.04 with ROS Noetic. In doing so, I have made some tutorials which are available [here](https://github.com/dnovischi/jetson-tutorials).

The most relevant  being  [Realsense D435i Installation for ROS Noetic on Ubuntu 18.04](https://github.com/dnovischi/jetson-tutorials/blob/main/jetson-nano-realsense-notetic-ubuntu-18-04-install.md#realsense-d435i-installation-for-ros-noetic-on-ubuntu-1804) where I have put in all the gory details of what I have done for: librealsense, realsense-ros and realsense firmware, to get it all working.

Anyway, I think some of the people here may have some use for it.

MartyG-RealSense, 2023-01-08:
Thanks so much @dnovischi for creating your guides and sharing them with the RealSense community!

dnovischi, 2023-01-10:
Hi @MartyG-RealSense,

I have tested the installation steps several times on Ubuntu 20.04 and the installation steps are the exact same.
I updated the guide to reflect that and also the link in the previous comment.

Side note:
It seems that somehow the D435i works better on Jetson Nano with Ubuntu 20.04. I receive all streams in a multiple host network setup at the frame rates I specify for each of them. However, in Ubuntu 20.04 I also receive the pointcloud (i.e. filter:=pointcloud)  stream data with no delay (real-time view even in rviz over the network). In Ubuntu 18.04 it works but there are delays. In both case I have used the exact same hardware setup (including wires and power source).
This is most likely due to a network configuration option in ubuntu 18.04. But I haven't been able to figure out exactly which one. Maybe others can pitch in.

MartyG-RealSense, 2023-01-11:
Thanks again @dnovischi for the continued feedback and updating of your guide!

CJdev99, 2023-01-31:
Thanks for the tutorial @dnovischi! I've been struggling with that for a while. I finally was able to get the camera working with the jetson nano but have some weird errors. In realsense viewer, I am able to view the stereo module, but there are no frames recieved for the rgb camera. In the terminal I get this error

`Exception was thrown during callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_`

and when I run the ROS package i get this error:

```
[ERROR] [1675203716.132865997]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1675203716.135037954]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1675203716.152188834]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-3] process has died [pid 9271, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/chase/.ros/log/a94f5750-a1b5-11ed-ba63-474209bbe0ee/camera-realsense2_camera-3.log].
log file: /home/chase/.ros/log/a94f5750-a1b5-11ed-ba63-474209bbe0ee/camera-realsense2_camera-3*.log
```
Any idea where the problem could be coming from? Thanks.

dnovischi, 2023-02-01:
Check the Realsense firmware version, it must be compatible with the librealsense version you have installed. And, deactivate auto-update option in realsense-viewer. The firmware list is located [here](https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads).

CJdev99, 2023-02-01:
I checked beforehand - I have firmware version 5.13.0.50 on realsense v2.50.0. I use the same setup on ubuntu 20.04 on a laptop with no problems.

dnovischi, 2023-02-02:
Its a very wired problem... you can try to delete the udev rules and put them again, maybe its an access problem. 

I have done again the whole setup on a brand new board and sdcard. It works as expected. The only suggestion I have is to start from scratch if everything else fails.